OE4T/meta-tegra #744: TX2 bootload update issue

lfdmn, 2021-06-29:
Hi,

I'm porting a TX2 image from warrior to dunfell-l4t-r32.4.3 after a recent shipment of D02 modules not compatible with prior jetpack revision. 

The booloader update succeeds
```
nv_update_engine -i
Nvidia A/B-Redundancy Update tool Version 1.2
current slot 0
Start updating slot 1
Tegra User Block Device: /dev/disk/by-partlabel
Tegra Boot Block Device: /dev/mmcblk0boot0
HEADER: MAGIC NVIDIA__BLOB__V2
HEX_VALUE 131072
BLOB_SIZE 42891591
HEADER_SIZE 40
NUMBER_OF_ELEMENTS 49
HEADER_TYPE 0
UNCOMP_SIZE 42891591
MB1_RATCHET_LV 98
MTS_RATCHET_LV 112
ROLLBACK_FUSE_LV 109
config COMPATIBLE_SPEC not found in /etc/nv_boot_control.conf
Device TN Spec: 3310-D01-1000-B.0-1-0-astro-tx2-mmcblk0p1
config COMPATIBLE_SPEC not found in /etc/nv_boot_control.conf
Device TN Spec: 3310-D01-1000-B.0-1-0-astro-tx2-mmcblk0p1
Device is fused board.
ENTRY_TABLE:
PART  POS  LEN  VER TNSPEC TYPE UPDATABLE
bpmp-fw  5920  533904  2    1  1
sce-fw  539824  125568  2    1  1
cpu-bootloader  665392  285216  2    1  1
mb2  950608  117568  2    0  1
spe-fw  1068176  95984  2    0  1
secure-os  1164160  366400  2    1  1
adsp-fw  1530560  108208  2    1  1
BMP  1638768  75973  2    1  1
mb1  1714741  99664  2    2  1
mts-bootpack  1814405  2082144  2    1  1
mts-preboot  3896549  63104  2    0  1
sc7  3959653  23632  2    1  1
bpmp-fw-dtb  3983285  605120  2  3310-B00-1000--1-0-astro-tx2-mmcblk0p1  1  0
bpmp-fw-dtb  4588405  605120  2  3310-B02-1000--1-0-astro-tx2-mmcblk0p1  1  0
bpmp-fw-dtb  5193525  605120  2  3310-C04-1000--1-0-astro-tx2-mmcblk0p1  1  0
bpmp-fw-dtb  5798645  605120  2  3310-D00-1000--1-0-astro-tx2-mmcblk0p1  1  0
bpmp-fw-dtb  6403765  605120  2  3310-D01-1000--1-0-astro-tx2-mmcblk0p1  1  1
bpmp-fw-dtb  7008885  605120  2  3310-D02-1000--1-0-astro-tx2-mmcblk0p1  1  0
bootloader-dtb  7614005  240576  2  3310-B00-1000--1-0-astro-tx2-mmcblk0p1  1  0
bootloader-dtb  7854581  240576  2  3310-B02-1000--1-0-astro-tx2-mmcblk0p1  1  0
bootloader-dtb  8095157  240576  2  3310-C04-1000--1-0-astro-tx2-mmcblk0p1  1  0
bootloader-dtb  8335733  240576  2  3310-D00-1000--1-0-astro-tx2-mmcblk0p1  1  0
bootloader-dtb  8576309  240576  2  3310-D01-1000--1-0-astro-tx2-mmcblk0p1  1  1
bootloader-dtb  8816885  240576  2  3310-D02-1000--1-0-astro-tx2-mmcblk0p1  1  0
VER  9057461  102  2  3310-B00-1000--1-0-astro-tx2-mmcblk0p1  0  0
VER  9057563  102  2  3310-B02-1000--1-0-astro-tx2-mmcblk0p1  0  0
VER  9057665  102  2  3310-C04-1000--1-0-astro-tx2-mmcblk0p1  0  0
VER  9057767  101  2  3310-D00-1000--1-0-astro-tx2-mmcblk0p1  0  0
VER  9057868  102  2  3310-D01-1000--1-0-astro-tx2-mmcblk0p1  0  1
VER  9057970  101  2  3310-D02-1000--1-0-astro-tx2-mmcblk0p1  0  0
BCT  9058071  3584  2  3310-B00-1000--1-0-astro-tx2-mmcblk0p1  2  0
BCT  9061655  3584  2  3310-B02-1000--1-0-astro-tx2-mmcblk0p1  2  0
BCT  9065239  3584  2  3310-C04-1000--1-0-astro-tx2-mmcblk0p1  2  0
BCT  9068823  3584  2  3310-D00-1000--1-0-astro-tx2-mmcblk0p1  2  0
BCT  9072407  3584  2  3310-D01-1000--1-0-astro-tx2-mmcblk0p1  2  1
BCT  9075991  3584  2  3310-D02-1000--1-0-astro-tx2-mmcblk0p1  2  0
MB1_BCT  9079575  50096  2  3310-B00-1000--1-0-astro-tx2-mmcblk0p1  0  0
MB1_BCT  9129671  50096  2  3310-B02-1000--1-0-astro-tx2-mmcblk0p1  0  0
MB1_BCT  9179767  50096  2  3310-C04-1000--1-0-astro-tx2-mmcblk0p1  0  0
MB1_BCT  9229863  50096  2  3310-D00-1000--1-0-astro-tx2-mmcblk0p1  0  0
MB1_BCT  9279959  50096  2  3310-D01-1000--1-0-astro-tx2-mmcblk0p1  0  1
[... 171 lines cut ...]
lseek(3, 2063360, SEEK_SET)             = 2063360
read(3, "\0NBC\3\3\2\0\17_a\7\1\0_b\0\0\246\337\326\262", 22) = 22
close(3)                                = 0
write(1, "Previous update on slot 1 failed"..., 33Previous update on slot 1 failed
) = 33
exit_group(0)                           = ?
+++ exited with 0 +++
```

The kernel cmd has the correct slot info ```... boot.slot_suffix=_b ...```

```/etc/nv_boot_control.conf``` looks ok to me
```
cat /etc/nv_boot_control.conf 
TNSPEC 3310-D01-1000-B.0-1-0-astro-tx2-mmcblk0p1
TEGRA_CHIPID 0x18
TEGRA_OTA_BOOT_DEVICE /dev/mmcblk0boot0
TEGRA_OTA_GPT_DEVICE /dev/mmcblk0boot1
``` 

The bootloader update works with Xavier based machines on the same branch.

Any hint where to look next?

Thanks,

madisongh, 2021-06-29:
What version of the BSP was on your A partition vs what was being installed on B?  If they are different, then the next question:

Do you `chroot` into the newly-installed (B) partition to run `nv-update-engine`, or were using the one in the A partition?  I know I've run into issues with cross-overs (trying to install a BUP from a later BSP using the tools/config from an older BSP version) not working, which is why I took the `chroot` path for the Mender (and other) integration.  Doing that should also eliminate the need to copy the `entry_table` file over, too, since it should end up in the B partition's filesystem.

lfdmn, 2021-06-30:
For this test I'm re-deploying the same bootloader payload. So the BSP is the same.

On slot A I ```nv_update_engine_install --install no-reboot```, mount the B partition, copy the ```entry_table```, reboot to slot B and run ```nv_update_engine_install --verify```

The entry_table looks fine to me too:
```
bpmp-fw,5920,533904,2,,1,1
sce-fw,539824,125568,2,,1,1
cpu-bootloader,665392,285216,2,,1,1
mb2,950608,117568,2,,0,1
spe-fw,1068176,95984,2,,0,1
secure-os,1164160,366400,2,,1,1
adsp-fw,1530560,108208,2,,1,1
BMP,1638768,75973,2,,1,1
mb1,1714741,99664,2,,2,1
mts-bootpack,1814405,2082144,2,,1,1
mts-preboot,3896549,63104,2,,0,1
sc7,3959653,23632,2,,1,1
bpmp-fw-dtb,6403765,605120,2,3310-D01-1000--1-0-astro-tx2-mmcblk0p1,1,1
bootloader-dtb,8576309,240576,2,3310-D01-1000--1-0-astro-tx2-mmcblk0p1,1,1
VER,9057869,101,2,3310-D01-1000--1-0-astro-tx2-mmcblk0p1,0,1
BCT,9072408,3584,2,3310-D01-1000--1-0-astro-tx2-mmcblk0p1,2,1
MB1_BCT,9279960,50096,2,3310-D01-1000--1-0-astro-tx2-mmcblk0p1,0,1
kernel,9380152,32067984,2,,1,1
kernel-dtb,42410440,240576,2,3310-D01-1000--1-0-astro-tx2-mmcblk0p1,1,1
```

```chroot``` sounds like a good idea. I'll give this a try.

lfdmn, 2021-07-01:
I'm actually getting the same problem with ```tegra-demo-distro```.

I did the following:
```bash
git clone https://github.com/OE4T/tegra-demo-distro.git
cd tegra-demo-distro
git checkout dunfell-l4t-r32.4.3
git submodule update --init

```
Changed ```demo-image-base.bb``` as:
```
DESCRIPTION = "Tegra demo base image"

require demo-image-common.inc

IMAGE_INSTALL_append = "\
    tegra-redundant-boot \
    tegra-bup-payload \
"
```
Build and flash
```
. ./setup-env --machine jetson-tx2
bitbake `demo-image-base
cd tmp/deploy/images/jetson-tx2/
mkdir demo-image-base && cd demo-image-base
tar -xvzf ../demo-image-base-jetson-tx2.tegraflash.tar.gz
# boot TX2 devkit to flashing mode
sudo ./doflash.sh
```
After the first boot
```
$ nv_update_engine --enable-ab
$ nvbootctrl dump-slots-info
magic:0x43424e00,             version: 3             features: 3             num_slots: 2
slot: 0,             priority: 15,             suffix: _a,             retry_count: 7,             boot_successful: 1
slot: 1,             priority: 14,             suffix: _b,             retry_count: 7,             boot_successful: 1

$ nv_update_engine --install

# after reboot
$ nvbootctrl dump-slots-info
magic:0x43424e00,             version: 3             features: 3             num_slots: 2
slot: 0,             priority: 0,             suffix: _a,             retry_count: 0,             boot_successful: 0
slot: 1,             priority: 15,             suffix: _b,             retry_count: 7,             boot_successful: 1

$ ls -la /opt/ota_package/
total 12664
drwxr-xr-x    2 root     root          4096 Sep 20 10:47 .
drwxr-xr-x    4 root     root          4096 Mar  9  2018 ..
-rw-r--r--    1 root     root      13008176 Mar  9  2018 bl_update_payload
-rw-r--r--    1 root     root           796 Sep 20 10:47 entry_table

$ cat /opt/ota_package/entry_table 
bpmp-fw,5920,533904,2,,1,1
sce-fw,539824,125568,2,,1,1
cpu-bootloader,665392,287232,2,,1,1
mb2,952624,117568,2,,0,1
spe-fw,1070192,95984,2,,0,1
secure-os,1166176,366400,2,,1,1
adsp-fw,1532576,108208,2,,1,1
BMP,1640784,70988,2,,1,1
mb1,1711772,99664,2,,2,1
mts-bootpack,1811436,2082144,2,,1,1
mts-preboot,3893580,63104,2,,0,1
sc7,3956684,23632,2,,1,1
bpmp-fw-dtb,6400796,605120,2,3310-D01-1000--1-0-jetson-tx2-mmcblk0p1,1,1
bootloader-dtb,9098268,371808,2,3310-D01-1000--1-0-jetson-tx2-mmcblk0p1,1,1
VER,9842292,102,2,3310-D01-1000--1-0-jetson-tx2-mmcblk0p1,0,1
BCT,9856832,3584,2,3310-D01-1000--1-0-jetson-tx2-mmcblk0p1,2,1
MB1_BCT,10064384,50096,2,3310-D01-1000--1-0-jetson-tx2-mmcblk0p1,0,1
kernel,10164576,612752,2,,1,1
kernel-dtb,12264560,371808,2,3310-D01-1000--1-0-jetson-tx2-mmcblk0p1,1,1

$ nv_update_engine --verify
Nvidia A/B-Redundancy Update tool Version 1.2
verifying update
The current slot 1 is marked as boot successful
SM: S22
Previous update on slot 0 failed
```
Any chance someone could try this?

Thanks

madisongh, 2021-07-01:
I tested this with both the demo distro and with stock L4T R32.4.3 and yes, the verification fails on reboot. Looks like `nv_update_engine` (actually, `nv_bootloader_payload_updater`, which is called on by `nv_update_engine`) has a bug in it in R32.4.3 where it doesn't account for the fact that the partition offsets in the partition table at the end of `/dev/mmcblk0boot1` are based on treating the two eMMC boot partitions as a single device. So when it seeks to the start of the `VER` or `VER_b` partition, which is located in the `boot1` device, it gets an EINVAL error from the kernel (because it's seeking on the `boot0` device to an offset greater than the size of that device), which it doesn't check before reading/writing. On the verification pass, it then reads whatever garbage happens to be at its current position (because the seek failed) so the comparision fails.

This appears to be fixed in R32.4.4, and it doesn't really affect stock L4T R32.4.3 because they do bootloader updates only to update to R32.4.4 - so after it reboots, it's running the fixed version of the tools and everything checks out OK.

We don't see this with the Mender-enabled demo distro because we disable the `nv_update_verifier` service - each rootfs is tightly coupled to the bootloader contents associated with the boot slot (since the bootloader contents has stuff like firmware, device tree, and maybe the kernel as well, depending on your setup), so with a full dual-rootfs A/B setup, you don't really want the verifier (which copies the current bootslot contents back to the other slot).

lfdmn, 2021-07-02:
Thanks for the explanation!

The dunfell-l4t-r32.5.0 branch works for me for TX2. Checking with Xavier now.

I found nv_update_engine issues in gatesgarth branch. nv_update_engine -i fails saying eks partition isn't found. Looking at the system traces, it tries to mount eks_b partition which is missing from the stock flash layout.

I added it, the install went through successfully but the next boot hangs. I have not stored encryption keys myself. Out of curiousity, has there been a need for a _b partition for the keys?

madisongh, 2021-07-02:
> I found nv_update_engine issues in gatesgarth branch. nv_update_engine -i fails saying eks partition isn't found.

Yeah, I saw that, too, and pushed 9c0bf75a3a1827215ead20503b1e372137f48e3d to fix it. 

> Out of curiousity, has there been a need for a _b partition for the keys?

It provides a way to add/update/rotate keys to go with an updated TOS, if you need to.

closed 2021-08-09 by lfdmn