OE4T/meta-tegra #746: Jetson NX SD Card Creation

bmagistro, 2021-07-02:
I feel like I'm probably missing a step somewhere but am unable to boot using the `tegra-demo-distro` or a custom image built off of `meta-tegra`.  I am not using mender in either build.  I am building the SD card based on instructions here (https://github.com/OE4T/meta-tegra/wiki/Flashing-the-Jetson-Dev-Kit) with the short version of extract tegraflash and use dosdcard.sh.  I'm new to the nvidia stuff but have done some work with oe before, can share additional info on the build if helpful.

The jetson powers on and starts the boot process but logs the following output and drops to limited shell:
```
Mounting /dev/mmcblk0p1
EXT4-fs (mmcblk0p1): couldn't mount as ext3 due to feature incompatibilities
EXT4-fs (mmcblk0p1): couldn't mount as ext2 due to feature incompatibilities
EXT4-fs error (device mmcblk0p1): ext4_iget:4591: inode #8 comm mount: checksum invalid
EXT4-fs (mmcblk0p1): no journal found
mount: mounting /dev/mmcblk0p1 on /mnt failed: Invalid argument
```

madisongh, 2021-07-02:
So the filesystem for the rootfs is corrupted. After you run `dosdcard.sh`, I'd suggest running `e2fsck -f /dev/sda1` (or whatever the right device name is for partition 1 on the SDcard) to verify it before you eject it from the the writer on your development host.

If it verifies OK on the host, but the kernel on the NX still can't read it, then maybe there's some kind of compatibility issue with the SDcard and the controller on the NX.  What kind of card is it?  You might try swapping in a different one.

bmagistro, 2021-07-02:
It looks like the fs is corrupted when validating on the development/flash host.  I am using a SanDisk Extreme 64GB card but using the stock 16GB layout for now.  I wanted to get to a bootable image before I made too many changes.  Not sure if I have any other types of cards available to test with.

```
$ sudo e2fsck -f /dev/sdb1
e2fsck 1.44.1 (24-Mar-2018)
Superblock has an invalid journal (inode 8).
Clear<y>? no
e2fsck: The journal superblock is corrupt while checking journal for /dev/sdb1
e2fsck: Cannot proceed with file system check

/dev/sdb1: ********** WARNING: Filesystem still has errors **********
```

Based on that checked the `core-image-sato-dev.ext4 ` contained in the `tegraflash.tar.gz` and that looks clean.
```
$ e2fsck core-image-sato-dev.ext4 
e2fsck 1.44.1 (24-Mar-2018)
core-image-sato-dev.ext4: clean, 49581/3670016 files, 1352959/3670016 blocks
```

The below are config and layer info (can provide commit hashes but should be close to current dunfell head).  For meta-tegra it is dunfell-l4t-r32.4.3. 

Layer Info:
```
layer                 path                                      priority
==========================================================================
meta                  /home/user1/jetson/poky/meta           5
meta-poky             /home/user1/jetson/poky/meta-poky      5
meta-oe               /home/user1/jetson/poky/../meta-openembedded/meta-oe  6
meta-python           /home/user1/jetson/poky/../meta-openembedded/meta-python  7
meta-filesystems      /home/user1/jetson/poky/../meta-openembedded/meta-filesystems  6
meta-networking       /home/user1/jetson/poky/../meta-openembedded/meta-networking  5
meta-virtualization   /home/user1/jetson/poky/../meta-virtualization  8
meta-tegra            /home/user1/jetson/poky/../meta-tegra  5
contrib               /home/user1/jetson/poky/../meta-tegra/contrib  4
```

local.conf, in general this should be at defaults and overrides added at the end of the file.  the wks file was from the ridgerun wiki and is not being used but hasn't been pulled out yet.
```
IMAGE_CLASSES += "image_types_tegra"
WKS_FILE="jetson-sdcard.wks"
IMAGE_FSTYPES = "tegraflash tar.gz wic.gz wic.bmap"
TEGRA_ROOTFS_AND_KERNEL_ON_SDCARD = "1"
ROOTFSPART_SIZE = "15032385536"

PREFERRED_VERSION_python3 = "3.8%"
PREFERRED_VERSION_python3-native = "3.8%"

# older version of gcc for cuda
GCCVERSION = "8.%"

require conf/include/gstreamer-1.14.conf

MACHINE = "jetson-xavier-nx-devkit"
DISTRO_FEATURES_append = " x11 opengl ldconfig virtualization"
EXTRA_IMAGE_FEATURES += " tools-sdk"
IMAGE_INSTALL_append = " cuda-samples cudnn tensorrt gstreamer1.0-plugins-nvvideo4linux2 deepstream-5.0"
IMAGE_INSTALL_append = " nvidia-docker nvidia-container-runtime cudnn-container-csv tensorrt-container-csv libvisionworks-container-csv"
```

closed 2021-07-02 by bmagistro