OE4T/meta-tegra #751: cboot-t19x: lookup_linear_dir: fails to stop lookup due incorrect entry

ricardosalveti, 2021-07-14:
I'm trying to use extlinux.conf support on jatson agx and noticed that the lookup logic in ext4 seems to be retrieving an invalid entry when the file is not available on the specified folder, only causing it to error out after file_blocknum reaches the maximum 1024 size (basically iterating over the same directory until it reaches 1024).

I noticed this issue after enabling trace support for ext2/4 and trying to debug why it wasn't able to read the file I specified. 

The log from cboot:

```
[0014.743] I> ext2_walk:147: inum 20
[0014.747] I> ext2_walk:132: component 'vmlinuz-4.9.201-l4t-r32.5+gc68230c5b46a.sig', done 1
[0014.755] I> ext2_walk:133: inode flags: 0x00080000
[0014.760] I> lookup_linear_dir: entry
[0014.769] I> ext4_read_data_from_extent: entry
[0014.773] I> ext4_read_data_from_extent:270: Extent: depth: 0, entries: 1
[0014.780] I> ext4_read_extent: entry
[0014.783] I> ext4_read_extent:197: Extent: depth: 0, entries: 1
[0014.789] I> ext4_read_extent:203: block_no: 0, next_block_no: 0
[0014.795] I> ext4_read_extent:220: entry:0: start data blk: 31793, addr: 0x7c36000, len: 1
[0014.803] I> ext4_read_extent:221: num bytes to read: 4096, buf ptr: 0xa9b00000
[0014.816] I> ext4_read_extent:240: bytes_read 4096
[0014.816] I> ext4_read_data_from_extent:304: err 0, bytes_read 4096
[0014.821] I> lookup_linear_dir:431: ent 0:0: inode 0x14, reclen 12, namelen 1, name: .
[0014.829] I> lookup_linear_dir:431: ent 0:12: inode 0x13, reclen 12, namelen 2, name: ..
[0014.837] I> lookup_linear_dir:431: ent 0:24: inode 0x15, reclen 52, namelen 42, name: devicetree-4.9.201-l4t-r32.5+gc68230c5b46a
[0014.848] I> lookup_linear_dir:431: ent 0:76: inode 0x16, reclen 56, namelen 45, name: initramfs-4.9.201-l4t-r32.5+gc68230c5b46a.img
[0014.860] I> lookup_linear_dir:431: ent 0:132: inode 0x17, reclen 3952, namelen 39, name: vmlinuz-4.9.201-l4t-r32.5+gc68230c5b46a
[0014.872] I> lookup_linear_dir:431: ent 0:4084: inode 0x0, reclen 12, namelen 0, name: ��\8��ހ�
[0014.880] I> ext4_read_data_from_extent: entry
[0014.885] I> ext4_read_data_from_extent:270: Extent: depth: 0, entries: 1
[0014.891] I> ext4_read_extent: entry
[0014.895] I> ext4_read_extent:197: Extent: depth: 0, entries: 1
[0014.901] I> ext4_read_extent:203: block_no: 0, next_block_no: 0
[0014.906] I> ext4_read_extent:220: entry:0: start data blk: 31793, addr: 0x7c36000, len: 1
[0014.915] I> ext4_read_extent:221: num bytes to read: 4096, buf ptr: 0xa9b00000
[0014.928] I> ext4_read_extent:240: bytes_read 4096
[0014.928] I> ext4_read_data_from_extent:304: err 0, bytes_read 4096
[0014.932] I> lookup_linear_dir:431: ent 1:0: inode 0x14, reclen 12, namelen 1, name: .
[0014.940] I> lookup_linear_dir:431: ent 1:12: inode 0x13, reclen 12, namelen 2, name: ..
[0014.948] I> lookup_linear_dir:431: ent 1:24: inode 0x15, reclen 52, namelen 42, name: devicetree-4.9.201-l4t-r32.5+gc68230c5b46a
[0014.960] I> lookup_linear_dir:431: ent 1:76: inode 0x16, reclen 56, namelen 45, name: initramfs-4.9.201-l4t-r32.5+gc68230c5b46a.img
[0014.972] I> lookup_linear_dir:431: ent 1:132: inode 0x17, reclen 3952, namelen 39, name: vmlinuz-4.9.201-l4t-r32.5+gc68230c5b46a
[0014.983] I> lookup_linear_dir:431: ent 1:4084: inode 0x0, reclen 12, namelen 0, name: ��\8��ހ�
```

This last line, with inode 0x0 and reclen 12 seems to be the pattern for the end of the extent block, and the check to abort the lookup when the file is not found is only checking for reclen 0 at the end, which causes the loop until 1024.

bootloader/partner/common/lib/fs/ext4/ext4.c

```
   file_blocknum = 0;
    for (;;) {
        err = ext4_read_data_from_extent(ext2, dir_inode, buf);
        if (err <= 0) {
            return -1;
        }

        /* walk through the directory entries, looking for the one that matches */
        pos = 0;
        while (pos < E2FS_BLOCK_SIZE(ext2->super_blk)) {

            ent = (struct ext2fs_dir_entry_2 *)&buf[pos];
            LTRACEF("ent %d:%d: inode 0x%x, reclen %d, namelen %d, name: %s\n",
                    file_blocknum, pos, LE32(ent->e2d_inode), LE16(ent->e2d_rec_len), ent->e2d_name_len, ent->e2d_name);

            /* Exit if no more file entries are present */
            if (LE16(ent->e2d_rec_len) == 0) {
                LTRACEF("record len 0\n");
                break;
            }

            /* match */
            if (ent->e2d_name_len == namelen && memcmp(name, ent->e2d_name, ent->e2d_name_len) == 0) {
                *inum = LE32(ent->e2d_inode);
                LTRACEF("match: inode %d\n", *inum);
                file_entry_found = true;
                break;
            }

            pos += ROUNDUP(LE16(ent->e2d_rec_len), 4);
        }

        if (file_entry_found) {
            break;
        }

        file_blocknum++;

        /* sanity check the directory. 4MB should be enough */
        if (file_blocknum > 1024) {
            TRACEF("Invalid file block num\n");
            return -1;
        }
    }
```

It is not clear to me if this is a side effect after converting to sparse images, so wonder if anyone has any idea why this behavior / incorrect check here.

In the end the board is still able to boot since the function eventually returns -1, but it aborts due an incorrect cause from what I understand.

ricardosalveti, 2021-07-14:
This is using the hardknott branch.

ricardosalveti, 2021-07-14:
Looking a bit more, it seems the bad print with the last ent of a block (inode 0x0 reclen 12) is just a red herring, as even a better check there would still cause the loop to go over until it reaches the blocknum limit.

Maybe the issue here is that file_blocknum is being increased on every iteration (until the > 1024 check) but it is not used anywhere else. I'm not so familiar with ext4 yet, but I guess the expected behavior here would be to iterate over all the blocks from the extent, and then return if not found?

ricardosalveti, 2021-07-14:
It seems ext4_read_data_from_extent is able to detect if there is more than one entry already, and read all the content to the buffer given, but lookup_linear_dir still only really checks for one block (file_blocknum not used to calculate the offset).

Maybe the fix would just be to extract the amount of entries from dir_inode extent_header->entries and iterate over that, using the entry index to guide the buf pointer?

madisongh, 2021-07-14:
I think you're close.  If I'm reading the traces above and the ext4 spec correctly, what's missing is iterating through all of the inlined data in the directory inode. It looks like someone just cut-and-pasted the `ext2_dir_lookup()` code and didn't fully complete the ext4 conversion.  It works OK if the directory is fully contained in a single 4K block, so I guess that was good enough. :(

To fix the last bug we found in the cboot ext4 implementation, I used the [ext4 wiki](https://ext4.wiki.kernel.org/index.php/Ext4_Disk_Layout) and the Linux kernel implementation as guides.

ricardosalveti, 2021-07-15:
> I think you're close. If I'm reading the traces above and the ext4 spec correctly, what's missing is iterating through all of the inlined data in the directory inode. It looks like someone just [...]

You're right, it seems it was just a bad copy-and-paste from `ext2_dir_lookup()`, not taking care of all the inlined data as you said.
 
> To fix the last bug we found in the cboot ext4 implementation, I used the [ext4 wiki](https://ext4.wiki.kernel.org/index.php/Ext4_Disk_Layout) and the Linux kernel implementation as guides.

Cool, thanks for the pointer, was quite helpful.

This ended up being a bit more complex than I envisioned as there are several scenarios we have to cover when managing an ext4 filesystem, but I think I got a better understanding now.

The interesting piece here is that `ext4_dir_lookup` only gives a buffer (`buf`) to `lookup_linear_dir` enough to cover one block (`E2FS_BLOCK_SIZE(ext2->super_blk)`), but later it calls `ext4_read_data_from_extent`, which basically inlines all the extent data (covering both interior and leaf nodes), without really checking the buffer size, so it could in theory quite easily go over the original buffer size given, causing an abort / seg fault. This works fine when there is a single block, but would probably crash on cases covering interior nodes.

Looking at the ext4 documentation, it says "the end of the entire directory is of course signified by reaching the end of the file", but the current code doesn't really validate that, and from what it seems it was supposed to only handle up to 4MB directories (part of the `file_blocknum > 1024` check).

I believe the most correct fix here would be to combine `lookup_linear_dir` with `ext4_read_data_from_extent`, but it would of course duplicate quite a bit of logic (so it can navigate the tree while searching for the respective file/directory), so not sure if worthy at this point as it would be more code to maintain as part of this layer.

Given that, I came to a simple diff that is better (good enough) from what I noticed. Let me know if this would be worthy proposing as a PR to this layer (`buf` is basically not used here, and we can improve by not even allocating it):

```
diff --git a/bootloader/partner/common/lib/fs/ext4/ext4.c b/bootloader/partner/common/lib/fs/ext4/ext4.c
index d57fb27..424b0fe 100644
--- a/bootloader/partner/common/lib/fs/ext4/ext4.c
+++ b/bootloader/partner/common/lib/fs/ext4/ext4.c
@@ -407,27 +407,45 @@ static int lookup_linear_dir(ext2_t *ext2, struct ext2fs_dinode *dir_inode, cons
 {
     uint file_blocknum;
     size_t namelen = strlen(name);
+    uint8_t *dir_buf, *entry_buf;
     struct ext2fs_dir_entry_2 *ent;
     uint32_t pos;
     bool file_entry_found = false;
-    int err;
+    ssize_t bytes_read;
+    int err = 0;

     LTRACE_ENTRY;

+    /* up to 4MB for the directory data */
+    dir_buf = calloc(1, 1024 * E2FS_BLOCK_SIZE(ext2->super_blk));
+    if (dir_buf == NULL) {
+        TRACEF("Failed to allocate memory for directory data\n");
+        return -1;
+    }
+
+    bytes_read = ext4_read_data_from_extent(ext2, dir_inode, dir_buf);
+    if (bytes_read <= 0) {
+        err = -1;
+        goto fail;
+    }
+
     file_blocknum = 0;
     for (;;) {
-        err = ext4_read_data_from_extent(ext2, dir_inode, buf);
-        if (err <= 0) {
-            return -1;
+        entry_buf = dir_buf + file_blocknum * E2FS_BLOCK_SIZE(ext2->super_blk);
+        if (entry_buf - dir_buf >= bytes_read) {
+            break;
         }

         /* walk through the directory entries, looking for the one that matches */
         pos = 0;
         while (pos < E2FS_BLOCK_SIZE(ext2->super_blk)) {

-            ent = (struct ext2fs_dir_entry_2 *)&buf[pos];
-            LTRACEF("ent %d:%d: inode 0x%x, reclen %d, namelen %d, name: %s\n",
-                    file_blocknum, pos, LE32(ent->e2d_inode), LE16(ent->e2d_rec_len), ent->e2d_name_len, ent->e2d_name);
+            ent = (struct ext2fs_dir_entry_2 *)&entry_buf[pos];
+            if (LE32(ent->e2d_inode)) {
+                LTRACEF("ent %d:%d: inode 0x%x, reclen %d, namelen %d, name: %s\n",
+                        file_blocknum, pos, LE32(ent->e2d_inode), LE16(ent->e2d_rec_len),
+                        ent->e2d_name_len, ent->e2d_name);
+            }

             /* Exit if no more file entries are present */
             if (LE16(ent->e2d_rec_len) == 0) {
@@ -455,10 +473,14 @@ static int lookup_linear_dir(ext2_t *ext2, struct ext2fs_dinode *dir_inode, cons
         /* sanity check the directory. 4MB should be enough */
         if (file_blocknum > 1024) {
             TRACEF("Invalid file block num\n");
-            return -1;
+            err = -1;
+            goto fail;
         }
     }

+fail:
+    free(dir_buf);
+
     return err;
 }
```

madisongh, 2021-07-15:
I came up with a simpler fix, borrowing the logic from the hashed directory lookup function.  It works fine in my testing:
* disabling the `dir_index` feature on the rootfs
* populating the directories with a bunch of junk files to get the size to exceed a single block
* making sure the extlinux and Image file names end up at the end of their respective directories
* enabling tracing to see it traverse to the second block correctly

ricardosalveti, 2021-07-15:
> I came up with a simpler fix, borrowing the logic from the hashed directory lookup function. It works fine in my testing:
>
> * disabling the `dir_index` feature on the rootfs
> * populating the [...]

Great, this is better indeed, it wasn't clear to me if the number of blocks would be available for the directory inode on linear directories (mostly because of the > 1024 check).

Will give it a try and report back to your PR. Thanks!

PR #754 (2021-07-15): cboot-*: add patch for ext4 linear directory traversal
The existing code wasn't handling multi-block directories
correctly.

Fixes #751
closed 2021-07-15 by madisongh