OE4T/meta-tegra #752: Jetson TX2 NX flash hangs

mothacehe, 2021-07-14:
Hello,

I have produced `master` and `hardknott` images for the `jetson-xavier-nx-devkit-tx2-nx` machine. When flashing those images on a TX2 NX module (P3636) mounted on the Xavier NX devkit (P3518), using `doflash`, the flashing hangs at this step:

```
[   6.7777 ] Sending bootloader and pre-requisite binaries
[   6.7785 ] tegrarcm_v2 --download blob blob.bin
[   6.7791 ] Applet version 01.00.0000
[   6.7966 ] Sending blob
[   6.7967 ] [................................................] 100%
[   7.2629 ] 
[   7.2643 ] tegrarcm_v2 --boot recovery
[   7.2651 ] Applet version 01.00.0000
[   7.2896 ] 
[   8.2918 ] tegrarcm_v2 --isapplet
```

Any idea what's going wrong?

Thanks,

Mathieu

madisongh, 2021-07-14:
It's working OK for me in my builds off both branches.  The `doflash.sh` output continues like this:
```
[   6.6976 ] Sending blob
[   6.6977 ] [................................................] 100%
[   7.1030 ] 
[   7.1046 ] tegrarcm_v2 --boot recovery
[   7.1059 ] Applet version 01.00.0000
[   7.1092 ] 
[   8.1121 ] tegrarcm_v2 --isapplet
[   8.1137 ] USB communication failed.Check if device is in recovery
[   8.1141 ] 
[   8.1159 ] tegradevflash_v2 --iscpubl
[   8.1175 ] Cannot Open USB
[   8.1179 ] 
[   9.1208 ] tegrarcm_v2 --isapplet
[   9.1242 ] 
[   9.1259 ] tegradevflash_v2 --iscpubl
[   9.1274 ] Bootloader version 01.00.0000
[   9.1295 ] Bootloader version 01.00.0000
[   9.1306 ] 
[   9.1306 ] Retrieving storage infomation
[   9.1320 ] tegrarcm_v2 --oem platformdetails storage storage_info.bin
[   9.1334 ] Applet is not running on device. Continue with Bootloader
[   9.1367 ] 
[   9.1382 ] tegradevflash_v2 --oem platformdetails storage storage_info.bin
[   9.1396 ] Bootloader version 01.00.0000
[   9.1417 ] Saved platform info in storage_info.bin
```
and the serial console shows:
```
[0890.110] I> Welcome to MB2(TBoot-BPMP) Recovery(version: 01.00.160913-t186-M-00.00-mobile-2814ffb8)
[0890.120] I> Boot-device: eMMC
[0890.144] I> sdmmc DDR50 mode
[0890.148] I> sdmmc bdev is already initialized
[0890.153] I> pmic: reset reason (nverc)	: 0x0
[0890.161] I> Found 19 partitions in SDMMC_BOOT (instance 3)
[0890.169] I> Found 33 partitions in SDMMC_USER (instance 3)
[0890.178] I> Binary(16) of size 534416 is loaded @ 0xd7800000
[0890.185] I> Binary(17) of size 113296 is loaded @ 0xd79e4560
[0890.212] I> Copy BTCM section
[0890.217] I> Binary(13) of size 221376 is loaded @ 0x96000000
[0890.224] I> Binary(20) of size 191536 is loaded @ 0x8520f400
[0890.232] I> Binary(14) of size 394672 is loaded @ 0x8530f600
[0890.239] I> TOS boot-params @ 0x85000000
[0890.243] I> TOS params prepared
[0890.246] I> Loading EKS ...
[0890.249] I> Binary(15) of size 1040 is loaded @ 0x8590f800
[0890.255] I> EKB detected (length: 0x400) @ 0x8590f800
[0890.260] I> Copied encrypted keys
[0890.264] I> boot profiler @ 0x175844000
[0890.268] I> boot profiler for TOS @ 0x175844000
[0890.272] I> Unhalting SCE
[0890.275] I> Primary Memory Start:80000000 Size:70000000
[0890.280] I> Extended Memory Start:f0110000 Size:856f0000
[0890.287] I> MB2(TBoot-BPMP) Recovery done

NOTICE:  BL31: v1.3(release):5b49e7f80
NOTICE:  BL31: Built : 08:38:54, Feb 19 2021
ipc-unittest-main: 1519: Welcome to IPC unittest!!!
ipc-unittest-main: 1531: waiting forever
ipc-unittest-srv: 329: Init unittest services!!!
hwkey-agent: 40: hwkey-agent is running!!
hwkey-agent: 197: key_mgnt_processing .......
hwkey-agent: 189: Setting EKB key 0 to slot 14
hwkey-agent: 167: Init hweky-agent services!!
luks-srv: 40: luks-srv is running!!
luks-srv: 157: Init luks-srv IPC services!!
platform_bootstrap_epilog: trusty bootstrap complete
[0890.510] I> Welcome to TBoot-CPU Recovery(version: 01.00.160913-t186-M-00.00-mobile-bf9ced17)
[0890.519] I> gpio framework initialized
[... 59 lines cut ...]
[0890.814] I> 39) Base:0x96000000 Size:0x02000000
[0890.818] I> 40) Base:0x85000000 Size:0x01200000
[0890.823] I> 41) Base:0x175800000 Size:0x00500000
[0890.827] I> 42) Base:0x00000000 Size:0x00000000
[0890.831] I> 43) Base:0x00000000 Size:0x00000000
[0890.836] I> Boot-device: eMMC
[0890.858] I> sdmmc DDR50 mode
[0890.862] I> sdmmc bdev is already initialized
[0890.868] I> Found 19 partitions in SDMMC_BOOT (instance 3)
[0890.874] I> Found 33 partitions in SDMMC_USER (instance 3)
[0890.879] I> bl dtb load address = @0x8520f400
[0890.884] I> Recovery boot_type: 0
[0890.889] I> fixed regulator driver initialized
[0890.917] I> register 'maxim' power off handle
[0890.922] I> virtual i2c enabled
[0890.925] I> registered 'maxim,max77620' pmic
[0890.929] I> tegrabl_gpio_driver_register: register 'max77620-gpio' driver
[0890.936] I> Entering 3p server
[0890.939] I>  HIGH SPEED
[0891.431] I> USB configuration success
[0892.085] I> Populate storage info
```
Is your USB cable directly connected to a port on your development host (as opposed to a hub), and are you flashing from a natively-running Linux (as opposed to a VM), with a good-quality USB cable (e.g., the one that comes with the Jetson development kit)?

Have you tried flashing with stock NVIDIA L4T/JetPack  (I don't think so, based on the UART output)?

mothacehe, 2021-07-20:
Hello,

Thanks for trying it! Turns out I can reproduce the exact same issue with the stock JetPack 4.5.1 release. I reported the issue [here](https://forums.developer.nvidia.com/t/jetson-tx2-nx-flash-hangs/183629/). I am still unable to flash the board but maybe we can close this ticket as this is does not appear to be related to this layer.

Thanks,

Mathieu

madisongh, 2021-07-20:
OK, let's close this here. Hopefully the NVIDIA folks can help you figure this out.

closed 2021-07-20 by madisongh