OE4T/meta-tegra #760: Container support broken on master

madisongh, 2021-07-23:
**Describe the bug**
The `nvidia-container-toolkit` program is crashing with a segmentation fault when trying to start a container.

The segfault is happening during teardown of the RPC communication it uses, which appears to be due to the newer `libtirpc` version (1.3.2) in OE-Core master.  Replacing the use of that version with a statically-linked copy of the `libtirpc` pulled from OE-Core dunfell eliminates the segfault, but setup still fails with:
```
docker: Error response from daemon: failed to create shim: OCI runtime create failed: container_linux.go:380: starting container process caused: process_linux.go:545: container init caused: Running hook #0:: error running hook: exit status 1, stdout: , stderr: nvidia-container-cli: initialization error: driver client creation failed: RPC: Remote system error - Cannot allocate memory: unknown.
```

**To Reproduce**
Steps to reproduce the behavior:
1. Use `tegra-demo-distro`, branch `master`
2. Build `demo-image-full` 
3. Load onto target (tested with Xavier NX devkit)
4. Try `docker run --net=host --runtime nvidia --rm --ipc=host --cap-add SYS_PTRACE -e DISPLAY=$DISPLAY -it nvcr.io/nvidia/l4t-base:r32.5.0`

madisongh, 2021-07-25:
This is due libtirpc trying to allocate arrays based on the fd table size, which has gone from thousands to billions in size.  It also appears that libtripc isn't properly handling memory allocation failures in some of its code paths, leading to the segmentation faults.

You can work around the problem by explicitly using `--ulimit nofile=1024:4096` , or some other more reasonable limits on the `docker run` command, but #763 patches the version of the RPC library statically linked into the container tools to cap the array sizes down to 1K, to work around the problem.

ichergui, 2021-07-25:
Hey @madisongh 

I tried `docker` and I got the same issue as you. 

* Here is the logs 
```
root@jetson-tx2-devkit:~# docker run -it --rm --runtime nvidia --network host nvcr.io/nvidia/l4t-ml:r32.5.0-py3
docker: Error response from daemon: failed to create shim: OCI runtime create failed: container_linux.go:380: starting container process caused: process_linux.go:545: container init caused: Running hook #0:: error running hook: exit status 1, stdout: , stderr: nvidia-container-cli: initialization error: driver client creation failed: RPC: Remote system error - Cannot allocate memory: unknown.
root@jetson-tx2-devkit:~# 
```

To Reproduce
Steps to reproduce the behavior:

1. Use tegra-demo-distro, branch master
2. Build demo-image-full
3. Load onto target (tested with Jetson TX2 devkit)
4. Try the following commands 
```
# docker pull nvcr.io/nvidia/l4t-ml:r32.5.0-py3
# docker run -it --rm --runtime nvidia --network host nvcr.io/nvidia/l4t-ml:r32.5.0-py3
```

PR #763 (2021-07-25): external/virtualization-layer: patch libtirpc126 to limit fd table sizes
With recent systemd versions, process limits on the number of open files
have been raised by a couple of orders of magnitude (K to G). The libtirpc
library, which is unfortunately used by libnvidia-container-tools to implement
some inter-process communication, allocates some arrays of tracking structures based
on the open-files limit (e.g., dtablesze) - leading to memory allocation failures
as it tries to allocate multi-GiB arrays off the heap.

Add a modified patch, back-ported from libtirpc 1.3.2, that caps the array si [... cut]
closed 2021-07-25 by madisongh