OE4T/meta-tegra #765: Mount fails - Nvdia jetson xavier NX dev kit

DhanushLingeswaran15, 2021-08-04:
I am using the meta-tegra from dunfell-l4t-r32.5.0 branch. I tired building a sato-image-dev. The image gets build perfectly fine. I am using the dosdcard.sh script to flash the image on to my sdcard. But when my device boots it shows 
"mounting /dev/mmcblk0p1 on /mnt failed: invalid arguments"

what could be the problem?

ichergui, 2021-08-04:
Hey @DhanushLingeswaran15 

What was the original image before you flashed the SDCard ? 
I think you need to use `doflash.sh` script instead

DhanushLingeswaran15, 2021-08-04:
Hi @ichergui 
It's a new device and a fresh SD card. I think the dev kit that I have doesn't have an emmc on board . Just the SD card. So I don't think I will be able to use doflash.sh.
I think doflash.sh pushes image to emmc , please correct me if I am wrong

DhanushLingeswaran15, 2021-08-04:
Just for information , trying to use doflash.sh , say me that the image size is not in the multiples of 4096

ichergui, 2021-08-04:
Could you please share the logs ? 
I try the branch and it works fine to me.
I will double check later and I will let you know 
Please refer to the [link](https://github.com/OE4T/meta-tegra/wiki/Flashing-the-Jetson-Dev-Kit) for device flashing

DhanushLingeswaran15, 2021-08-05:
Here is the log, when i try to run doflash.sh (My device is in recovery mde and connected to pc via micro usb)

dhanush@dhanush-HP-EliteBook-850-G6:~/JetsonDunfell/build/tmp/deploy/images/jetson-xavier-nx-devkit/core-image-sato-dev-jetson-xavier-nx-devkit.tegraflash$ sudo ./doflash.sh
Welcome to Tegra Flash
version 1.0.0
Type ? or help for help and q or quit to exit
Use ! to execute system commands
 
[   0.0028 ] Generating RCM messages
[   0.0035 ] tegrahost_v2 --chip 0x19 0 --magicid MB1B --appendsigheader mb1_t194_prod.bin zerosbk
[   0.0042 ] Header already present for mb1_t194_prod.bin
[   0.0066 ] 
[   0.0071 ] tegrasign_v2 --key None --getmode mode.txt
[   0.0076 ] Assuming zero filled SBK key
[   0.0077 ] 
[   0.0086 ] tegrasign_v2 --key None --file mb1_t194_prod_sigheader.bin --offset 2960 --length 1136 --pubkeyhash pub_key.key
[   0.0092 ] Assuming zero filled SBK key
[   0.0095 ] 
[   0.0099 ] tegrahost_v2 --chip 0x19 0 --updatesigheader mb1_t194_prod_sigheader.bin mb1_t194_prod_sigheader.hash zerosbk
[   0.0128 ] 
[   0.0134 ] tegrabct_v2 --chip 0x19 0 --sfuse tegra194-mb1-soft-fuses-l4t.cfg.pdf sfuse.bin
[   0.0141 ] 
[   0.0150 ] tegrarcm_v2 --listrcm rcm_list.xml --chip 0x19 0 --sfuses sfuse.bin --download rcm mb1_t194_prod_sigheader.bin 0 0
[   0.0156 ] RCM 0 is saved as rcm_0.rcm
[   0.0195 ] RCM 1 is saved as rcm_1.rcm
[   0.0195 ] RCM 2 is saved as rcm_2.rcm
[   0.0195 ] List of rcm files are saved in rcm_list.xml
[   0.0195 ] 
[   0.0196 ] Signing RCM messages
[   0.0215 ] tegrasign_v2 --key None --list rcm_list.xml --pubkeyhash pub_key.key --getmontgomeryvalues montgomery.bin
[   0.0222 ] Assuming zero filled SBK key
[   0.0226 ] 
[   0.0227 ] Copying signature to RCM mesages
[   0.0233 ] tegrarcm_v2 --chip 0x19 0 --updatesig rcm_list_signed.xml
[   0.0242 ] 
[   0.0243 ] Boot Rom communication
[   0.0248 ] tegrarcm_v2 --chip 0x19 0 --rcm rcm_list_signed.xml --skipuid
[   0.0254 ] RCM version 0X190001
[   0.1150 ] Boot Rom communication completed
[   1.1306 ] 
[   2.1360 ] tegrarcm_v2 --isapplet
[   2.1392 ] Applet version 01.00.0000
[   2.3206 ] 
[   2.3245 ] tegrarcm_v2 --ismb2
[   2.5043 ] 
[   2.5079 ] tegrahost_v2 --chip 0x19 --align nvtboot_applet_t194.bin
[   2.5111 ] 
[   2.5141 ] tegrahost_v2 --chip 0x19 0 --magicid PLDT --appendsigheader nvtboot_applet_t194.bin zerosbk
[   2.5170 ] adding BCH for nvtboot_applet_t194.bin
[   2.5237 ] 
[   2.5270 ] tegrasign_v2 --key None --list nvtboot_applet_t194_sigheader.bin_list.xml --pubkeyhash pub_key.key
[   2.5279 ] Assuming zero filled SBK key
[   2.5282 ] 
[   2.5293 ] tegrahost_v2 --chip 0x19 0 --updatesigheader nvtboot_applet_t194_sigheader.bin.encrypt nvtboot_applet_t194_sigheader.bin.hash zerosbk
[   2.5317 ] 
[   2.5326 ] tegrarcm_v2 --download mb2 nvtboot_applet_t194_sigheader.bin.encrypt
[   2.5335 ] Applet version 01.00.0000
[   2.7128 ] Sending mb2
[   2.7131 ] [................................................] 100%
[   2.7261 ] 
[   2.7288 ] tegrarcm_v2 --boot recovery
[   2.7317 ] Applet version 01.00.0000
[   2.9120 ] 
[   3.9179 ] tegrarcm_v2 --isapplet
[   4.1048 ] 
[   4.1085 ] tegrarcm_v2 --ismb2
[   4.1114 ] MB2 Applet version 01.00.0000
[   4.2927 ] 
[   4.2963 ] tegrarcm_v2 --ismb2
[   4.2994 ] MB2 Applet version 01.00.0000
[   4.4810 ] 
[   4.4846 ] Retrieving board information
[   4.4878 ] tegrarcm_v2 --oem platformdetails chip chip_info.bin
[   4.4908 ] MB2 Applet version 01.00.0000
[   4.6539 ] Saved platform info in chip_info.bin
[   4.6583 ] Chip minor revision: 2
[   4.6584 ] Bootrom revision: 0xf
[   4.6584 ] Ram code: 0x0
[   4.6585 ] Chip sku: 0xde
[   4.6585 ] Chip Sample: non es
[   4.6585 ] 
[   4.6592 ] Retrieving EEPROM data
[   4.6594 ] tegrarcm_v2 --oem platformdetails eeprom cvm /home/dhanush/JetsonDunfell/build/tmp/deploy/images/jetson-xavier-nx-devkit/core-image-sato-dev-jetson-xavier-nx-devkit.tegraflash/cvm.bin.L6GP8
[   4.6624 ] MB2 Applet version 01.00.0000
[   4.8286 ] Saved platform info in /home/dhanush/JetsonDunfell/build/tmp/deploy/images/jetson-xavier-nx-devkit/core-image-sato-dev-jetson-xavier-nx-devkit.tegraflash/cvm.bin.L6GP8
[   4.8637 ] 
[   4.8640 ] Rebooting to recovery mode
[   4.8680 ] tegrarcm_v2 --ismb2
[   4.8710 ] MB2 Applet version 01.00.0000
[   5.0340 ] 
[   5.0341 ] Rebooting to recovery mode
[   5.0359 ] tegrarcm_v2 --reboot recovery
[   5.0374 ] MB2 Applet version 01.00.0000
[   5.2362 ] 
size of core-image-sato-dev.ext4 is not mulple of 4096.

lfdmn, 2021-08-05:
@DhanushLingeswaran15 check the value of ```ROOTFSPART_SIZE``` in your machine config and make sure it is dividable by 4096.

DhanushLingeswaran15, 2021-08-05:
Hi @lfdmn 
ROOTFSPART_SIZE ?= "15032385536" 

this is the size that i use.

DhanushLingeswaran15, 2021-08-05:
My bad! i missed this line in the wiki 
"Be sure to use the tar command from a terminal window. Some users have reported issues with incorrect results when extracting files using GUI-based tools."

GUI based extraction seems to mess up the image.ext4 file during extraction. Hence the file size error in ext4 file. 
I used tar command to extract and everything worked fine!

Thanks a lot for your help guys!!!

closed 2021-08-05 by DhanushLingeswaran15