OE4T/meta-tegra #766: u-boot-tegra nano: L4T 32.5.0 /boot/extlinux.conf append is not honoured

FransSorama, 2021-08-04:
After moving from L4T 32.4.3 to 32.5.0 the append section from /boot/extlinux/extlinux.conf is not passed on to the kernel.
In my case I wanted to pass the root partition to the device

I did some analysis on this. In arch/armlib/bootm.c function fdt_chosen() it says:
        str = env_get("bootargs");
        if (str) {
                err = fdt_setprop(fdt, nodeoffset, "bootargs", str,
                                  strlen(str) + 1);
A debug statement after the env_get indicates that the root partition is there.

Adding debug printf's in fdt_setprop showed that "bootargs" is written twice. The first time as expected. This call was invoked from fdt_chosen.
The second invocation seems to come somewhere below the ft_system_setup() call (code in arch/arm/mach-tegra/dt-setup.c).
Quite likely this comes from the call from fdt_copy_env_proplist(blob).
This second invocation overwrites the bootargs written bij fdt_chosen with a copy that does not have the append to the command line from /boot/extlinux/extlinux.conf, so my added root partition is lost from the cmd line.

While searching for a solution I found the following posts that probably describe the same problem.
https://forums.developer.nvidia.com/t/nano-problem-about-modify-extlinux-conf/165372
https://forums.developer.nvidia.com/t/extlinux-conf-kernel-command-line-options-not-appended-when-using-fdt/179010
message 102 of https://forums.developer.nvidia.com/t/jetpack-4-5-production-release-with-l4t-32-5/166475/106

**To Reproduce**
Steps to reproduce the behavior:
1. Build `meta-tegra` commit fd2c51cbe94c156944240607a3d92b9bdc6c609c with `MACHINE` based on jetson-nano-devkit-emmc
2. Build with bitbake argument '...'
I was building our own image based upon the changes in tegra-demo-distro.
We're dunfell based so I used 
commit ca26fd53d5ef04c56c0752e00259f1a42dbb59cd (HEAD -> dunfell-l4t-r32.5.0)
make sure you have a root fs with a /boot/extlinux.conf on it

u-boot-tegra is on hash 4bca3f387ee0f08f6f9b7daa80c6c71c765d7ddd from nov 29, 2020
3. Deploy to hardware with method 'doflash.sh'; connect serial cable
4. notice that system does not boot properly, in the console see that in u-boot there is a printf starting with:
append:
that contains the new bootargs.
When booting the kernel the boot cmdline given does not have the additional args that are added by extlinux.conf

**Additional context**

This is my /boot/extlinux/extlinux.conf

# Generic Distro Configuration file generated by OpenEmbedded
LABEL Sorama
        KERNEL ../Image
        FDT ../tegra210-p3448-0002-p3449-0000-b00.dtb
        INITRD ../initrd
        APPEND ${cbootargs} root=/dev/mmcblk${devnum}p${distro_bootpart} rw rootwait ${bootargs}

Thinking of it it might be that the code in ft_system_setup is writing cbootargs as bootargs? Puzzled!

madisongh, 2021-08-05:
Looks like NVIDIA has a fix for this in the L4T R32.6.1 U-Boot sources.  I've pulled the R32.6 changes into the `patches-v2020.04` branch in the u-boot-tegra repository, if you'd like to give them a try.

nvl1109, 2021-08-09:
That branch fixes the issue. Thank you @madisongh

FransSorama, 2021-08-11:
Thanks @madisongh 

I've tested two things and they both fix the issue.

The first one was just to cherry-pick e9ee05aa8e07b7da360a63a24ed85f1b6a40b8bb from the u-boot-tegra repo onto the u-boot we had.
The second one was to use the e9ee05aa8e07b7da360a63a24ed85f1b6a40b8bb commit and everything below it.
This second solution required me to update my u-boot mender patches, I took the patches from tegra-demo-distro, directory 
tegra-demo-distro/layers/meta-tegra-support/dynamic-layers/meta-mender-tegra/recipes-bsp/u-boot/patches branch wip-dunfell-l4t-r32.6 commit 2471759d6c81eb4aeeb9f4c06553ca9ab7cc05a3

These patches applied flawlessly and also resulted in a u-boot working as expected.

closed 2021-08-11 by FransSorama