OE4T/meta-tegra #767: R32.6.1-t210: USB dropouts

madisongh, 2021-08-05:
**Describe the bug**
Logitech USB wireless keyboard/mouse receiver isn't coming up.  Unplugging and reinserting will sometimes get it to work, at least for a while.  Kernel logs show errors trying to probe the device, at least sometimes, and sometimes it looks like the PHY just powers off the port.

Tested on wip-dunfell-l4t-r32.6.1, jetson-nano-devkit-emmc, demo-image-sato.

Needs investigation.

ichergui, 2021-08-05:
I will give it a try with `jetson xavier agx` and will let you know.
Maybe something related with `DTB`, I'm not sure if Nvidia update it

madisongh, 2021-08-05:
It could be a problem with the peripheral... just had the same thing happen testing a build off another branch on one of my TX2s.

madisongh, 2021-08-09:
This does a appear to be a problem with my keyboard/mouse peripheral, and not a Nano issue.  Closing.

closed 2021-08-09 by madisongh