OE4T/meta-tegra #768: R32.6.1-t210: gstreamer nv3dsink plugin not working properly

madisongh, 2021-08-05:
**Describe the bug**
Testing with wip-dunfell-l4t-r32.6, jetson-nano-devkit-emmc, core-image-sato.

The `sinteltest` gstreamer test program does not display video, showing mostly black with maybe an occasional still frame from the video.  This test does work OK on TX2 and Xavier. 

Using `nvgstplayer-1.0` to play the video does work OK, though.  That player uses the (long-deprecated) `nveglglessink` plugin for output, instead of the `nv3dsink` plugin, hence the suspicion that there's something wrong with that plugin vs. others in the pipeline.

Needs further investigation.

ichergui, 2021-08-08:
Hey @madisongh 

Could you please share the `Gstreamer` pipeline ?

madisongh, 2021-08-09:
If you're building one of the images in the demo distro that includes the [gstreamer-tests](https://github.com/OE4T/tegra-demo-distro/blob/master/layers/meta-tegra-support/recipes-test/tegra-tests/gstreamer-tests_1.0.bb) recipe, you'll find it in the `sinteltest` script that gets installed.

ichergui, 2021-08-09:
I will give it a try

madisongh, 2021-08-09:
I've tried substituting `nveglglessink` for `nv3dsink` in the script's pipeline, and ran into the same issue.  So perhaps there is some issue with the way the pipeline is constructed that's causing the problem.  It's just odd that it works OK on the t186 and t194 Jetsons.

madisongh, 2021-08-18:
I re-ran the the test today, and it's working fine.  I'll close this as unreproducible now, and just put it down to things being somewhat messy while the the update was still a work in progress.

closed 2021-08-18 by madisongh