OE4T/meta-tegra #769: R32.6.1-t210: gpu_burn test fails

madisongh, 2021-08-05:
**Describe the bug**
Testing on wip-dunfell-l4t-r32.6, jetson-nano-emmc, core-image-sato.

The `gpu_burn` test hangs without displaying any progress.  Process must be manually killed to exit the test.

The same test does work OK on TX2 and Xavier.

Needs investigation.

ichergui, 2021-08-08:
Hey @madisongh 
Could you please share the command line that you used here ?

madisongh, 2021-08-08:
```
cd /opt/cudatests/cpu_burn
./gpu_burn
```

ichergui, 2021-08-08:
This is working fine on `Jetson Xavier AGX`

madisongh, 2021-08-08:
Yes, it was only on the Nano I had the problem, as I mentioned.

madisongh, 2021-08-09:
I re-ran this with a fresh build and didn't encounter the hanging problem, so this was probably an issue with the environment I had at the time.  Closing.

closed 2021-08-09 by madisongh