OE4T/meta-tegra #770: Container doesn't work properly on L4T R32.6.1

ichergui, 2021-08-08:
**Describe the bug**

Testing on wip-dunfell-l4t-r32.6, jetson-agx-xavier-devkit, demo-image-full.

Docker container does't work on `L4T R32.6.1`, when running the `docker` container got the following issue

```
root@jetson-agx-xavier-devkit:~# docker run -it --rm --runtime nvidia --network host nvcr.io/nvidia/l4t-ml:r32.6.1-py3
docker: Error response from daemon: OCI runtime create failed: container_linux.go:345: starting container process caused "process_linux.go:430: container init caused \"process_linux.go:413: running prestart hook 0 caused \\\"error running hook: exit status 1, stdout: , stderr: nvidia-container-cli: mount error: file creation failed: /var/lib/docker/overlay2/7386e75626ea397b81f50e58d83867856c546e7ca1e4d4afa26d458be3d243ba/merged/usr/lib/aarch64-linux-gnu/libvulkan.so.1.2.141: file exists\\\\n\\\"\"": unknown.
root@jetson-agx-xavier-devkit:~#
``` 

When using docker container with tag `r32.5.0-py3` on the same L4T release, It works fine

Needs investigation.

**To Reproduce**
Steps to reproduce the behavior:
1. Use `tegra-demo-distro`, branch `wip-dunfell-l4t-r32.6`
2. Build `demo-image-full`
3. Load onto target (tested with `Xavier AGX devkit`)
4. try the following commands 

```
$ sudo docker pull nvcr.io/nvidia/l4t-ml:r32.6.1-py3
$ sudo docker run -it --rm --runtime nvidia --network host nvcr.io/nvidia/l4t-ml:r32.6.1-py3
```

ichergui, 2021-08-08:
I tried the following command and it works fine.

```
$ sudo docker run -it --rm --network host nvcr.io/nvidia/l4t-ml:r32.6.1-py3
```

I think that the docker container with tag `r32.6.1-py3` is not stable.

madisongh, 2021-08-08:
NVIDIA published a new 0.10-jetson version of libnvidia-container.  We need to adapt the patches we have for the older version onto the new version.

madisongh, 2021-08-08:
We'll also need to review the way we manage the CSV files, since one of the features in the new version is to allow for reduced mapping to the underlying OS by treating "base" mappings differently from others.

ichergui, 2021-08-08:
We should be compatible with the old version, I mean `0.9.0` of libnvidia-container, right ?

madisongh, 2021-08-08:
For R32.6 we need to update to the new version, since all of the NVIDIA R32.6-based containers will expect to be using the new libnvidia-container interface.

ichergui, 2021-08-08:
I'm working on it

ichergui, 2021-08-08:
#771 PR is ready for review

ichergui, 2021-08-09:
@madisongh 

Should we upgrade the `docker-ce` as well to be aligned with `L4T R32.6.1` ?

madisongh, 2021-08-09:
> Should we upgrade the docker-ce as well to be aligned with L4T R32.6.1 ?

Let's focus on the NVIDIA-specific packages, not stock Ubuntu/upstream packages.

madisongh, 2021-08-09:
This error:
```
nvidia-container-cli: mount error: file creation failed: /var/lib/docker/overlay2/7386e75626ea397b81f50e58d83867856c546e7ca1e4d4afa26d458be3d243ba/merged/usr/lib/aarch64-linux-gnu/libvulkan.so.1.2.141: file exists
```
is due to that library being installed by the `tegra-libraries` recipe.  I think the solution is to remove it.

ichergui, 2021-08-09:
Ok, 
I tried `docker` yesterday and it works fine when using `Xavier AGX`, I will double check later. Otherwise, I will remove the library.

madisongh, 2021-08-09:
I've pushed 7d2bc945e3735e010fc83734738d388765f10012 to the branch, and now the `l4t-ml` container starts with no errors.

ichergui, 2021-08-09:
Docker is working fine after fixes from PR #771 and commit  [7d2bc94](https://github.com/OE4T/meta-tegra/commit/7d2bc945e3735e010fc83734738d388765f10012) 

Thanks @madisongh 

I tried [l4t-ml](https://ngc.nvidia.com/catalog/containers/nvidia:l4t-ml) container from Nvidia NGC and it works fine.

```
root@jetson-agx-xavier-devkit:~# docker pull nvcr.io/nvidia/l4t-ml:r32.6.1-py3
r32.6.1-py3: Pulling from nvidia/l4t-ml
e47a8a86d66c: Pull complete 
bdce3430dad6: Pull complete 
c26a6b81c746: Pull complete 
a70635e646de: Pull complete 
24cbe60e3161: Pull complete 
c7f64cc97a39: Pull complete 
8777adb92eda: Pull complete 
542a24b3572f: Pull complete 
3d5ade2b8849: Pull complete 
3e865584f789: Pull complete 
4811af6cacf1: Pull complete 
db645757aac7: Pull complete 
0105dc23dc93: Pull complete 
c8aff3baf097: Pull complete 
5f50b3e38480: Pull complete 
b5418d85b074: Pull complete 
d09ffaf0f11a: Pull complete 
f1cf5625978f: Pull complete 
514fd4c3f8c5: Pull complete 
0988d0ccf8a6: Pull complete 
ffce97733d6c: Pull complete 
80d5eedfe895: Pull complete 
d19792019860: Pull complete 
3ee7b31a8890: Pull complete 
8aca18eb38a5: Pull complete 
153000266b1a: Pull complete 
23bf7741f5d8: Pull complete 
d03fa2111a1c: Pull complete 
095c23b57578: Pull complete 
2de9347ccaf2: Pull complete 
0f26c4b0a765: Pull complete 
ee3bc588c8ee: Pull complete 
aa25eb953d98: Pull complete 
97fd80a69cc6: Pull complete 
Digest: sha256:1f4ef02343223cab6bbe5bf85d5575364b7ee0ee23d72c2e3b89416ca3806f7e
Status: Downloaded newer image for nvcr.io/nvidia/l4t-ml:r32.6.1-py3
nvcr.io/nvidia/l4t-ml:r32.6.1-py3
root@jetson-agx-xavier-devkit:~# docker run -it --rm --runtime nvidia --network host nvcr.io/nvidia/l4t-ml:r32.6.1-py3
allow 10 sec for JupyterLab to start @ http://192.168.0.24:8888 (password nvidia)
JupterLab logging location:  /var/log/jupyter.log  (inside the container)
root@jetson-agx-xavier-devkit:/# python3
Python 3.6.9 (default, Jan 26 2021, 15:33:00) 
[GCC 8.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import torch 
>>> import torchvision
>>> torch.__version__
'1.9.0'
>>> torchvision.__version__
'0.10.0a0+300a8a4'
>>> torch.cuda.is_available()
True
>>> torch.version.cuda
'10.2'
>>> 
root@jetson-agx-xavier-devkit:/# 
```
[image]

closed 2021-08-09 by ichergui