OE4T/meta-tegra #781: R32.6.1-t210: vulkaninfo test fails

ichergui, 2021-08-23:
Testing on `dunfell`, `jetson-nano-devkit-emmc`, `demo-image-weston`.

* Here is the logs
```
sh-5.0# vulkaninfo
Xlib:  extension "NV-GLX" missing on display ":0".
Xlib:  extension "NV-GLX" missing on display ":0".
Xlib:  extension "NV-GLX" missing on display ":0".
/usr/src/debug/vulkan-tools/1.1.126.0-r0/git/vulkaninfo/vulkaninfo.h:399: failed with ERROR_INITIALIZATION_FAILED
sh-5.0#
```
Needs investigation.

To Reproduce
Steps to reproduce the behavior:

Use `tegra-demo-distro`, branch `dunfell`
Build `demo-image-weston`
Load onto target (tested with jetson-nano-devkit-emmc)
Try the following command line:

```
sh-5.0# vulkaninfo
```

madisongh, 2021-08-23:
IIRC, the Jetson Vulkan support requires X11, and has never run on Wayland/Weston.

ichergui, 2021-08-28:
Since Vulkan support requires X11, we should not test it with `demo-image-weston` image 
Closing.

closed 2021-08-28 by ichergui