OE4T/meta-tegra #782: R32.6.1-t210: Deep Sleep (SC7) doesn't work properly

ichergui, 2021-08-23:
Testing on `dunfell`, `jetson-nano-devkit-emmc`, `demo-image-weston`.

Deep sleep (SC7) as described in the [Nvidia documentation](https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%20Linux%20Driver%20Package%20Development%20Guide/power_management_nano.html#) doesn't work when using `Jetson Nano eMMC`

```
root@jetson-nano-devkit-emmc:~# systemctl suspend
root@jetson-nano-devkit-emmc:~# [  513.886729] PM: suspend entry 2021-08-23 21:34:47.895088858 UTC
[  513.892829] PM: Syncing filesystems ... done.
[  513.915435] tegradc tegradc.0: blank - powerdown
[  513.957754] extcon-disp-state extcon:disp-state: cable 47 state 0
[  513.957756] Extcon AUX1(HDMI) disable
[  513.978088] tegradc tegradc.0: unblank
[  514.013901] tegradc tegradc.0: nominal-pclk:148500000 parent:148500000 div:1.0 pclk:148500000 147015000~161865000
[  514.013964] tegradc tegradc.0: hdmi: tmds rate:148500K prod-setting:prod_c_hdmi_75m_150m
[  514.014963] tegradc tegradc.0: hdmi: get YCC quant from EDID.
[  514.050612] extcon-disp-state extcon:disp-state: cable 47 state 1
[  514.050614] Extcon AUX1(HDMI) enable
[  514.065238] tegradc tegradc.0: sync windows ret = 247
[  514.120999] tegra_dvfs: suspended
[  514.121003] Freezing user space processes ... (elapsed 0.050 seconds) done.
[  514.178436] Freezing remaining freezable tasks ... (elapsed 0.001 seconds) done.
[  514.193917] usb 1-2.1: usb_suspend_both: status 0
[  514.194131] usb 2-1: usb_suspend_both: status 0
[  514.198785] usb usb2: usb_suspend_both: status 0
[  514.208860] tegra_cec 70015000.tegra_cec: suspended
[  514.213870] tegra-xudc-new 700d0000.xudc: device mode off: 0
[  514.219658] tegra-xusb-padctl 7009f000.xusb_padctl: power down UTMI pad 0
[  514.226503] tegra-xudc-new 700d0000.xudc: entering ELPG
[  514.231082] usb 1-2.3: usb_suspend_both: status 0
[  514.236946] tegra-xudc-new 700d0000.xudc: entering ELPG done
[  514.244123] tegradc tegradc.1: suspend
[  514.244174] tegradc tegradc.0: suspend
[  514.251953] hpd: state 3 (Disabled), hpd 0, pending_hpd_evt 1
[  514.257743] hpd: request suspend
[  514.259033] tegra-xusb-padctl 7009f000.xusb_padctl: power down UTMI pad 1
[  514.265266] extcon-disp-state extcon:disp-state: cable 47 state 0
[  514.265268] Extcon AUX1(HDMI) disable
[  514.277630] hpd: switching from state 3 (Disabled) to state 3 (Disabled)
[  514.287030] usb 1-2: usb_suspend_both: status 0
[  514.291775] usb usb1: usb_suspend_both: status 0
[  514.319249] tegra-xusb 70090000.xusb: entering ELPG
[  514.325215] tegra-pmc: PMC tegra_pmc_utmi_phy_enable_sleepwalk : port 1, speed 3
[  514.332846] tegra-pmc: PMC tegra_pmc_utmi_phy_enable_sleepwalk : port 2, speed 0
[  514.341830] tegra-xusb 70090000.xusb: entering ELPG done
[  514.347316] PM: suspend of devices complete after 153.591 msecs
[  514.355098] host1x 50000000.host1x: suspended
[  514.359702] PM: late suspend of devices complete after 6.425 msecs
[  514.368880] PM: noirq suspend of devices complete after 2.942 msecs
[  514.375227] Disabling non-boot CPUs ...
[  514.411410] IRQ10 no longer affine to CPU1
[  514.415802] CPU1: shutdown
[  514.418613] psci: CPU1 killed (polled 0 ms)
[  514.463386] IRQ11 no longer affine to CPU2
[  514.467910] CPU2: shutdown
[  514.470699] psci: CPU2 killed (polled 0 ms)
[  514.499391] IRQ12 no longer affine to CPU3
[  514.503932] CPU3: shutdown
[  514.506746] psci: CPU3 killed (polled 0 ms)
[  514.515704] tegra-pmc: PMC wake level = 0x40178000010000
[  514.515704] tegra-pmc: PMC wake enable = 0x48178001010001
[  514.515704] Entered SC7
ERROR:   BPMP not powered on

```

Needs investigation.

To Reproduce
Steps to reproduce the behavior:

Use `tegra-demo-distro`, branch `dunfell`
Build `demo-image-weston`
Load onto target (tested with jetson-nano-devkit-emmc)
Try to follow the steps below:

```
root@jetson-nano-devkit-emmc:~# systemctl suspend
```
or 

```
root@jetson-nano-devkit-emmc:~# bash -c "echo mem > /sys/power/state"
```

madisongh, 2021-08-24:
Does it work with the stock L4T Ubuntu image?

ichergui, 2021-08-24:
I will try it and will update the issue