OE4T/meta-tegra #831: Need to change URIs for at least linux-tegra's git pulls in the recipe.

madscientist42, 2021-11-02:
**Describe the bug**
Effectively, at least linux-tegra is broken with the new "security" features (Which includes no more un-auth git pulls from GitHub...)

https://github.blog/2021-09-01-improving-git-protocol-security-github/

**To Reproduce**

Just build with the OE4T meta-tegra involving one of the supported targets or a derived one.  They're "browning out" right now (So...none of our builds are working...) so all branches are broken with this for anything using a, "ssh://" URI and will be permanently out of order without changes by March 15, 2022.  So, we're going to need to fix this...

madisongh, 2021-11-02:
None of the recipes in the currently-maintained branches in meta-tegra use bare git, as far as I can see; they all specify `protocol=https`  for the Github SRC_URIs. Older branches probably do have some bare git SRC_URIs that need updating.  I'd be happy to take patches to fix those older branches.

The ssh-related changes mentioned in the blog post shouldn't affect the layer at all, since none of the recipes use ssh protocol with Github URLs.

madscientist42, 2021-11-02:
We've got Warrior branch fails, Matt.  So...while we're working on getting off of Warrior onto Gatesgarth and later...it's a bit of a bind.  Good to know that it got fixed later on.  Knowing you'll accept patches/PR's works- we'll get on it for things.  Thanks!

madscientist42, 2021-11-02:
Looks like Richard Purdie's proposing a working "fix" to Bitbake to be dropped in the earlier versions of Yocto- that being said, it rather needs to be fixed, so we're still on a plan to do what you're suggesting...it's just going to not be as pressing for us to do it.

madisongh, 2021-11-02:
Yep, I've seen the conversation on the mailing list.  You might find git's configurable rewrites a useful workaround as well (e.g., `git config --global url.https://github.com/.insteadOf git://github.com/`)

madisongh, 2021-11-10:
The bitbake patch to automatically set `protocol=https` for `github.com` SRC_URIs has been back-ported to OE-Core releases going back to thud, which hopefully will provide cover for folks stuck on older branches when Github throws the big switch for good.

I'll close this now.   PRs for SRC_URI fixups for older branches are still welcome.

closed 2021-11-10 by madisongh