OE4T/meta-tegra #837: Deepstream 6.0 support

Smoolak, 2021-11-08:
Hi,

Is Deepstream 6.0 planned? It would be a nice addition to this project.

dwalkes, 2021-11-08:
I think PRs are welcome for this.  Our current implementation uses in docker so we aren't planning to put in the base OS in the near term.  Perhaps others are.  This could be a topic for the [monthly meeting](https://github.com/OE4T/meta-tegra/discussions/515) this week.

madisongh, 2021-11-08:
I missed the fact that version 6.0 was actually released.  I've taken a quick look at it, and adapting the existing recipe for the new version shouldn't be too bad, except there are some additional runtime dependencies on NVIDIA SDKs that we don't have recipes for yet: Triton and Rivermax.  If the plugins and libraries that have those dependencies can be made optional via PACKAGECONFIG and excluded by default, that should be OK (assuming they aren't  necessary for DS to work).

There are some libraries that are linked against OpenSSL 1.1.1, while OE-Core master has updated to OpenSSL 3.0, so something would need to be done to address that. That shouldn't be an issue in dunfell or honister.

It looks like the Python bindings are available for building [from source](https://github.com/NVIDIA-AI-IOT/deepstream_python_apps) now (including for DS-5.1, it looks like).

madisongh, 2021-11-11:
With thanks to @ichergui , we now have ds-6.0 recipes in honister and dunfell. I'm not sure what to do about the openssl issue in master, but I'll open a separate issue to track that.

closed 2021-11-11 by madisongh