OE4T/meta-tegra #859: CONFIG_PSTORE_RTRACE is invalid kernel config

mtayler, 2021-11-18:
**Describe the bug**
Trying to enable Nvidia's [Register Trace](https://docs.nvidia.com/jetson/l4t/index.html#page/Tegra%20Linux%20Driver%20Package%20Development%20Guide/debugging_rtrace.html#wwpID0EKHA) utility on Jetson Nano in Yocto image.

When building `CONFIG_PSTORE_RTRACE` config is ignored and moved to `invalid.cfg` and `mismatch.cfg`

**To Reproduce**
Steps to reproduce the behavior:
1. Build `meta-tegra` branch 'dunfell' with `MACHINE` based on 'jetson-nano-devkit-emmc'
2. Build with bitbake argument 'linux-tegra'
3. N/A
4. See error `CONFIG_PSTORE_RTRACE` is moved to `invalid.cfg` and `mismatch.cfg`

madisongh, 2021-11-19:
It's considered invalid because there is no such kernel configuration option; you can check out the kernel sources to see that for yourself.

That looks like old documentation.  The feature was present back in the 4.4-based kernels in R28.x and earlier, but looks like it was removed when NVIDIA switched to the current 4.9-based kernel.

mtayler, 2021-11-19:
Thanks @madisongh I was still seeing the option in some of Nvidia's kernel module code. We are using 32.6.1 so it looks like that functionality is no longer present.

closed 2021-11-19 by mtayler