OE4T/meta-tegra #869: kernel-bup-payload fails for xavier nx

reyapo, 2021-12-17:
**Describe the bug**
I'm trying to update our jetson-xavier-nx-devkit-emmc image from `dunfell-r32.5.0` to `dunfell` (r32.6.0) but am getting an error in the `kernel-bup-payload.bb` `do_deploy` step:
```
| creating xusb payload
|
| File /yocto/yocto-configurations/nvidia/jetson-xavier-nx/tmp/work/jetson_xavier_nx_devkit_emmc-poky-linux/kernel-bup-payload/1.0-r0/bup-payload/signed/xusb_sil_rel_fw_sigheader.encrypt does not exist
| PARTITION INFO   :                    signed/xusb_sil_rel_fw_sigheader.encrypt xusb-fw 3261 0 common
| Traceback (most recent call last):
|   File "/yocto/yocto-configurations/nvidia/jetson-xavier-nx/tmp/work/jetson_xavier_nx_devkit_emmc-poky-linux/kernel-bup-payload/1.0-r0/bup-payload/BUP_generator.py", line 659, in <module>
|     main(param)
|   File "/yocto/yocto-configurations/nvidia/jetson-xavier-nx/tmp/work/jetson_xavier_nx_devkit_emmc-poky-linux/kernel-bup-payload/1.0-r0/bup-payload/BUP_generator.py", line 652, in main
|     generate_BUP(arg)
|   File "/yocto/yocto-configurations/nvidia/jetson-xavier-nx/tmp/work/jetson_xavier_nx_devkit_emmc-poky-linux/kernel-bup-payload/1.0-r0/bup-payload/BUP_generator.py", line 175, in generate_BUP
|     payload_obj.fill_image(blob)
|   File "/yocto/yocto-configurations/nvidia/jetson-xavier-nx/tmp/work/jetson_xavier_nx_devkit_emmc-poky-linux/kernel-bup-payload/1.0-r0/bup-payload/BUP_generator.py", line 293, in fill_image
|     binary_handle = open(entry_info[0], 'rb')
| TypeError: expected str, bytes or os.PathLike object, not NoneType
|
| FAILURE: xusb_only_payload not created
```
It seems related to [this issue](https://github.com/OE4T/meta-tegra/issues/336#issuecomment-647841575), however for me the signing steps completes without an error:
```
...
| [   3.8830 ] Copying signed file in /yocto/yocto-configurations/nvidia/jetson-xavier-nx/tmp/work/jetson_xavier_nx_devkit_emmc-poky-linux/kernel-bup-payload/1.0-r0/bup-payload/signed
| [   4.0526 ] tegraparser_v2 --pt flash.xml.bin --generateflashindex /yocto/yocto-configurations/nvidia/jetson-xavier-nx/tmp/work/jetson_xavier_nx_devkit_emmc-poky-linux/kernel-bup-payload/1.0-r0/bup-payload/signed/flash.xml.tmp flash.idx
| [   4.3445 ]
|
| signing images succeeded
...
```

I can only find references to `xusb_sil_rel_fw` in recipes, could it be that an extra signed file was added? Or am I simply missing a migration step.

**To Reproduce**
Steps to reproduce the behavior:
1. Build `meta-tegra` branch 'dunfell' with `jetson-xavier-nx-devkit-emmc` based on *core-image-x11*
2. Build with bitbake argument 'core-image-x11'

**Aditional context**
- Completely removed tmp, sstace-cache, cache and deploy dirs

Please let me know if additional info is needed or I need to check something.

Thanks in advance!

Rens

madisongh, 2021-12-17:
> I now see this in the wiki, but not that I should do anything because of it. Is there a step needed for the change?

If you're using a custom flash layout XML that you created with the older BSP version, you need to update it as mentioned in that section of the wiki page.  Based on the error you're seeing, that looks like the problem.

If you don't think you're using a custom flash layout, maybe it's being introduced by another layer you're including in your builds?

reyapo, 2021-12-17:
Hi,

Thanks for you quick reply! And you hit the mark :) I forgot to update the `meta-mender-community/meta-mender-tegra` layer. Doing so fixes it.

Thanks a lot!

closed 2021-12-17 by reyapo

dwalkes, 2024-06-18:
@paroque28 I think you need to follow the instructions at https://github.com/OE4T/meta-tegra/wiki/L4T-R32.6.1-Notes#flash-layout-file-changes-for-t194-xavier-platforms linked above regarding any custom flash layouts in your layer:

> The flash layout XML files for the Xavier platforms in the L4T BSP have been modified to include new attributes (but no location/size changes). In particular, the xusb-fw partition is now marked oem_signed="true", as cboot now performs signature validation on the USB controller firmware.

See example at https://github.com/mendersoftware/meta-mender-community/blame/kirkstone/meta-mender-tegra/recipes-bsp/tegra-binaries/mender-custom-flash-layout/jetson-xavier-nx-devkit-emmc/flash_mender.xml#L217 - if your layer overrides this layout you need to make the same changes in your layer.

paroque28, 2024-06-19:
@dwalkes I am using this same xml, same  example that you mentioned without modifications, just that I am using the dunfell instead of kirkstone branch (it also has the oem_signed="true").
Unfortunately, no luck.
Thanks for the suggestion

paroque28, 2024-06-19:
@dwalkes I didn try tegra-demo-distro but I checked out the meta-mender commits tegra-demo-distro is using:

[meta-mender @ 35886d9](https://github.com/mendersoftware/meta-mender/tree/35886d930a16ddf10a43e81a5f80d29bd6b33d16)
[meta-mender-community @ dd4621c](https://github.com/mendersoftware/meta-mender-community/tree/dd4621ca3304cf536105416baff7be529b9f543a)

That solved my issue.