OE4T/meta-tegra #879: cboot 32.6.1 loads incorrect boot slots for kernel and kernel-dtb ?

acostach, 2022-01-10:
Hi @madisongh , I've previously tried this with L4T 32.5.1 and tegra boot tools and switching the boot with tegra-boot-control -a 1 worked fine, the second partitions (_b) were used for loading the kernel and dtb.

However, I'm now trying with tegra-boot-tools 2.5.2 and it seems that cboot still loads kernel and dtb from slots 0. I've also tried with tegra-boot-tools 2.5.1 and see the same thing.

My usecase is that I'm starting from L4T 32.5.1, update mtd0 with the new 32.6.1 boot blob (which I construct with dd from signed binaries and boots fine), write the kernel and dtb to the kernel_b and kernel-dtb_b and then do:
tegra-boot-control -e # ensure redundancy is enabled after writing mtd0
tegra-boot-control -a 1 # switch from slot 0 to slot 1
tegra-boot-control -s # ensure slot 1 is set

I then reboot the device and see that slots kernel and kernel-dtb are loaded. I can confirm this by the kernel cmdline, which should've switched to resin-rootB. After a few minutes, the device starts to reboot and eventually (after 3 attempts that I configured in tegra-boot-tools, was originally 7), it switches to loading the new slots _b.

```
  [0001.017] I> Welcome to Cboot
��WARNING: no registered clock for FMON_NAFLL_CLUSTER1 (id 281)
fmon_post initialized
��[0001.017] I> Cboot Version: 32.06.01-poky-t194-59698135
[0001.020] I> CPU-BL Params @ 0xf2820000
[0001.024] I>  0) Base:0x00000000 Size:0x00000000
[0001.028] I>  1) Base:0xf1100000 Size:0x00100000
[0001.033] I>  2) Base:0xf2000000 Size:0x00200000
��clk_set_parent failed for clk i2c2, parent pll_aon (-22)
clk_set_parent failed for clk i2c8, parent pll_aon (-22)
clk_dt_late initialized
machine_check initialized
pm_post initialized
dbells initialized
avfs_clk_platform_post initialized
dmce initialized
cvc initialized
ccplex_avfs_hw_init: nafll_cluster0: not monitored
ccplex_avfs_hw_init: nafll_cluster2: not monitored
ccplex_avfs_hw_init: nafll_cluster3: not monitored
avfs_clk_mach_post initialized
regulator_post initialized
rm initialized
sc7_diag initialized
thermal_test initialized
serial_late initialized
clk_post initialized
clk_dt_post initialized
mc_reg initialized
pg_post initialized
dyn_modules initialized
sku_debugfs initialized
speedo_debugfs initialized
adc_debugfs initialized
Failed to register PTO counter for id 281
Failed to register PTO counter for id 281
Failed to register PTO counter for id 281
Failed to register PTO counter for id 281
clk_debugfs initialized
emc_debugfs initialized
dvs_debugfs initialized
fmon_debugfs_init_one: no clock debugfs node to attach FMON_NAFLL_CLUSTER1
fmon_debugfs initialized
vmon_debugfs initialized
pg_debugfs initialized
profile_fs initialized
debugfs_cons initialized
mail_fs initialized
profile initialized
cvc_debugfs initialized
dmce_debugfs initialized
ec_debugfs initialized
rm_rail_debugfs_init: /rm/vdd_cpu: failed
rm_rail_debugfs_init: /rm/vdd_cpu: failed
rm_debugfs initialized
soctherm_debug initialized
gr_reader initialized
mods initialized
[... 188 lines cut ...]
[0009.797] I> T19x: Authenticate kernel (bin_type: 37), max size 0x5000000
[0010.407] I> Encryption fuse is not ON
[0010.422] I> Validate kernel-dtb ...
[0010.422] I> T19x: Authenticate kernel-dtb (bin_type: 38), max size 0x400000
[0010.428] I> Encryption fuse is not ON
[0010.428] I> Checking boot.img header magic ... [0010.429] I> [OK]
[0010.429] I> Kernel hdr @0xa42e0000
[0010.429] I> Kernel dtb @0x90000000
[0010.429] I> decompressor handler not found
[0010.430] I> Copying kernel image (40108040 bytes) from 0xa42e0800 to 0x80080000 ... [0010.441] I> Done
[0010.442] I> Move ramdisk (len: 0) from 0xa6921000 to 0x91000000
[0010.443] I> Updated bpmp info to DTB
[0010.447] I> Ramdisk: Base: 0x91000000; Size: 0x0
[0010.451] I> Updated initrd info to DTB
[0010.454] I> Active rootfs suffix: 
[0010.458] E> tegrabl_linuxboot_add_disp_param, du 0 failed to get display params
[0010.465] E> tegrabl_linuxboot_add_disp_param, du 0 failed to get display params
[0010.472] E> tegrabl_linuxboot_add_disp_param, du 0 failed to get display params
[0010.479] I> Active slot suffix: _b
[0010.483] I> add_boot_slot_suffix: slot_suffix = _b
[0010.488] I> Linux Cmdline: console=ttyTCU0,115200 root=LABEL=resin-rootA ro rootwait rootfstype=ext4 console=ttyTCU0,115200 console=tty0 fb  <-- should have been resin-rootB, which is contained by kernel-dtb-b
[0010.526] I> Updated bootarg info to DTB
```

I see this log here: '[0004.311] W> tegrabl_get_kernel_bootctrl: use default dummy boot control data' which I suspect might be causing this. How can I prevent it, maybe I can patch cboot to get around this? Thank you!

madisongh, 2022-01-10:
Hmm.  Looks like I forgot to carry forward the patch to the SD/USB boot code to enable A/B slot selection.

acostach, 2022-01-10:
Ok, so that would be only for SD I guess, I haven't tried the eMMC variant yet but it's probably best if l try with the patch applied.

madisongh, 2022-01-10:
I've pushed updates to the master and honister branches to fix this.  I'll follow with an update to the dunfell branch after running a couple more tests.

madisongh, 2022-01-10:
OK, the missing patches are restored for the cboot-t19x recipe in all of the R32.6.1 branches now.

acostach, 2022-01-14:
Thank you @madisongh and sorry for my late reply. I've tested the new patches on the honister branch 2ec80cc94bcd7428c80f505467be2f50ec3d64bd and it seems that I'm still running into this problem. The slot suffix can't be read by the code added in the new patches for some reason, but cboot is  aware of the selected slot:

```
[0004.276] I> fs_detect:173: Unsupported or no filesystem present
[0004.276] I> Loading kernel-bootctrl from partition
[0004.276] I> Loading partition kernel-bootctrl at 0xa42e0000 from device(0x6)
[0004.305] W> tegrabl_get_kernel_bootctrl: magic number(0x00000000) is invalid
[0004.306] W> tegrabl_get_kernel_bootctrl: use default dummy boot control data
[0004.306] I> Active slot suffix: _b <---
[0004.306] W> Could not obtain bootslot suffix[0004.307] I> Loading kernel ... <--- Could not obtain bootslot suffix
[0009.750] I> Loading kernel-dtb ...
```

After setting the new active slot with tegra-boot-control I reboot the device and I don't touch the mtd0 partition in any way. Am I doing something wrong?

Full log:

```
parent failed for clk i2c2, parent pll_aon (-22)
clk_set_parent failed for clk i2c8, parent pll_aon (-22)
clk_dt_late initialized
machine_check initialized
pm_post initialized
dbells initialized
avfs_clk_platform_post initialized
dmce initialized
cvc initialized
ccplex_avfs_hw_init: nafll_cluster0: not monitored
ccplex_avfs_hw_init: nafll_cluster2: not monitored
ccplex_avfs_hw_init: nafll_cluster3: not monitored
avfs_clk_mach_post initialized
regulator_post initialized
rm initialized
sc7_diag initialized
thermal_test initialized
serial_late initialized
clk_post initialized
clk_dt_post initialized
mc_reg initialized
pg_post initialized
dyn_modules initialized
sku_debugfs initialized
speedo_debugfs initialized
adc_debugfs initialized
Failed to register PTO counter for id 281
Failed to register PTO counter for id 281
Failed to register PTO counter for id 281
Failed to register PTO counter for id 281
clk_debugfs initialized
emc_debugfs initialized
dvs_debugfs initialized
fmon_debugfs_init_one: no clock debugfs node to attach FMON_NAFLL_CLUSTER1
fmon_debugfs initialized
vmon_debugfs initialized
pg_debugfs initialized
profile_fs initialized
debugfs_cons initialized
mail_fs initialized
profile initialized
cvc_debugfs initialized
dmce_debugfs initialized
ec_debugfs initialized
rm_rail_debugfs_init: /rm/vdd_cpu: failed
rm_rail_debugfs_init: /rm/vdd_cpu: failed
rm_debugfs initialized
soctherm_debug initialized
gr_reader initialized
mods initialized
dt_fs initialized
debugfs_mrq initialized
[... 249 lines cut ...]
[0010.707] I> - update property: interrupt-affinity
[0010.713] W> "misc-data" doesn't exist, creating
[0010.717] I> Boot-device: QSPI
[0010.719] W> Unknown boot device
[0010.722] I> Add storage-qspi to plugin-manager/misc-data
[0010.728] W> Unknown storage device
[0010.731] I> Add serial number:1423220028426 as DT property
[0010.737] I> Plugin-manager override starting
[0010.741] I> node /plugin-manager/fragment-pcie-c5-rp matches
[0010.747] I> node /plugin-manager/fragement-tegra-wdt-en matches
[0010.754] I> node /plugin-manager/fragement-tegra-sdhci-emmc-dis matches
[0010.761] I> Disable plugin-manager status in FDT
[0010.763] I> Plugin-manager override finished successfully
[0010.769] I> tegrabl_load_kernel_and_dtb: Done
[0010.773] I> SMD partition is updated.
[0010.776] I> QSPI: erasing sectors from 177 - 177
[0010.781] I> QSPI: recoverying tail blocks from 22664 - 22784
[0011.480] I> QSPI: erasing sectors from 176 - 176
[0011.481] I> QSPI: recoverying tail blocks from 22536 - 22656
[0012.154] E> tegrabl_display_clear: display is not initialized
[0012.155] W> Boot logo display failed...
[0012.155] I> Kernel EP: 0x80080000, DTB: 0x90000000
```

Thank you

madisongh, 2022-01-14:
Oops, it looks like I got the logic reversed when checking the return code from the function that gets the bootslot suffix. I'll update the patch to fix it.

acostach, 2022-01-14:
Thank you! I'll test now and let you know

acostach, 2022-01-14:
Thanks again @madisongh , with the latest patches it worked and rolled back to slot _a.

acostach, 2022-01-27:
Hi @madisongh , for the Xaviers (AGX, NX SD/eMMC) the process went fine, the system booted from the previous slot after X unsuccessful attempts. However, on the TX2 I'm seeing a different problem: booting forward from 32.5.1 to 32.6.1 there is no issue, second slot is used and all is loaded fine. However, if there are some unsuccessful attempts and slot 0 is picked for booting, I'm getting:
```
[0002.006] C> MTS error (4) : Unknown or unhandled failure
[0002.011] C> cpu way point1 failed
[0002.014] C> ERROR: Highest Layer Module = 0x32, Lowest Layer Module = 0x32,
Aux Info = 0x2, Reason = 0x6
```

I suspect this happens because the new contents of mmcblk0boot0 (which now contains 32.6.1 binaries) don't like the 32.5.1 binaries on the old slot 0 in the raw mmcblk0pX partitions, and this so far happens only on the TX2 .

If this is the case, do the tools you wrote in tegra-boot-tools accept creating a bup and updating /dev/mmcblk0boot0  slot contents *only*, from a custom flash.xml and signed binaries that were already generated at build-time? Or maybe from a any valid boot0.img that would be stored on mmcblk0boot0?

Thanks you!

madisongh, 2022-01-27:
I think it's because the USB firmware is signed in 32.6.1 where it wasn't in 32.5.1 (module 0x32 is related to USB PHY, from what I see in the cboot sources), although I could be wrong about the specifics.

An update to a new version of the BSP that has changes like these needs careful handling. You perhaps could use `tegra-bootloader-update --slot_suffix _a` (or `_b` - whichever slot is not the current one) after successfully booting with the update to write the update to the other slot using the `bl_only.bup_payload` to just write the boot firmware and not the kernel... I'm just not 100% sure the 32.5.1-based kernel will be compatible with the 32.6.1 boot firmware.  You could test it to see if it's sufficiently compatible for your use case. Otherwise, it might be safer to just trigger a second full update (boot + kernel + rootfs) after you successfully complete the first one, so both slots will be fully compatible.

acostach, 2022-01-27:
I see, thanks Matt. Unfortunately on the TX2 the 32.6.1 boot firmware won't load 32.5.1 bootloader (above error) and it doesn't work the other way around either, 32.5.1 boot firmware doesn't load the newer 32.6.1 bootloader due to the same error.

Best approach I see for now on the TX2 is to fallback to loading new boot firmware & new bootloader + old kernel & old rootfs in case of failure and then re-write the old boot firmware to restore to 32.5.1, like we did so far.

acostach, 2022-02-09:
Sorry to bother again @madisongh , I was testing the fallback on Xavier AGX and it works fine between 32.6.1 and 32.6.1, however from 32.6.1 to 32.5.1 it doesn't appear to work (i.e started from 32.5.1, updated to 32.6.1 but failed to boot in 32.6.1 enough times so fallback to loading 32.5.1 dtb and kernel). It looks like after the retries number is reached, the bootloaders enter recovery mode.

Although cboot 32.6.1 is loaded on the last attempt, it goes trough `#### Fixed storage boot ####`, and fixed_boot.c doesn't seem to have the A/B patches that removable_boot.c has.

This is the log from the last failed attempt:

```
��
  ��mrq initialized
��[0001.077] I> Welcome to Cboot
��fmon_post initialized
��[0001.078] I> Cboot Version: 32.06.01-poky-t194-dc175a98
[0001.078] I> CPU-BL Params @ 0xf2820000
[0001.079] I>  0) Base:0x00000000 Size:0x00000000
[0001.084] I>  1) Base:0xf1100000 Size:0x00100000
[0001.088] I>  2) Base:0xf2000000 Size:0x00200000
[0001.093] I>  3) Base:0xf1200000 Size:0x00200000
��clk_set_parent failed for clk i2c2, parent pll_aon (-22)
clk_set_parent failed for clk i2c8, parent pll_aon (-22)
clk_dt_late initialized
machine_check initialized
pm_post initialized
dbells initialized
avfs_clk_platform_post initialized
dmce initialized
cvc initialized
ccplex_avfs_hw_init: nafll_cluster0: not monitored
ccplex_avfs_hw_init: nafll_cluster1: not monitored
ccplex_avfs_hw_init: nafll_cluster2: not monitored
ccplex_avfs_hw_init: nafll_cluster3: not monitored
avfs_clk_mach_post initialized
regulator_post initialized
rm initialized
sc7_diag initialized
thermal_test initialized
serial_late initialized
clk_post initialized
clk_dt_post initialized
mc_reg initialized
pg_post initialized
dyn_modules initialized
sku_debugfs initialized
speedo_debugfs initialized
adc_debugfs initialized
clk_debugfs initialized
��[0001.097] I>  4) Base:0xf1000000 Size:0x00100000
[0001.176] I>  5) Base:0xf0f00000 Size:0x00100000
[0001.181] I>  6) Base:0xf3800000 Size:0x00400000
[0001.185] I>  7) Base:0xf1c00000 Size:0x00400000
[0001.190] I>  8) Base:0xf0e00000 Size:0x00100000
[0001.194] I>  9) Base:0xf0d00000 Size:0x00100000
[0001.199] I> 10) Base:0xf3000000 Size:0x00800000
[0001.203] I> 11) Base:0x40000000 Size:0x00040000
[0001.208] I> 12) Base:0xf0c00000 Size:0x00100000
[0001.212] I> 13) Base:0x40046000 Size:0x00002000
[0001.217] I> 14) Base:0x40048000 Size:0x00002000
��emc_debugfs initialized
dvs_debugfs initialized
fmon_debugfs initialized
vmon_debugfs initialized
pg_debugfs initialized
profile_fs initialized
debugfs_cons initialized
mail_fs initialized
profile initialized
cvc_debugfs initialized
dmce_debugfs initialized
ec_debugfs initialized
rm_debugfs initialized
soctherm_debug initialized
[... 570 lines cut ...]
[0000.157] I> Boot-device: eMMC
[0000.167] I> Sdmmc: HS400 mode enabled
[0000.171] I> ECC region[0]: Start:0x0, End:0x0
[0000.175] I> ECC region[1]: Start:0x0, End:0x0
[0000.179] I> ECC region[2]: Start:0x0, End:0x0
[0000.184] I> ECC region[3]: Start:0x0, End:0x0
[0000.188] I> ECC region[4]: Start:0x0, End:0x0
[0000.192] I> Non-ECC region[0]: Start:0x80000000, End:0x100000000
[0000.198] I> Non-ECC region[1]: Start:0x0, End:0x0
[0000.202] I> Non-ECC region[2]: Start:0x0, End:0x0
[0000.206] I> Non-ECC region[3]: Start:0x0, End:0x0
[0000.211] I> Non-ECC region[4]: Start:0x0, End:0x0
[0000.216] E> FAILED: Thermal config
[0000.224] E> FAILED: MEMIO rail config
[0000.237] I> Boot-device: eMMC
[0000.246] I> sdmmc bdev is already initialized
[0001.790] E> WP-1 ACK pending
[0001.793] E> Error: 19 
[0001.795] E> Task 72 failed (err: 0x32320006)
[0001.799] E> Top caller module: CPUINIT, error module: CPUINIT, reason: 0x06, aux_info: 0x00
[0001.807] I> MB1(1.5.1.7-t194-41334769-98030a79) BIT boot status dump :
0000000000011111111110111111111111111111111100111111101110011111111011011000000000000000000000000000000000000000000000000000000011111011000000
[0001.837] I> Reset to recovery mode

```

madisongh, 2022-02-09:
@acostach I'm not sure I understand.  cboot has supported A/B slot selection for fixed boot devices for quite a while, so no patches to that code path are needed.  It was only the removable devices that didn't have that support.

> [0008.572] I> A/B: bin_type (37) slot 1 <-- Should have loaded from slot 0

That slot number is read from a register, and should match up with the slot number reported by the earlier bootloaders (e.g. MB1).

> (i.e started from 32.5.1, updated to 32.6.1 but failed to boot in 32.6.1 enough times so fallback to loading 32.5.1 dtb and kernel). It looks like after the retries number is reached, the bootloaders enter recovery mode.

The reversion to recovery mode is a drawback of NVIDIA's boot count/priority implementation, which is why for the Mender integration we mark the boot as successful in the initramfs and use a separate boot counter to track OS boot failures.

How did you implement your updates?  Is it not a full boot chain/kernel/rootfs per boot slot? I'm not sure how the fallback isn't pairing the 32.5.x bootloader with your 32.5.x kernel/dtb/etc.

acostach, 2022-02-10:
Hi @madisongh and thank you for your reply, I'll try to provide a more detailed explanation of the update process on the AGX:

1) Start from a 32.5.1 based system
2) Update MB1_b, MB1_BCT_b, spe-fw_b, mb2_b, mts-preboot_b, smd_b, MEM_BCT_b inside /dev/mmcblk0boot0 (I write 32.6.1 new signed binaries at their _b offset, leaving slots _a inside mmcblk0boot0 intact.
3) Write mts_b, mts-proper_b, cboot_b, bootloader-dtb_b, tos_b etc .kernel_b with 32.6.1 signed binaries insider their mmcblk0pX_b partitions.
4) run tegra-boot-control -e and  tegra-boot-control -a 1 and then check with tegra-boot-control -s that slot 1 (_b) has been activated, then reboot
5) on the next boot (32.6.1) I have a rootfs that causes the new kernel to panic, so that I can check redundant boot. For normal boot cases, I use tegra-boot-control -m in the next rootfs boot (root_b)  to mark the current slot as successful.

Unfortunately after 7 failed boots, I'm seeing the MB1 from 32.6.1 running and putting the system in recovery mode, while I was expecting MB1 from 32.5.1 to run, because slots _a were not updated inside /dev/mmcblk0boot0, and nor were any /dev/mmcblk0pX_a partitions updated.

Also, if at step 2) I update both _a and _b slots inside mmcblk0boot0 with 32.6.1 signed binaries (so not only _b slots), I'm seeing the exact same problem, MB1 from 32.6.1 puts the device in recovery mode after 7 failed attempts. This doesn't happen on the Xavier NX though, when jumping from 32.5.1 to 32.6.1.

Next, if at step 2) I update only slots _b inside mmcblk0boot0 with 32.6.1 signed binaries and leave slots _a on 32.5.1  and don't use tegra-boot-control -a 1 to switch to slot _b, the system will reboot into the same 32.5.1 OS without problems, by running MB1 from 32.5.1. This confirms that updating only slots _a or _b inside mmcblk0boot0 works as expected, and does not affect booting 32.5.1, nor 32.6.1. 

Finally, if I do perform the same steps above but update from 32.6.1 to 32.6.1, so just switching between slots _a and _b that both have the same 32.6.1 binaries and just a rootfs that fails to mark the current boot as successful (kernel panic), the system falls back to slot _a instead of going into recovery mode.

Thank you

acostach, 2022-02-10:
Ok, seems that I found a solution for the AGX, when starting from 32.5.1 I have to update both _a and _b slots of the hw_partitions ( xusb-fw, secure_os, cpu-bootloader etc)  to 32.6.1 and leave the old kernel and kerner-dtb partitions on 32.5.1.

This way MB1, while still at the version from 32.6.1, will successfully fallback to the old slot and not enter debug mode. This is enough to boot the old OS and do a full recovery to 32.5.1. Thanks again for looking into this!

closed 2022-02-10 by acostach