OE4T/meta-tegra #891: [Xavier NX Devkit SD-CARD] Occasional cboot panic and halt on 32.6.1 after reboot

acostach, 2022-01-30:
Not sure if this has been happening with older L4Ts but I've noticed this sporadic panic in cboot 32.6.1, happens occasionally after rebooting the device multiple times:

```
] ��[0001.031] I>  2) Base:0xf2000000 Size:0x00200000
[0001.185] I>  3) Base:0xf1200000 Size:0x00200000
[0001.190] I>  4) Base:0xf1000000 Size:0x00100000
[0001.194] I>  5) Base:0xf0f00000 Size:0x00100000
[0001.199] I>  6) Base:0xf3800000 Size:0x00400000
[0001.203] I>  7) Base:0xf1c00000 Size:0x00400000
[0001.208] I>  8) Base:0xf0e00000 Size:0x00100000
[0001.212] I>  9) Base:0xf0d00000 Size:0x00100000
[0001.217] I> 10) Base:0xf3000000 Size:0x00800000
[0001.221] I> 11) Base:0x40000000 Size:0x00040000
[0001.225] I> 12) Base:0xf0c00000 Size:0x00100000
[0001.230] I> 13) Base:0x40046000 Size:0x00002000
[0001.234] I> 14) Base:0x40048000 Size:0x00002000
[0001.239] I> 15) Base:0xac000000 Size:0x00004000
[0001.243] I> 16) Base:0x4004a000 Size:0x00002000
[0001.248] I> 17) Base:0xf0b00000 Size:0x00100000
[0001.252] I> 18) Base:0x4004c000 Size:0x00002000
[0001.257] I> 19) Base:0xf2200000 Size:0x00600000
[0001.261] I> 20) Base:0x4004e000 Size:0x00002000
[0001.266] I> 21) Base:0xf0ad0000 Size:0x0000c000
[0001.270] I> 22) Base:0x00000000 Size:0x00000000
[0001.275] I> 23) Base:0xf0ae0000 Size:0x00020000
[0001.279] I> 24) Base:0xf6000000 Size:0x02000000
[0001.284] I> 25) Base:0x40050000 Size:0x00002000
[0001.288] I> 26) Base:0x40040000 Size:0x00006000
[0001.292] I> 27) Base:0xf1800000 Size:0x00400000
[0001.297] I> 28) Base:0xf4c00000 Size:0x01400000
[0001.301] I> 29) Base:0xf1400000 Size:0x00400000
[0001.306] I> 30) Base:0x00000000 Size:0x00000000
[0001.310] I> 31) Base:0x00000000 Size:0x00000000
[0001.315] I> 32) Base:0xf8000000 Size:0x08000000
[0001.319] I> 33) Base:0x00000000 Size:0x00000000
[0001.324] I> 34) Base:0xf3c00000 Size:0x01000000
[0001.328] I> 35) Base:0xab000000 Size:0x01000000
[0001.333] I> 36) Base:0xa0000000 Size:0x0b000000
[0001.337] I> 37) Base:0xf2800000 Size:0x00800000
[0001.342] I> 38) Base:0x80000000 Size:0x20000000
[0001.346] I> 39) Base:0xb0000000 Size:0x08000000
[0001.350] I> 40) Base:0x00000000 Size:0x00000000
[0001.355] I> 41) Base:0x00000000 Size:0x00000000
[0001.359] I> 42) Base:0x00000000 Size:0x00000000
[0001.364] I> 43) Base:0x00000000 Size:0x00000000
[0001.368] I> 44) Base:0x00000000 Size:0x00000000
[0001.373] I> 45) Base:0x00000000 Size:0x00000000
[0001.377] GIC-SPI Target CPU: 0
[0001.380] Interrupts Init done
[0001.383] calling constructors
[0001.386] initializing heap
[0001.389] I> Heap: [0xa069cb70 ... 0xab000000]
[0001.393] initializing threads
[0001.396] initializing timers
[0001.399] creating bootstrap completion thread
[0001.403] top of bootstrap2()
[0001.406] CPU: MIDR: 0x4E0F0040, MPIDR: 0x80000000
[0001.410] initializing platform
[0001.413] E> DEVICE_PROD: Invalid value data = 0, size = 0.
[0001.419] W> device prod register failed
[0001.422] I> Bl_dtb @0xaaf00000
[0001.431] W> "plugin-manager" doesn't exist, creating
[0001.431] W> "ids" doesn't exist, creating
[0001.435] W> "connection" doesn't exist, creating
[0001.439] W> "configs" doesn't exist, creating
[0001.450] E> failed to read label property for node 158736: 13
[0001.452] E> failed to read reg property for node 158800: 13
[0001.456] E> failed to read reg property for node 158852: 13
[0001.461] E> failed to read reg property for node 158936: 13
[0001.469] I> Find /i2c@3160000's alias i2c0
[... 343 lines cut ...]
[0006.480] => pc:0x00000000, sp:0xA06A7F30
[0006.484] => pc:0xA060F880, sp:0xA06A8160
[0006.488] => pc:0xA060F89C, sp:0xA06A81B0
[0006.492] => pc:0xA060F644, sp:0xA06A81E0
[0006.495] => pc:0xA060EB68, sp:0xA06A81F0
[0006.499] => pc:0xA060EB30, sp:0xA06A8200
[0006.503] -----------------------------------------------
[0006.508] iframe 0xa06a7e40:
[0006.511] x0  0x        90000000 x1  0x               0 x2  0x               0 x3  0x               0
[0006.520] x4  0x               0 x5  0x              20 x6  0x         b200123 x7  0x        ffffffc0
[0006.529] x8  0x               1 x9  0xffffffffffffffff x10 0x               6 x11 0x               2
[0006.538] x12 0x               1 x13 0x              40 x14 0x               1 x15 0x             2c0
[0006.548] x16 0x        a061c4a8 x17 0x               0 x18 0x               0 x19 0x               0
[0006.557] x20 0x               0 x21 0x               0 x22 0x               0 x23 0x               0
[0006.566] x24 0x               0 x25 0x               0 x26 0x               0 x27 0x               0
[0006.575] x28 0x               0 x29 0x        a06a8160 lr  0x        a060f7f4 sp  0x        a06a7f30
[0006.584] elr 0x               0
[0006.587] spsr 0x        400003c9
[0006.590] -----------------------------------------------
[0006.595] panic (caller 0xa0601238): die
[0006.599] HALT: spinning forever...

```

I guess the sd-card is sometimes in a bad state so I'm wondering if it would it be possible to configure cboot to simply reboot in this case instead of halting? Thank you

madisongh, 2022-01-30:
That it's triggering an exception like that is probably a bug - maybe not properly handling the case where no boot device is found? Considering how complicated the t19x boot sequence has become as they've tacked on support for NVME, extlinux, etc., CBO, etc., it wouldn't surprise me that some infrequently-exercised code path is broken.

Still, bugs happen, and cboot should have a watchdog timer enabled by default, unless you've configured otherwise. I don't remember the default timeout off the top of my head, but it's on the order of minutes.

acostach, 2022-01-30:
Thanks Matt, I left it for about 5 minutes and nothing happened but I'll reproduce this again and leave it for longer. I was reading in the docs that the PMIC WDT is disabled by default from odmdata on the Xavier NX and that's why I was asking. I'll get back with an update.

acostach, 2022-01-31:
@madisongh Unfortunately the device did not restart after 40 minutes in this state, perhaps that watchdog is not enabled by default?

madisongh, 2022-01-31:
Sure enough, the default ODMDATA (`0xB8190000`) disables the PMIC WDT by default.  I thought they at least enabled the internal WDT in the processor, but perhaps not.  Try flashing with ODMDATA set to `0xB81A0000`, that should enable the PMIC WDT, if I'm reading the cboot sources correctly.

acostach, 2022-02-01:
Hi @madisongh , unfortunately the PMIC WDT doesn't appear to work with systemd, the device reboots as soon as it finishes booting. We're using systemd to kick the watchdog with RuntimeWatchdogSec set to 10s. This doesn't happen with PMIC WDT disabled in the previous ODMDATA though.

madisongh, 2022-02-01:
Ah, right, you may need to modify your device tree and/or kernel config and/or systemd configuration to switch to using the PMIC WDT from Linux, too.  I believe the driver is enabled by default in the default kernel config (CONFIG_MAX77620_WATCHDOG), but it may not be enabled by default in the device tree.  Even if it is, if there are multiple watchdog devices, systemd uses `/dev/watchdog0` by default, which could be the wrong one.

acostach, 2022-02-02:
@madisongh yes, the config is enabled and from what I see the plugin manager also enables it in the device tree based on the selected odmdata.

There are no other watchdog devices created (only watchdog and watchdog0) and if I remove the watchdog.conf file systemd no longer opens the device and the board doesn't reboot. If however i do a cat on /dev/watchdog the board is rebooted immediately.

madisongh, 2022-02-02:
Now that you mention it, I vaguely remember running into a similar problem on the TX2 a while back. IIRC there was a bug somewhere that kept the SoC internal WDT around even when I chose the PMIC WDT.  Looking at my kernel config, I think the solution was to just disable the support for the other WDTs there - here's my `pmic-watchdog-only.cfg` fragment:

```
# CONFIG_TEGRA21X_WATCHDOG is not set
# CONFIG_TEGRA18X_WATCHDOG is not set
# CONFIG_SOFT_PLATFORM_WATCHDOG is not set
CONFIG_MAX77620_WATCHDOG=y
```

I don't remember now whether the bug was in cboot or the kernel, but there could be something similar going on here.

acostach, 2022-02-02:
Many thanks @madisongh! indeed, disabling the other tegra watchdog modules made the max77620-watchdog work with systemd.

acostach, 2022-02-04:
Hi @madisongh, unfortunately enabling the PMIC WDT did not solve the original cboot problem, I managed to reproduce the issue once more and left the board running for 30 minutes, the pmic did not reset it though.

The easiest way to reproduce the watchdog not resetting the device is to flash the NX SD with that odmdata and then remove the sd-card or any other medium from which it could boot before powering it on.

I guess that maybe cboot doesn't actually start this watchdog?

I see it's overriding some nodes in the dtb but a not sure if they have anything to do with this:

```
[0001.653] I> Plugin-manager override starting
[0001.658] I> node /plugin-manager/fragment-pcie-c5-rp matches
[0001.666] I> node /plugin-manager/fragement-pmic-wdt-en matches
[0001.670] I> node /plugin-manager/fragement-tegra-wdt-dis matches
[0001.676] I> node /plugin-manager/fragement-tegra-sdhci-emmc-dis matches
[0001.684] I> Disable plugin-manager status in FDT
[0001.686] I> Plugin-manager override finished successfully
```

madisongh, 2022-02-05:
Hmm.  I took a closer look at the cboot code, and sure enough, it disables the WDT as part of the kernel handoff. I don't know if that's a recent change, or whether I was misremembering how it worked, but that surprised me. :(

Also, it turns out that cboot always uses the internal WDT, even if you've chosen the PMIC WDT via ODMDATA.

Some additional patches to cboot will be needed to addresses these shortcomings.

acostach, 2022-02-05:
Thanks again for looking into this @madisongh ! I see that tegrabl_reset() makes the boot sequence restart from MB1, at least on the NX devkit, and am wondering if resetting instead of halting is a good idea. I'll try to reproduce the problem to see if cboot is able to load the kernel and dtb from the raw partitions after reset with this:

```
+++ b/bootloader/partner/t18x/cboot/platform/tegra_shared/debug.c
@@ -15,6 +15,7 @@
 #include <platform_c.h>
 #include <printf.h>
 #include <tegrabl_timer.h>
+#include <tegrabl_exit.h>
 #include <tegrabl_debug.h>

 #if defined(CONFIG_DEBUG_TIMESTAMP)
@@ -91,7 +92,9 @@ int platform_dgetc(char *c, bool wait)

 void platform_halt(void)
 {
-       dprintf(ALWAYS, "HALT: spinning forever...\n");
+       dprintf(ALWAYS, "Will reset in 10 seconds...\n");
+       tegrabl_mdelay(10 * 1000);
+       tegrabl_reset();

```

acostach, 2022-02-05:
Yes, after the tegrabl_reset() the device is able to boot normally:

```
[0006.571] x24 0x               0 x25 0x               0 x26 0x               0 x27 0x               0
[0006.580] x28 0x               0 x29 0x        a06a8160 lr  0x        a060f7f4 sp  0x        a06a7f30
[0006.589] elr 0x               0
[0006.592] spsr 0x        400003c9
[0006.595] -----------------------------------------------
[0006.600] panic (caller 0xa0601238): die
[0006.604] Will reset in 10 seconds...
[0016.607] E> tegrabl_display_shutdown: display is not initialized
����Shutdown state requested 1
Rebooting system ...
��
[0000.024] W> RATCHET: MB1 binary ratchet value 4 is too large than ratchet level 2 from HW fuses.
[0000.033] I> MB1 (prd-version: 1.5.1.7-t194-41334769-98030a79)
[0000.038] I> Boot-mode: Coldboot
[0000.041] I> Chip revision : A02P
[0000.044] I> Bootrom patch version : 15 (correctly patched)
[0000.049] I> ATE fuse revision : 0x200
[0000.053] I> Ram repair fuse : 0x0
[0000.056] I> Ram Code : 0x0
[0000.058] I> rst_source : 0xb
[0000.061] I> rst_level : 0x1
[0000.065] I> Boot-device: QSPI
[0000.067] I> Qspi flash params source = brbct

```

madisongh, 2022-02-06:
@acostach Great, that looks like a much simpler way to solve the problem you're seeing.  Please leave this issue open, though, as I will try and solve the problems with cboot's WDT handling, when I get a chance.

ichergui, 2025-01-26:
There is not update on this issue for a while. I'm closing it 
Don't hesitate to re-open a new issue.

closed 2025-01-26 by ichergui