OE4T/meta-tegra #913: R32.6.1-t186: WiFi bcmdhd driver doesn't work properly

ichergui, 2022-02-20:
**Describe the bug**
Testing on `master`, `jetson-tx2-devkit`, `demo-image-base`

WiFi driver `bcmdhd` doesn't work properly when enabling `KERNEL_INTERNAL_WIRELESS_REGDB`

```
[    9.591243] systemd-journald[4042]: Received client request to flush runtime journal.
[    9.978273] dhd_module_init in
[    9.978382] found wifi platform device bcmdhd_wlan
[    9.981178] wifi_platform_get_country_code_map: could not get country_code_map
[    9.981181] wifi_plat_dev_drv_probe:platform country code map is not available
[    9.981196] Power-up adapter 'DHD generic adapter'
[    9.981203] wifi_platform_set_power = 1
[   10.018057] gpio tegra-gpio wake18 for gpio=101(M:5)
[   10.019956] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[   10.129381] gpio tegra-gpio-aon wake30 for gpio=60(FF:4)
[   10.189263] wifi_platform_bus_enumerate device present 1
[   10.233467] wifi_platform_bus_enumerate device present 0
[   10.250602] F1 signature read @0x18000000=0x17214354
[   10.278983] F1 signature OK, socitype:0x1 chip:0x4354 rev:0x1 pkg:0x2
[   10.294586] DHD: dongle ram size is set to 786432(orig 786432) at 0x180000
[   10.320929] wifi_platform_get_mac_addr
[   10.333306] ids dhd_log_netlink_init
[   10.340724]  wl_cfg80211_reg_notifier : [   10.352025] reg_notifier for intiator:0 not supported : set default
[   10.361257]  wl_cfg80211_reg_notifier : [   10.365934] Set country code 00 from User
[   10.373280] Internal error: Accessing user space memory outside uaccess.h routines: 96000005 [#1] PREEMPT SMP
[   10.385867] Modules linked in: bluedroid_pm bcmdhd(+) nvgpu cfg80211 fuse
[   10.395381] CPU: 0 PID: 4216 Comm: systemd-udevd Not tainted 4.9.253-l4t-r32.6+g40e3027fe477 #1
[   10.406727] Hardware name: quill (DT)
[   10.412967] task: ffffffc1e25c0000 task.stack: ffffffc1e25c8000
[   10.421617] PC is at wl_cfg80211_reg_notifier+0x9c/0x1d0 [bcmdhd]
[   10.430565] LR is at wl_cfg80211_reg_notifier+0x16c/0x1d0 [bcmdhd]
[   10.439255] pc : [<ffffff800141ae1c>] lr : [<ffffff800141aeec>] pstate: 80400045
[   10.449117] sp : ffffffc1e25cb470
[   10.454818] x29: ffffffc1e25cb470 x28: 0000000000000001 
[   10.462496] x27: ffffff800a5df978 x26: ffffffc1e0b60000 
[   10.470189] x25: ffffff80010efd90 x24: ffffff800149fb68 
[   10.477861] x23: ffffffc1e0b2f900 x22: 0000000000000000 
[   10.485476] x21: ffffff800149fb68 x20: ffffffc1e0b2f900 
[   10.493079] x19: ffffffc1da4802a0 x18: ffffffffffffffff 
[   10.500637] x17: 0000000000000000 x16: 0000000000000000 
[   10.508094] x15: 0000000000000001 x14: 0000000000000020 
[   10.515524] x13: ffffff80090d2d68 x12: 0000000000000000 
[   10.522944] x11: 00000000000003bd x10: 0000000000000001 
[   10.530257] x9 : ffffffc1eafafc40 x8 : ffffff8008414780 
[   10.537581] x7 : ffffff800a3146e8 x6 : 000000000000002d 
[   10.544863] x5 : 0000000000000000 x4 : ffffffc1f5f74bf0 
[   10.552092] x3 : 0000000000000000 x2 : 0000000000000000 
[   10.559303] x1 : ffffffc1e0b2f91c x0 : 0000000000000000 
[   10.566454] 
[   10.569740] Process systemd-udevd (pid: 4216, stack limit = 0xffffffc1e25c8000)
[   10.578820] Call trace:
[   10.583191] [<ffffff800141ae1c>] wl_cfg80211_reg_notifier+0x9c/0x1d0 [bcmdhd]
[   10.592248] [<ffffff800107c714>] wiphy_update_regulatory+0x174/0x1b0 [cfg80211]
[   10.601434] [<ffffff800107e148>] wiphy_regulatory_register+0x48/0x60 [cfg80211]
[   10.610861] [<ffffff8001072c64>] wiphy_register+0x384/0x580 [cfg80211]
[   10.628312] [<ffffff800141e640>] wl_setup_wiphy+0x1e0/0x3a0 [bcmdhd]
[   10.645571] [<ffffff800142ac6c>] wl_cfg80211_attach+0x7c/0x410 [bcmdhd]
[   10.662971] [<ffffff80013f6794>] dhd_attach+0x214/0x7a0 [bcmdhd]
[   10.679639] [<ffffff800146165c>] dhdsdio_probe+0x16c/0x710 [bcmdhd]
[   10.696553] [<ffffff800144af9c>] bcmsdh_probe+0xfc/0x180 [bcmdhd]
[   10.713338] [<ffffff800144d518>] bcmsdh_sdmmc_probe+0xf8/0x160 [bcmdhd]
[   10.729960] [<ffffff8008c513a0>] sdio_bus_probe+0xa0/0x150
[   10.745465] [<ffffff80087ed840>] really_probe+0x160/0x3b0
[   10.760877] [<ffffff80087eddc4>] driver_probe_device+0x54/0xd0
[   10.776766] [<ffffff80087edf48>] __driver_attach+0x108/0x130
[   10.792482] [<ffffff80087eb0b0>] bus_for_each_dev+0x80/0xc0
[   10.808149] [<ffffff80087ed1c8>] driver_attach+0x38/0x50
[   10.823507] [<ffffff80087ecbec>] bus_add_driver+0x178/0x27c
[   10.839176] [<ffffff80087eefb4>] driver_register+0x84/0x130
[   10.854863] [<ffffff8008c51198>] sdio_register_driver+0x38/0x50
[   10.871561] [<ffffff800144d6f4>] bcmsdh_register_client_driver+0x24/0x40 [bcmdhd]
[   10.889944] [<ffffff800144b208>] bcmsdh_register+0x38/0x50 [bcmdhd]
[   10.907072] [<ffffff80014664d0>] dhd_bus_register+0x30/0x70 [bcmdhd]
[   10.924311] [<ffffff8001444588>] dhd_wifi_platform_load_sdio+0x208/0x4f0 [bcmdhd]
[   10.942676] [<ffffff8001444a94>] wifi_plat_dev_drv_probe+0x184/0x420 [bcmdhd]
[   10.959994] [<ffffff80087f0768>] platform_drv_probe+0x68/0xe4
[   10.975855] [<ffffff80087ed840>] really_probe+0x160/0x3b0
[   10.991343] [<ffffff80087eddc4>] driver_probe_device+0x54/0xd0
[   11.007249] [<ffffff80087edf48>] __driver_attach+0x108/0x130
[   11.022920] [<ffffff80087eb0b0>] bus_for_each_dev+0x80/0xc0
[   11.038460] [<ffffff80087ed1c8>] driver_attach+0x38/0x50
[   11.053678] [<ffffff80087ecbec>] bus_add_driver+0x178/0x27c
[   11.069097] [<ffffff80087eefb4>] driver_register+0x84/0x130
[   11.084441] [<ffffff80087f0688>] __platform_driver_register+0x64/0x7c
[   11.101373] [<ffffff8001444f24>] wifi_ctrlfunc_register_drv+0x1f4/0x300 [bcmdhd]
[   11.119332] [<ffffff800144509c>] dhd_wifi_platform_register_drv+0x6c/0x160 [bcmdhd]
[   11.137512] [<ffffff80011360a8>] dhd_module_init+0xa8/0x1e0 [bcmdhd]
[   11.153665] [<ffffff8008083abc>] do_one_initcall+0x4c/0x130
[   11.169020] [<ffffff80081615f8>] do_init_module+0x68/0x210
[   11.184113] [<ffffff80081629d8>] load_module+0x718/0x8b0
[   11.198943] [<ffffff8008162dd8>] SyS_finit_module+0x98/0xf0
[   11.213847] [<ffffff800808395c>] __sys_trace_return+0x0/0x4
[   11.228587] ---[ end trace a077d4d1c74e2d1d ]---
[   11.432262] CPU1: shutdown
[   11.436589] psci: CPU1 killed (polled 0 ms)
[   11.537368] CPU2: shutdown
[   11.541684] psci: CPU2 killed (polled 0 ms)
[   12.928232] eqos 2490000.ether_qos eth0: Link is Up - 1Gbps/Full - flow control rx/tx
```  

Needs investigation.

**To Reproduce**
Steps to reproduce the behavior:

Use [tegra-demo-distro](https://github.com/OE4T/tegra-demo-distro), branch `master`
Build `demo-image-base`
Load onto target (`jetson-tx2-devkit`)

madisongh, 2022-02-21:
Looks like there's a missing return after reporting the error (line 8654 in wl_cfg80211.c).
And if the driver doesn't support REGDOM_SET_BY_CORE, there is a flag it can set in its wiphy structure to tell the core not to send those notifications.

madisongh, 2022-02-21:
I've fixed the code path in the driver and updated the recipe to point to the hash with the fix.

closed 2022-02-21 by madisongh