OE4T/meta-tegra #923: perf bbappend causes taskhash mismatch

sapiippo, 2022-02-25:
The perf.bbappend included in meta-tegra is causing taskhash mismatch errors during the build.

```
ERROR: perf-1.0-r9 do_configure: Taskhash mismatch c89f42fa090d8796d7ddfd80fda6997fc6acaa631b6dfa87ecfd1e1b19eeb0c1 versus 61af608e350dc15893018d5e54306af10a0db93538a60aa8df9c3227dcbe2e02 for /home/qt/sources/poky/meta/recipes-kernel/perf/perf.bb:do_configure
ERROR: Taskhash mismatch c89f42fa090d8796d7ddfd80fda6997fc6acaa631b6dfa87ecfd1e1b19eeb0c1 versus 61af608e350dc15893018d5e54306af10a0db93538a60aa8df9c3227dcbe2e02 for /home/qt/sources/poky/meta/recipes-kernel/perf/perf.bb:do_configure
ERROR: When reparsing /home/qt/sources/poky/meta/recipes-kernel/perf/perf.bb:do_configure, the basehash value changed from 8aa1073b829aaf78c986cad3eca4670e24d9be248d5f5e4e40aa77811c17cf26 to 48daecbbb1f7cbd0c3c0d475e72ea1736e38ffb4e138b03706cdbc5abbccd09b. The metadata is not deterministic and this needs to be fixed.
```

This is happens because the value of KERNEL_VERSION is read from a file in STAGING_KERNEL_BUILDDIR which doesn't exists when perf recipe is first parsed.

madisongh, 2022-02-25:
We probably need to simplify the PACKAGECONFIG override in the bbappend to just omit the KERNEL_VERSION check, but I'd like to make sure I can reproduce the issue before making any fixes.

Which branch are you using?  I'm not seeing any task hash mismatches with dunfell, but I am on master (along with compilation failures).

sapiippo, 2022-02-28:
I'm currently using master branches, not quite the HEAD, but just before the incompatible variable renames.
I was able reproduce this by running:
bitbake linux-tegra -c clean
bitbake perf

madisongh, 2022-02-28:
Thanks for confirming.  Should be fixed in master now.