OE4T/meta-tegra #943: nvargus-daemon going into an endless error loop

brgl, 2022-03-22:
**Describe the bug**
`nvargus-daemon` is going into an endless error loop at startup. It makes it impossible to use the CSI camera using jetcam. This happens on Nano and Xavier NX. If I do `systemctl restart nvargus-daemon` the daemon goes back into a sane state and can be used which leads me to believe it's some kind of a system dependency issue.

The journal output when this happens looks like this:

```
-- Logs begin at Sat 2000-01-01 01:34:03 UTC, end at Sat 2000-01-01 01:58:46 UTC. --
Jan 01 01:34:05 veki systemd[1]: Started NVIDIA Argus daemon.
Jan 01 01:34:12 veki nvargus-daemon[4299]: === NVIDIA Libargus Camera Service (0.97.3)=== Listening for connections...=== python3[4506]: Connection established (7F949FE1C0)OFParserListModules: module list: /proc/device-tree/tegra-camera-platform/modules/module0
Jan 01 01:34:12 veki nvargus-daemon[4299]: OFParserListModules: module list: /proc/device-tree/tegra-camera-platform/modules/module1
Jan 01 01:34:13 veki nvargus-daemon[4299]: OFParserGetVirtualDevice: NVIDIA Camera virtual enumerator not found in proc device-tree
Jan 01 01:34:13 veki nvargus-daemon[4299]: ---- imager: No override file found. ----
Jan 01 01:34:13 veki nvargus-daemon[4299]: LSC: LSC surface is not based on full res!
Jan 01 01:34:13 veki nvargus-daemon[4299]: ---- imager: No override file found. ----
Jan 01 01:34:13 veki nvargus-daemon[4299]: (NvCamV4l2) Error ModuleNotPresent: V4L2Device not available (in /dvs/git/dirty/git-master_linux/camera/utils/nvcamv4l2/v4l2_device.cpp, function findDevice(), line 256)
Jan 01 01:34:13 veki nvargus-daemon[4299]: (NvCamV4l2) Error ModuleNotPresent:  (propagating from /dvs/git/dirty/git-master_linux/camera/utils/nvcamv4l2/v4l2_device.cpp, function initialize(), line 60)
Jan 01 01:34:13 veki nvargus-daemon[4299]: (NvOdmDevice) Error ModuleNotPresent:  (propagating from dvs/git/dirty/git-master_linux/camera-partner/imager/src/devices/V4L2SensorViCsi.cpp, function initialize(), line 107)
Jan 01 01:34:13 veki nvargus-daemon[4299]: NvPclDriverInitializeData: Unable to initialize driver v4l2_sensor
Jan 01 01:34:13 veki nvargus-daemon[4299]: NvPclInitializeDrivers: error: Failed to init camera sub module v4l2_sensor
Jan 01 01:34:13 veki nvargus-daemon[4299]: NvPclStartPlatformDrivers: Failed to start module drivers
Jan 01 01:34:13 veki nvargus-daemon[4299]: NvPclStateControllerOpen: Failed ImagerGUID 0. (error 0xA000E)
Jan 01 01:34:13 veki nvargus-daemon[4299]: NvPclOpen: PCL Open Failed. Error: 0xf
Jan 01 01:34:13 veki nvargus-daemon[4299]: SCF: Error BadParameter: Sensor could not be opened. (in src/services/capture/CaptureServiceDeviceSensor.cpp, function getSourceFromGuid(), line 582)
Jan 01 01:34:13 veki nvargus-daemon[4299]: SCF: Error BadParameter:  (propagating from src/services/capture/CaptureService.cpp, function addSourceByGuid(), line 437)
Jan 01 01:34:13 veki nvargus-daemon[4299]: SCF: Error BadParameter:  (propagating from src/api/CameraDriver.cpp, function addSourceByIndex(), line 303)
Jan 01 01:34:13 veki nvargus-daemon[4299]: SCF: Error BadParameter:  (propagating from src/api/CameraDriver.cpp, function getSource(), line 466)
Jan 01 01:34:13 veki nvargus-daemon[4299]: Acquiring SCF Camera device source via index 1 has failed. === python3[4506]: CameraProvider initialized (0x7f90ba13a0)LSC: LSC surface is not based on full res!
Jan 01 01:34:13 veki nvargus-daemon[4299]: NvIspAfConfigParamsSanityCheck: Error: positionWorkingHigh is not larger than positionWorkingLow positionWorkingHigh = 0, positionWorkingLow = 0
Jan 01 01:34:13 veki nvargus-daemon[4299]: NvIspAfConfigParamsSanityCheck: Error: positionWorkingHigh is not larger than positionWorkingLow positionWorkingHigh = 0, positionWorkingLow = 0
Jan 01 01:34:13 veki nvargus-daemon[4299]: NvIspAfConfigParamsSanityCheck: Error: positionWorkingHigh is not larger than positionWorkingLow positionWorkingHigh = 0, positionWorkingLow = 0
Jan 01 01:34:31 veki nvargus-daemon[4299]: (Argus) Error OverFlow: Too many pending events, ignoring new events (in src/api/EventProviderImpl.cpp, function addEvent(), line 156)
Jan 01 01:34:31 veki nvargus-daemon[4299]: (Argus) Error OverFlow: Too many pending events, ignoring new events (in src/api/EventProviderImpl.cpp, function addEvent(), line 156)
Jan 01 01:34:31 veki nvargus-daemon[4299]: (Argus) Error OverFlow: Too many pending events, ignoring new events (in src/api/EventProviderImpl.cpp, function addEvent(), line 156)
Jan 01 01:34:31 veki nvargus-daemon[4299]: (Argus) Error OverFlow: Too many pending events, ignoring new events (in src/api/EventProviderImpl.cpp, function addEvent(), line 156)
Jan 01 01:34:31 veki nvargus-daemon[4299]: (Argus) Error OverFlow: Too many pending events, ignoring new events (in src/api/EventProviderImpl.cpp, function addEvent(), line 156)
Jan 01 01:34:31 veki nvargus-daemon[4299]: (Argus) Error OverFlow: Too many pending events, ignoring new events (in src/api/EventProviderImpl.cpp, function addEvent(), line 156)
```

And the same message repeats.

**To Reproduce**

Below is my build config.

```
Build Configuration:
BB_VERSION           = "1.46.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-20.04"
TARGET_SYS           = "aarch64-oe-linux"
MACHINE              = "jetson-nano-devkit"
DISTRO               = "veki"
DISTRO_VERSION       = "0.1.0-dev"
TUNE_FEATURES        = "aarch64 armv8a crc"
TARGET_FPU           = ""
meta                 = "dunfell:79ce9059f716546a7d6f4562ba194aedd90c22cd"
meta-oe              
meta-python          
meta-networking      
meta-multimedia      
meta-filesystems     = "dunfell:0722ff6f021df91542b5efa1ff5b5f6269f66add"
meta-tegra           
contrib              = "dunfell-l4t-r32.5.0:7e5b82c3a5fc774a8e8f2c66dd8c551a8c82f500"
meta-mender-core     = "dunfell:8b2840fc4766ca41619cb562d470239f29b7cac3"
meta-mender-tegra    = "dunfell:d52e47db3fadf97dfb081636c9565b4ac3ba616a"
meta-qt5             = "HEAD:6bfe29d9e8fdd5c2fd17c736814b6df859b3af50"
meta-rust            = "HEAD:bf6705fb7f41b16c3277cbf84af0b9757dcd3a01"
meta-tegra-community = "dunfell-l4t-r32.5.0:b197cc612aa97bd76f578da798ef8f3508bc9a4f"
meta-scikit-learn    = "dunfell:c8bbe97b8a6069f8c3e87c5f860434b60b335b5b"
meta-aws             = "dunfell:9979cfa676105cb68cfadfdaeabf044d7c919319"
```

**Additional context**
While googling for an answer I'm seeing people posting a similar issue since 2019 but none of these issues have any answers. I'd swear the same image used to work at some point but bisecting led me nowhere and we don't have a proper CI in place yet to spot it.

madisongh, 2022-03-22:
Argus logs make it hard to distinguish real errors from "normal" ones, but the ones about it not being able to find the V4L2 device sound like there's some kind of low-level issue -- missing (or not yet loaded) driver, missing device tree entries, or something.

Is this a 2-camera setup, and what kind of cameras are they?  Can you check that the `/dev/video0` and `/dev/video1` devices exist?  Any kernel logs from the camera drivers?

brgl, 2022-03-23:
The kernel is coming from linux-tegra and the device tree is the regular one for nano. I've only ever had `/dev/video0` and the camera we're using is imx219 connected over CSI. Both ports have the same issue.

Right before the argus daemon starts, I'm seeing this in v4l2-ctl:

```
Jan 01 01:00:20 veki v4l2-ctl[4330]: vi-output, imx219 8-0010 (platform:54080000.vi:4):
Jan 01 01:00:20 veki v4l2-ctl[4330]:         /dev/video0
Jan 01 01:00:20 veki v4l2-ctl[4330]: NVIDIA Tegra Video Input Device (platform:vi):
Jan 01 01:00:20 veki v4l2-ctl[4330]:         /dev/media0
```

So it looks like the camera is up.

From the kernel log I can see this:

```
root@veki:~# dmesg | grep imx
[    3.663287] imx219 7-0010: tegracam sensor driver:imx219_v2.0.6
[    3.686714] imx219 7-0010: imx219_board_setup: error during i2c read probe (-121)
[    3.686822] imx219 7-0010: board setup failed
[    3.686898] imx219: probe of 7-0010 failed with error -121
[    3.687507] imx219 8-0010: tegracam sensor driver:imx219_v2.0.6
[    5.293492] vi 54080000.vi: subdev imx219 8-0010 bound
```

Also looks good - the first port is empty, the second one goes up.

madisongh, 2022-03-23:
OK, I've booted up a Nano with an IMX219 attached to help sift through the argus messages, and yours compare to mine pretty well.

Do you have a systemd service for your camera application so it starts automatically at boot time?  I do in one of my projects, and I do have a hack/workaround for some timing issues with nvargus-daemon that delays the start of the app for 5 seconds after nvargus-daemon starts.  I didn't implement that particular workaround, but the comment on the relevant commit mentions it was to help with "camera stability", so I suspect we saw something similar at some point.

You could see if something like that would help.

brgl, 2022-03-23:
Alright, so I now realized what the logs are saying. Someone is indeed connecting to the daemon. Except it's not our camera app because I disabled it. Any idea how to determine who this is? There doesn't seem to be any debug switch in the executable.

madisongh, 2022-03-23:
You could try disabling the nvargus-daemon service and see if that causes another failure.

Or from the log you see the line with the `===` - that indicates the process that's connecting.  Something running `python3`, and I think the `6124` is a process ID.

brgl, 2022-03-25:
I managed to track the offending process and discovered a rather "interesting" design approach in jetcam. The object representing the CSI camera opens a connection to the nvargus daemon in the constructor and tells it to read frames from camera immediately. If you don't start consuming those frames in a matter of seconds from the daemon, it overflow and causes the above. The only way out seems to create the jetcam object only when it's needed and destroy it when it isn't or use gstreamer directly.

Anyway, not a meta-tegra issue so closing this.

closed 2022-03-25 by brgl

madisongh, 2022-03-25:
Thanks for the update, glad you figured out the issue.