OE4T/meta-tegra #955: Problems with Wayland/Weston

Achder, 2022-04-14:
I'm having some problem with wayland/weston.
It's basically three different scenarios of which 2 feel like crashes:

1. (crash?) When I move the mouse directly after boot (within the first 2-3 seconds after the desktop appears on screen), the desktop environment instantly vanishes and I get thrown into a mix of system log and terminal login. (it's basically both. I'm able to login as root and executes commands but there's also system log messages mixed into it) If I wait a few seconds after boot before using the mouse everything is fine.
2. (stand-by-like) When interacting with the mouse or keyboard I sometimes randomly get thrown into the mix of systemlog and login view the second I stop moving or typing. In this case weston pops up again when I start typing or move the mouse. It does not happen every time and I can't predict when it will. Also note this is different from the lock screen with the small window and the green circle in it. But it kind of feels a standy mode.
3. (crash?) Just now it crashed while a process was printing to the console without me doing anything else for the first time. Again it's only weston crashing. I still have terminal access.

Steps to reproduce:
For us it's just building the core-image-weston. We connect a display over HDMI and use USB mouse and keyboard.

Additional info:
I tried starting weston after it crashed but I get "error: XDG_RUNTIME_DIR not set in the environment".

board: jetson nano devkit
poky: dunfell (up to date)
meta-tegra: dunfell (also up to date)

kekiefer, 2022-04-14:
Can you confirm if these problems exist in the `dunfell-32.6.1` branch? There appear to be some graphics stack updates in the 32.7.1 release which are causing problems in other places as well.

It's worth noting that the master branch (and shortly, the kirkstone release) rework the weston integration to use gbm instead of egldevice/eglstreams, and it appears to be stable with the 32.7.1 graphics stack.

ichergui, 2022-06-23:
Hey @Achder 
Any updates concerning your issue ?
Did you try our master branch ?
Please share with us how it goes
Thanks

Achder, 2022-06-24:
Hey,

I‘m sorry. I‘m not working at that company anymore. But we couldn‘t switch branches so easily because we had lots of dependencies.

But thanks again for the great help with everything. You‘re a great team!

ichergui, 2022-06-28:
@kekiefer should we close this issue ? Or you think that we need at least a test of tegra-demo-Weston with the following branches:
* dunfell-l4t-r32.6.1 
* dunfell-L4T-r32.7.1
* master

kekiefer, 2022-07-03:
We have Weston support in master and Kirkstone that works with the latest BSP, and it took a different approach than forward-porting NVIDIA's Weston patches. If someone else needs the latest BSP and finds broken the patched Weston from the aging Dunfell release, I'm happy to review any pull request that fixes it, but I personally don't have the bandwidth to sort this out.

closed 2022-07-03 by kekiefer