OE4T/meta-tegra #956: Blank screen with Qt5 (EGLFS + KMS/DRM)

benediktley, 2022-04-14:
Since L4T32.7.1 (dunfell branch) my Qt application does not render any more. 
The application is starting and interaction with mouse/keyboard/touch is working but the screen is black. No issues on previous L4T32.6.1 branch. I do not use X11 or wayland, both are disabled.

Qt installation from [meta-qt5](https://github.com/meta-qt5/meta-qt5) (tested both, dunfell & master)
`PACKAGECONFIG_append_qtbase = " eglfs kms gles2 "`

runtime environment variables:
`QT_QPA_PLATFORM=eglfs`
`QT_QPA_EGLFS_INTEGRATION=eglfs_kms_egldevice`

Qt sample application:
[https://github.com/alpqr/quickmwtest](url)

kekiefer, 2022-04-14:
Confirming that I'm seeing the same with qt5's egldevice and 32.7.1 on the master branch. It's worth noting that qt5 isn't receiving updates anymore an there's a possibility of fixes upstream in qt6, but I haven't looked. Patching qt5's egldevice integration work with tegra-udrm is another avenue to explore. Do you know if NVIDIA's jetpack release works with qt5?

benediktley, 2022-04-14:
Just did a brief test on the latest jetpack release (4.6.1) and run the sample application. Had some issues compiling qt, so i did the test with the prebuilt deb version from repo (qt5.9.5). Same issue here: Application runs without any error but the screen remains black.

I'm testing the whole thing again with Qt6 but won't have time until next week as I'm travelling for a few days.

kekiefer, 2022-04-14:
I think I have a solution to this. It appears that the newer nvidia libdrm is rejecting the calls to `drmModeSetCrtc` with the framebuffer id `-1`, which was accepted before to allow qt to perform modesetting without flipping buffers (which was left to the egl). I can set up a dummy framebuffer to flip initially, and then things are happy. Patches are forthcoming.

PR #959 (2022-04-15): external/qt5-layer: fix egldevice backend for L4T 32.7.1
This change adds two patches to support tegra devices in qt5 with
the L4T release 32.7.1. The newer version of L4T appears to unify
the capabilities of tegra-udrm and drm-nvdc drivers in the NVIDIA's
libdrm, and this enables us to more easily support Qt using either
method, but does require some fixups.

The first patch is a backport from newer versions of Qt which will
allow the egldevice backend to function with the tegra-udrm. It is
not strictly necessary, but is nice because then users don't have
to worry about removing this driver i [... cut]
commit 47b14268 (2022-04-15): external/qt5-layer: fix egldevice backend for L4T 32.7.1

This change adds two patches to support tegra devices in qt5 with
the L4T release 32.7.1. The newer version of L4T appears to unify
the capabilities of tegra-udrm and drm-nvdc drivers in the NVIDIA's
libdrm, and this enables us to more easily support Qt using either
method, but does require some fixups.

The first patch is a backport from newer versions of Qt which will
allow the egldevice backend to function with the tegra-udrm. It is
not strictly necessary, but is nice because then users don't have
to worry about removing this driver  [... cut]
PR #960 (2022-05-06): external/qt5-layer: fix egldevice backend for L4T 32.7.1
This change adds two patches to support tegra devices in qt5 with
the L4T release 32.7.1. The newer version of L4T appears to unify
the capabilities of tegra-udrm and drm-nvdc drivers in the NVIDIA's
libdrm, and this enables us to more easily support Qt using either
method, but does require some fixups.

The first patch is a backport from newer versions of Qt which will
allow the egldevice backend to function with the tegra-udrm. It is
not strictly necessary, but is nice because then users don't have
to worry about removing this driver i [... cut]
commit 5d1a0b6e (2022-05-06): external/qt5-layer: fix egldevice backend for L4T 32.7.1

This change adds two patches to support tegra devices in qt5 with
the L4T release 32.7.1. The newer version of L4T appears to unify
the capabilities of tegra-udrm and drm-nvdc drivers in the NVIDIA's
libdrm, and this enables us to more easily support Qt using either
method, but does require some fixups.

The first patch is a backport from newer versions of Qt which will
allow the egldevice backend to function with the tegra-udrm. It is
not strictly necessary, but is nice because then users don't have
to worry about removing this driver  [... cut]
closed 2022-04-15 by madisongh