OE4T/meta-tegra #967: Add Propper ffmpeg

GavinDarkglider, 2022-04-22:
https://gitlab.com/switchroot/switch-l4t-multimedia/FFmpeg

add configure option: --enable-nvv4l

Either take patches, and rebase, or build this ffmpeg branch, and assuming your v4l2 multimedia libs/headers are installed you can build this, and it will force any app installed that uses system installed ffmpeg to use proper decoders/encoders.

You can also take these patches and build libffmpeg for chrome, but that requires dirty replacing the binary, and disabling chrome sandbox(VP9 decode doesnt work in chromium, as they use a seperate lib for that, use x264ify chrome plugin). Firefox has issues, and needs patches to work correctly with the decoder, as audio/video desyncs.

This has been tested in many distros, and bsps, and it is stable, also currently works with jetpack 5.x, but with the switch from nvbuf_utils to NVUtils, it might need a little work again.

madisongh, 2022-04-23:
That's the work that goes with the support for the hardware accelerator support in the mainline kernel, isn't it?  I'm not sure it will be compatible with the downstream L4T support.

NVIDIA does provide sources for ffmpeg for L4T, mentioned in [their documentation](https://docs.nvidia.com/jetson/l4t/Tegra%20Linux%20Driver%20Package%20Development%20Guide/multimedia.html#wwpID0E0JB0HA).  It should be possible to put together a recipe for building it from those sources.

GavinDarkglider, 2022-04-23:
> That's the work that goes with the support for the hardware accelerator support in the mainline kernel, isn't it? I'm not sure it will be compatible with the downstream L4T support.
>
> NVIDIA does [...]
We started with their nvv4l2_dec stuff, and CTCaer rewrote the whole implementation, and added a ton of support, increased encoder/decoder support by up to 190% in some cases(over example code, and what nvidia said was possible) This also has patches for moonlight support. This is also compatible without recompiling with all bsps 32.3.1-32.7.1 and should be good with 32.4 as well, until they completely remove nvbuf_utils.

closed 2024-03-24 by madisongh