OE4T/meta-tegra #998: Serial output for kernel boot on serial using wip-jetpack-5-dp

gsally-sarcos, 2022-06-13:
Using the wip-jetpack-5-dp branch, I'm not seeing output on the serial device (using 40 pin port, serial console works with NVidia jetpack image) past this from the bootloader: 

```
Jetson UEFI firmware (version r34.1-975eef6 built on 2022-05-16T20:58:45-07:00)
Press ESCAPE for boot options **  WARNING: Test Key is used.  **
......ExtLinuxBoot: Unable to load image: \boot\Image Not Found
L4TLauncher: Unable to boot via extlinux: Not Found
EFI stub: Booting Linux Kernel...
EFI stub: Using DTB from configuration table
EFI stub: ERROR: Failed to load initrd!
EFI stub: Exiting boot services and installing virtual address map...
```
In my local.conf, I've tried appending to the commandline via

```
APPEND += " console=ttyS0,115200n8"
```

But I'm not seeing this (or anything) on the linux command-line when I boot.

ichergui, 2022-06-13:
Hello @gsally-sarcos 
I assume that you are using `Jetson Xavier NX` right ?
if so, I didn't try it. I can do that later and keep you updated

gsally-sarcos, 2022-06-13:
@ichergui, 

In my local.conf, this is what I'm using for the machine: 

```
MACHINE ?= "jetson-agx-xavier-devkit"
```
This is booting, but the kernel command line is empty when I look at dmesg and /proc/cmdline.   Over in meta-tegra/conf/machine/jetson-agx-xavier-devkit.conf, I can see the parameters being set via 

```
KERNEL_ARGS ?= "mminit_loglevel=4 console=tty0 console=ttyTCU0,115200 fbcon=map:0"
```
I should be able to modify the kernel command line via the bootloader, the uefi bootloader with the 5.0 kit doesn't make that obvious. 

Thanks for your help!

kekiefer, 2022-06-13:
I've been playing with this today also on an AGX Xavier, and made the observation that when UEFI loads the kernel via extlinux, I've got console output (settings from `KERNEL_ARGS`, which are present in the `extlinux.conf`), but via the kernel partition I see no output, similarly. The UEFI menus have an option to set the boot args: `Device Manager -> NVIDIA Resource Configuration -> Append to Kernel Command Line`. Only when I manually put boot arguments here do I see console output when loading from the kernel partition.

ichergui, 2022-06-13:
Thanks @kekiefer for sharing this

gsally-sarcos, 2022-06-13:
@kekiefer, on your agx, what does /proc/cmdline report if no changes are made via uefi?

kekiefer, 2022-06-13:
It's blank.

With my build from tegra-demo-distro, I have to go into the menu an explictly tell it to boot from the kernel partition, since extlinux is default.

As an aside, with another build from a custom distro, it behaves differently in a few ways that I don't yet understand. Among them is that the extlinux boot fails, and it falls back to the kernel partition boot automatically, which is why I noticed this at all.

kekiefer, 2022-06-14:
@gsally-sarcos is your boot also falling back to the kernel partition rather than loading extlinux? A boot like this would print something like the following:

```
Jetson UEFI firmware (version r34.1-975eef6 built on 2022-05-16T20:58:45-07:00)
Press ESCAPE for boot options **  WARNING: Test Key is used.  **
......
      ExtLinuxBoot: Unable to load image: \boot\Image Not Found
L4TLauncher: Unable to boot via extlinux: Not Found
EFI stub: Booting Linux Kernel...
...
```

gsally-sarcos, 2022-06-14:
@kekiefer, good catch.  /proc/version shows kernel built May 22.  
This is what I'm seeing on the serial console:

```
Jetson UEFI firmware (version r34.1-975eef6 built on 2022-05-16T20:58:45-07:00)
Press ESCAPE for boot options **  WARNING: Test Key is used.  **
......ProcessExtLinuxConfig: Failed to open file: Not Found
L4TLauncher: Unable to process extlinux config: Not Found
EFI stub: Booting Linux Kernel...
EFI stub: Using DTB from configuration table
EFI stub: ERROR: Failed to load initrd!
EFI stub: Exiting boot services and installing virtual address map...
```
I loopback mounted the RFS (the .ext4 file created by yocto) and there was no extlinux.conf file, so I guessed it should belong in ```boot/extlinux``` directory.  I looked at my yocto build and found this file, which looked about right: 

```
MENU TITLE L4T boot options
TIMEOUT 30
DEFAULT primary
LABEL primary
	MENU LABEL primary
	LINUX /boot/Image
	APPEND root=/dev/mmcblk0p1 rootfstype=ext4 rw rootwait mminit_loglevel=4 console=tty0 console=ttyTCU0,115200 fbcon=map:0
```

When I did this, the error changed to 

```
......ExtLinuxBoot: Unable to load image: Image Not Found
L4TLauncher: Unable to boot via extlinux: Not Found
```

It looks like the file was found (as the ``` Failed to open file: Not Found``` went away, but it looks like the ```/boot/Image``` wasn't found, guessing this is looking at the root file system.  Trying to get verbose logging enabled, so I get some more hints as to what's happening.

kekiefer, 2022-06-14:
After some experimentation this morning it doesn't look like UEFI is able to read a symlink on ext4. If I hardlink the kernel or change the name in the extlinux, it loads it properly.

kekiefer, 2022-06-14:
Also, have a look at the recipe for `l4t-launcher-extlinux` and make sure this is included in your rootfs to install the extlinux config. Currently, the `l4t-launcher-extlinux` should be picked up by `TEGRA_ESSENTIAL_EXTRA_RDEPENDS` (which is included by default in `MACHINE_ESSENTIAL_EXTRA_RDEPENDS`).

gsally-sarcos, 2022-06-14:
@kekiefer, I appreciate your helpfulness, thank you! I changed the symlinks to hardlinks in /boot and the kernel built locally is booting.  No output on the serial console.  Added ```l4t-launcher-extlinux```, now I have my extlinux.conf file, and as noted the symlinks aren't being followed.  Not sure what I need to change to get a hardlink and not a sym-link 

USB keyboard no longer works with the kernel built on my host, not that important if I can get the serial communication working.

kekiefer, 2022-06-17:
@gsally-sarcos are you still having problems with this? With a5c03326510ee05843642d604e8134e9d12ebac1 the kernel recipe no longer clobbers the kernel image written by `l4t-launcher-extlinux` and the bootargs are picked up from the extlinux.conf.

gsally-sarcos, 2022-06-21:
@kekiefer, I can give this a try.  I created a bbappend that compiles this in as the command-line

```root=/dev/mmcblk0p1 rootfstype=ext4 rw rootwait   mminit_loglevel=4 console=tty0 console=ttyTCU0,115200 fbcon=map:0 ip=192.168.111.10:::255.255.255.0:jtsn:eth0:none```

and this did not result in any output on the serial port on the 40pin connector.  I can reach the device via the ssh, so this has enabled me to make progress.  

I'll pull and see if the change above makes any difference.

gsally-sarcos, 2022-06-30:
@kekiefer, after pulling and rebuilding, the /boot directory still has symlinks to the Image and Image.gz files (I mounted the generated ext4 via loopback); I'm doing some work on my board, so I haven't tried booting this image. 

I'm not so worried about the kernel parameters being clobbered; not having a working usb/serial is worrisome, but worked around by configuring address on command-line and using ssh.

gsally-sarcos, 2022-06-30:
@kekiefer, added PACKAGE_EXCLUDE:append to local.conf, clean build and the boot directory contained the correct links.  Just to make sure I didn't make a mistake, I commented the PACKAGE_EXCLUDE:append out, rebuilt cleanly, and the boot directory contained soft links.  

> If it's something softer (RRECOMMENDS) at least this should keep the offending packages out of your image.

Must be this issue then; I can try by pairing back to just the packagegroup-core-full-cmdline set of packages and go from there. 

Thanks for your help!

kekiefer, 2022-07-03:
OK, I'll attempt to wrangle this.

* Since we opened this issue, we've managed to uncover and fix an issue where `kernel-image-image` was getting included by default, and this was clobbering the kernel installed by `l4t-launcher-extlinux`.
* Other methods can pull this package in which are sort of outside the scope of what we can control here, which you have addressed for your own case.
* There remains a question as to how to get kernel bootargs passed to UEFI for the case where extlinux is not used. I believe this is still a valid use-case, since UEFI _can_ load the kernel from the boot partitions just like cboot could do.

It is this last issue which is still of some concern, so I'm leaving this issue open to address the final point.

gsally-sarcos, 2022-07-05:
@kekiefer, 
Sorry, I was wrong about the changes to local.conf having a difference (in my configuration) making a difference w/r/t the links created in the RFS /boot directory.   I have this line in my local.conf

`PACKAGE_EXCLUDE:append = " kernel-image kernel-image-image"
`

After a build, I look at the root file system and see this in the boot directory:

```
drwxr-xr-x  2 root root     4096 Mar  9  2018 extlinux
lrwxrwxrwx  1 root root       38 Mar  9  2018 Image -> Image-5.10.65-l4t-r34.dp+g4cfb4d6ec975
-rw-r--r--  1 root root 33010176 Mar  9  2018 Image-5.10.65-l4t-r34.dp+g4cfb4d6ec975
lrwxrwxrwx  1 root root       41 Mar  9  2018 Image.gz -> Image.gz-5.10.65-l4t-r34.dp+g4cfb4d6ec975
-rw-r--r--  1 root root 12602109 Mar  9  2018 Image.gz-5.10.65-l4t-r34.dp+g4cfb4d6ec975
```

I have a script that creates the proper links, so I can boot this kernel and make progress with getting the application running.  After crawling through the various log files and putting some distinctive strings in the kernel command line (that I can verify after boot) I'm convinced I'm seeing the kernel build from my project in this directory. 

I've reduced my recipe to  IMAGE_INSTALL containing only the following `packagegroup-core-full-cmdline, l4t-launcher-extlinux` and can still reproduce this problem, so I'm unsure on what other packages I need to remove that could be interfering with the links being created correctly.  

Would you be able to post what components were in the IMAGE_INSTALL used for verifying this issue was solved with a5c0332?  I will use those to see if I can duplicate a successful build.  To double check I'm running the same software, `git status` says

```
On branch wip-jetpack-5-dp
Your branch is up to date with 'origin/wip-jetpack-5-dp'.
```

and `git log -1 --oneline` reports

```
git log -1 --oneline
052a23ac (HEAD -> wip-jetpack-5-dp, origin/wip-jetpack-5-dp) cuda: update all CUDA packages to latest from JP5 deb feed
```

gsally-sarcos, 2022-07-05:
@kekiefer,
This was originally about the serial port on the 40 pin port not producing output.  That's still true for my configuration, I can interrupt the boot and use the console program, but nothing from the kernel on the same serial port.  The command line is

```
# cat /proc/cmdline 
root=/dev/mmcblk0p1 rootfstype=ext4 rw rootwait mminit_loglevel=4 console=tty0 console=ttyTCU0,115200 fbcon=map:0 ip=192.168.111.10:::255.255.255.0:jtsn:eth0:none 
```

The directly-attached usb keyboard is not working, so I get to the console via ssh over the eth0 port configured in the command-line.

tjnix, 2022-08-26:
I was having similar issues (i.e. `ExtLinuxBoot: Unable to load image: \boot\Image Not Found`) and https://github.com/OE4T/meta-tegra/commit/a5c03326510ee05843642d604e8134e9d12ebac1 fixed it for me.

closed 2023-07-15 by madisongh