OE4T/meta-tegra #999: kernel-image-* erroneously included in rootfs (kirkstone and later)

kekiefer, 2022-06-15:
As of commit https://github.com/openembedded/openembedded-core/commit/1c90b27d2c65cfb4f9debf0272820b6a95942f76 the kernel images are included in the rootfs, as the meta-tegra layer needs to be updated to either:

a) replace the linux-tegra recipe's existing fixup to `RDEPENDS` with a fixup to `RRECOMMENDS` instead
b) remove the fixup and add a `PACKAGE_EXCLUDE:append` to the tegra-common.inc

I've tested both approaches, and both will work. So this issue is to solicit feedback regarding which path makes more sense.

madisongh, 2022-06-15:
Option (a) looks better to me.   The PACKAGE_EXCLUDE approach looks more difficult to override for cases where someone _does_ want the kernel image included in the rootfs.

ichergui, 2022-06-15:
Hey @kekiefer 
I prefer option a. I think it is the appropriate way  as a fix 
Thanks for catching this :blush:

PR #1000 (2022-06-15): linux-tegra: update packaging RDEPENDS->RRECOMMENDS
Upstream kernel.bbclass changed the inclusion of kernel image
artifacts from RDEPENDS to RRECOMMENDS.

Update this recipe accordingly.

Fixes #999
commit a5c03326 (2022-06-15): linux-tegra: update packaging RDEPENDS->RRECOMMENDS

Upstream kernel.bbclass changed the inclusion of kernel image
artifacts from RDEPENDS to RRECOMMENDS.

Update this recipe accordingly.

Fixes #999
commit 1af84edd (2022-06-15): linux-tegra: update packaging RDEPENDS->RRECOMMENDS

Upstream kernel.bbclass changed the inclusion of kernel image
artifacts from RDEPENDS to RRECOMMENDS.

Update this recipe accordingly.

Fixes #999
closed 2022-06-15 by madisongh