OE4T/meta-tegra #1019: Orin PCIe hotfix

teebr, 2022-08-02:
**Describe the bug**

NVidia has acknowledged a bug with the Orin AGX device tree which means that the PCIe bus doesn't work. A [patch](https://forums.developer.nvidia.com/t/pcie-x16-supply-not-enabled-on-development-kit/214927/12) and [unofficial DTB](https://forums.developer.nvidia.com/t/pcie-card-not-working-and-not-getting-poweron/220747) are available.

I got it working in `tegra-demo-distro` in a hacky way (appending `l4t-launcher-extlinux` to include the DTB and then `sed`-ing `extlinux.conf` to add the FDT entry - I tried using `UBOOT_EXTLINUX_FDT` and `DTBFILE` but couldn't get it to work, though to be honest didn't spent a huge amount of time on it. Maybe using the patch is better anyway?

**To Reproduce**

Plug something into the PCIe bus -> `lspci` shows nothing much unless the patch is applied.

**Additional context**

It's not yet clear when this will be fixed - hopefully a minor revision to JP5 DP is in the works.

ichergui, 2022-08-04:
Hey @teebr 
I'm using `Jetson Orin AGX` and `tegra-demo-distro`, it is working for my side.
Could you please be explicit on your experience ?
What kind of hardware you are connecting to the `Jetson Orin AGX` ?

teebr, 2022-08-04:
In my case I'm connecting a GPU (a whole other world of pain, I know) to the PCIe bus. By default you won't see anything - `lspci` will just show something like :
```
0001:00:00.0 PCI bridge: NVIDIA Corporation Device 229e (rev a1)
0001:01:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8822CE 802.11ac PCIe Wireless Network Adapter
```

With the DTB update you should be able to see the specific device via `lspci`:
```
0001:00:00.0 PCI bridge: NVIDIA Corporation Device 229e (rev a1)
0001:01:00.0 Network controller: Realtek Semiconductor Co., Ltd. RTL8822CE 802.11ac PCIe Wireless Network Adapter
0005:00:00.0 PCI bridge: NVIDIA Corporation Device 229a (rev a1)
0005:01:00.0 <your device here>
```

ichergui, 2022-08-04:
I think that you have a power issue.
Is the FAN of the GPU spinning when you connect it with the root complex `Jetson Orin AGX` ?
Are you using an external power supply ? if not, sometime you need that.

teebr, 2022-08-04:
Yes to be clear the DTB fix relates to the power supply of the PCIe bus. We have tried a couple of different GPUs with and without external power and both only work with the fix mentioned above.

tjnix, 2022-11-21:
This is semi-related I think. I had to apply the two file changes from this forum [post ](https://forums.developer.nvidia.com/t/solution-tutorial-jetson-orin-enabling-pcie-power/222285) for PCIe to work on my Orin devkit. I looked at Nvidia's latest 5.10 kernel source and I don't see this fix but according to this [post](https://forums.developer.nvidia.com/t/jetson-agx-orin-board-not-able-to-enumerate-fpga-pcie-card-latest-kernel-35-1-0/226733/3) it is supposed to be fixed in 5.0.2.

teebr, 2022-11-22:
@tjnix yes - I saw that NVidia made some changes to the device tree around PCIe but I still had to use those same patches to get it to work.

closed 2023-05-03 by teebr