OE4T/meta-tegra #1046: devtool modify edk2-firmware-tegra fails with wip-jetpack-5-ga branch.

atharvanan1, 2022-08-26:
**Describe the bug**
`devtool modify edk2-firmware-tegra` fails with wip-jetpack-5-ga branch.

**To Reproduce**
Steps to reproduce the behavior:
1. Build `meta-tegra` branch 'wip-jetpack-5-ga' with `jetson-xavier-nx-devkit-emmc`
2. Set `PREFERRED_PROVIDER_virtual/bootloader = "edk2-firmware-tegra"`
3. Run with devtool argument `modify edk2-firmware-tegra`
5. See error log below:
```
oe@pop-os:~/build/tegra-demo-distro/build (wip-jetpack-5-ga)$ devtool modify edk2-firmware-tegra
NOTE: Starting bitbake server...
INFO: Creating workspace layer in /home/oe/build/tegra-demo-distro/build/workspace
NOTE: Reconnecting to bitbake server...
NOTE: Retrying server connection (#1)...
Loading cache: 100% |                                                                                           | ETA:  --:--:--
Loaded 0 entries from dependency cache.
Parsing recipes: 100% |##########################################################################################| Time: 0:00:17
Parsing of 2811 .bb files complete (0 cached, 2811 parsed). 4342 targets, 116 skipped, 0 masked, 0 errors.
Removing 1 recipes from the jetson_xavier_nx_devkit_emmc sysroot: 100% |#########################################| Time: 0:00:00
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "2.0.1"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-oe4t-linux"
MACHINE              = "jetson-xavier-nx-devkit-emmc"
DISTRO               = "tegrademo"
DISTRO_VERSION       = "4.1+snapshot-20220826"
TUNE_FEATURES        = "aarch64 armv8a crc"
TARGET_FPU           = ""
meta                 = "HEAD:21cfb871f7da604f1fab957d7747cb03b113b971"
meta-tegra           = "rootfs-ab-nx:ada84fa8932376178deb778ce83c745f16122a66"
meta-oe              
meta-python          
meta-networking      
meta-filesystems     = "HEAD:78fe37bb9c1456263df3a6ec450a486a2d8464db"
meta-virtualization  = "HEAD:884355eeedf0126c9cc7d5e704d4262feb753510"
meta-tegra-community = "HEAD:a81bb9319b4e4e0095fe30ec073bdc2e58dd44ad"
meta-tegra-support   
meta-demo-ci         
meta-tegrademo       
workspace            = "wip-jetpack-5-ga:ed60609e5802b0cfb8b86cb0e884bd44bbd0313b"

Initialising tasks: 100% |#######################################################################################| Time: 0:00:00
Sstate summary: Wanted 0 Local 0 Mirrors 0 Missed 0 Current 20 (0% match, 100% complete)
NOTE: Executing Tasks
ERROR: Applying patch '0002-Replace-libc-mem-calls-with-EDK2-defined-calls.patch' on target directory '/home/oe/build/tegra-demo-distro/build/tmp/work/jetson_xavier_nx_devkit_emmc-oe4t-linux/edk2-firmware-tegra/35.1.0-r0/devtooltmp-r4mdpsa5/workdir/edk2-tegra'
CmdError('sh -c \'PATCHFILE="0002-Replace-libc-mem-calls-with-EDK2-defined-calls.patch" git -c user.name="OpenEmbedded" -c user.email="oe.patch@oe" commit -F /tmp/tmpbwry16ib --author="Matt Madison <matt@madison.systems>" --date="Mon, 11 Apr 2022 03:13:25 -0700"\'', 0, 'stdout: On branch devtool
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
  (commit or discard the untracked or modified content in submodules)
\tmodified:   edk2-nvidia (modified content)

no changes added to commit (use "git add" and/or "git commit -a")

stderr: ')
ERROR: Logfile of failure stored in: /home/oe/build/tegra-demo-distro/build/tmp/work/jetson_xavier_nx_devkit_emmc-oe4t-linux/edk2-firmware-tegra/35.1.0-r0/devtooltmp-r4mdpsa5/temp/log.do_patch.675654
NOTE: Tasks Summary: Attempted 93 tasks of which 90 didn't need to be rerun and 1 failed.
ERROR: Extracting source for edk2-firmware-tegra failed
```

atharvanan1, 2022-08-26:
Just placing this here, I'll investigate what this is about.

madisongh, 2022-08-26:
Our source layout is a bit funky - two git repos next to each other - and the patch applies to just one of the two repos, so I set `S:task-patch` to point to the right one, since for everything else in the build, `S` has to point to the other one.  devtool is trying to convert the patches into git commits, and it probably can't handle that strangeness.

ichergui, 2023-08-28:
I checked with our branch `wip-l4t-r35.4.1`

I got an issue:

```
➜  build-testdistro git:(wip-l4t-r35.4.1)  (wip-l4t-r35.4.1)devtool modify edk2-firmware-tegra
NOTE: Starting bitbake server...
INFO: Creating workspace layer in /home/icg/projects/oe4t/tegra-demo-distro/build-testdistro/workspace
NOTE: Reconnecting to bitbake server...
NOTE: Retrying server connection (#1)... (21:38:47.866864)
NOTE: Reconnecting to bitbake server...
NOTE: Reconnecting to bitbake server...
NOTE: Retrying server connection (#1)... (21:38:47.866864)
NOTE: Retrying server connection (#1)... (21:38:47.866864)
NOTE: Starting bitbake server...
NOTE: Reconnecting to bitbake server...
NOTE: Retrying server connection (#1)... (21:38:48.544226)
Loading cache: 100% |                                                                                                                                                                        | ETA:  --:--:--
Loaded 0 entries from dependency cache.
Parsing recipes: 100% |#######################################################################################################################################################################| Time: 0:00:20
Parsing of 3011 .bb files complete (0 cached, 3011 parsed). 4884 targets, 130 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "2.4.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "aarch64-oe4t-linux"
MACHINE              = "jetson-agx-orin-devkit"
DISTRO               = "tegrademo"
DISTRO_VERSION       = "4.2+snapshot-b08a06a5bb4a862c00d8273d5c57fa847fdedb00"
TUNE_FEATURES        = "aarch64 armv8a crc"
TARGET_FPU           = ""
meta                 = "HEAD:b08a06a5bb4a862c00d8273d5c57fa847fdedb00"
meta-tegra           = "HEAD:7b3ff17eef73e0fedc95ab256f72f050633a039e"
meta-oe              
meta-python          
meta-networking      
meta-filesystems     = "HEAD:4bacc11b3a103435f480e3a63e8e06e98d650bc1"
meta-virtualization  = "HEAD:0e64f6382fea409c970eaa804f6043f1c5f109f6"
meta-tegra-community = "HEAD:37a3a31f309284df5dde528f39a28eebe885e130"
meta-tegra-support   
meta-demo-ci         
meta-tegrademo       
workspace            = "wip-l4t-r35.4.1:7aa23c1eb95af7e424d0641161f89c596f2adb20"

Initialising tasks: 100% |####################################################################################################################################################################| Time: 0:00:00
Sstate summary: Wanted 0 Local 0 Mirrors 0 Missed 0 Current 30 (0% match, 100% complete)
NOTE: Executing Tasks
ERROR: Applying patch '0002-Disable-outline-atomics-in-eqos-driver.patch' on target directory '/home/icg/projects/oe4t/tegra-demo-distro/build-testdistro/tmp/work/jetson_agx_orin_devkit-oe4t-linux/edk2-firmware-tegra/35.4.1/devtooltmp-q1wp2o0a/workdir/edk2-tegra/edk2/..'
CmdError('sh -c \'PATCHFILE="0002-Disable-outline-atomics-in-eqos-driver.patch" git -c user.name="OpenEmbedded" -c user.email="oe.patch@oe" commit -F /tmp/tmp7duvneef --author="Matt Madison <matt@madison.systems>" --date="Thu, 26 Jan 2023 03:33:12 -0800"\'', 0, 'stdout: On branch devtool
Changes not staged for commit:
  (use "git add <file>..." to update what will be committed)
  (use "git restore <file>..." to discard changes in working directory)
  (commit or discard the untracked or modified content in submodules)
\tmodified:   edk2-nvidia (modified content)

no changes added to commit (use "git add" and/or "git commit -a")

stderr: ')
ERROR: Logfile of failure stored in: /home/icg/projects/oe4t/tegra-demo-distro/build-testdistro/tmp/work/jetson_agx_orin_devkit-oe4t-linux/edk2-firmware-tegra/35.4.1/devtooltmp-q1wp2o0a/temp/log.do_patch.8532
NOTE: Tasks Summary: Attempted 103 tasks of which 100 didn't need to be rerun and 1 failed.
ERROR: Extracting source for edk2-firmware-tegra failed
➜  build-testdistro git:(wip-l4t-r35.4.1)  (wip-l4t-r35.4.1)

```

dwalkes, 2023-08-28:
Yea I'm not sure if there's going to be a quick fix to get devtool to play nicely with this setup. I just removed the patches before running devtool, then added them myself in the resulting work directory when I needed to modify this recipe.

quaresmajose, 2023-08-29:
With https://github.com/OE4T/meta-tegra/pull/1319/commits/8b94d0f8c3ec651c3dc67dd5e420e7926f79f6f9 from https://github.com/OE4T/meta-tegra/pull/1319 I can run `devtool modify edk2-firmware-tegra` and it works as expected. I am waiting for the [wip-l4t-r35.4.1](https://github.com/OE4T/meta-tegra/tree/wip-l4t-r35.4.1) merge and will re-submit just this in another PR.

quaresmajose, 2023-08-29:
I can also try doing a rabase on top of wip-l4t-r35.4.1 and open a PR for the wip branch.

ichergui, 2023-08-29:
> I can also try doing a rabase on top of wip-l4t-r35.4.1 and open a PR for the wip branch.

Please do

ichergui, 2023-08-29:
Thanks @quaresmajose

atharvanan1, 2023-09-04:
I've been away from the project. Let me know if this is resolved, I'll close this issue.

quaresmajose, 2023-09-04:
This issue was fixed and merged in https://github.com/OE4T/meta-tegra/pull/1352 so it can be closed.

closed 2023-09-04 by ichergui