OE4T/meta-tegra #1051: Update weston in kirkstone branch

jmlms, 2022-08-29:
**Describe the bug**
Build fails, because weston was updated to 10.0.2 in [poky's kirkstone branch](https://git.yoctoproject.org/poky/commit/?h=kirkstone&id=3e3e985aba7f4e0c34cfabac0790343821d0e55d) on the weekend:

> 2022-08-28 07:09:43 - ERROR    - ERROR: No recipes in default available for:
2022-08-28 07:09:43 - ERROR    - <...>/layers/meta-tegra/recipes-graphics/wayland/weston_10.0.1.bbappend

**To Reproduce**
Use latest kirkstone-poky.

**Additional context**
A cherrypick from https://github.com/OE4T/meta-tegra/commit/1f29a5a6f4ed0fbbd7ef38b27ecd81fe10e0552e might be enough to resolve this.

ichergui, 2022-08-29:
Hey @jmlms 
Thanks for reporting this. I will check that and fix the issue

madisongh, 2022-08-29:
Branch has been updated (d60193750012074a5726e879eeca117dc26e6951) with the update cherry-picked from master.

closed 2022-08-29 by madisongh

ichergui, 2022-08-29:
> Branch has been updated ([d601937](https://github.com/OE4T/meta-tegra/commit/d60193750012074a5726e879eeca117dc26e6951)) with the update cherry-picked from master.

Thanks @madisongh

jmlms, 2022-08-29:
@ichergui we are building a custom distro.
@madisongh thanks, it builds again :)