OE4T/meta-tegra #1054: Compiling python3-tensorrt for Xavier error on dunfell

przemoch-dev, 2022-09-02:
I'm trying to add python wrapper for tensorrt to dunfell latest release (R32.7.2/JP4.6.2).
During compile task I keep getting error as below

(just part of the log)
```
[2/12] /home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++  -Dtensorrt_EXPORTS -I/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/recipe-sysroot/usr/include/python3.8 -I/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/git/python/../include -I/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/git/python/include -I/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/git/python/docstrings -I/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/recipe-sysroot -march=armv8-a+crc -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0=/usr/src/debug/python3-tensorrt/8.2.1-r0                           -fdebug-prefix-map=/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/recipe-sysroot=                           -fdebug-prefix-map=/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden  -march=armv8-a+crc -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/recipe-sysroot -I/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/recipe-sysroot-native/usr/local/cuda-10.2/include -I=/usr/local/cuda-10.2/include -fvisibility=hidden -std=c++11 -Wno-deprecated-declarations -DNDEBUG -fPIC   -std=gnu++11 -MD -MT CMakeFiles/tensorrt.dir/src/parsers/pyCaffe.cpp.o -MF CMakeFiles/tensorrt.dir/src/parsers/pyCaffe.cpp.o.d -o CMakeFiles/tensorrt.dir/src/parsers/pyCaffe.cpp.o -c /home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/git/python/src/parsers/pyCaffe.cpp
FAILED: CMakeFiles/tensorrt.dir/src/parsers/pyCaffe.cpp.o 
/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++  -Dtensorrt_EXPORTS -I/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/recipe-sysroot/usr/include/python3.8 -I/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/git/python/../include -I/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/git/python/include -I/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/git/python/docstrings -I/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/recipe-sysroot -march=armv8-a+crc -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0=/usr/src/debug/python3-tensorrt/8.2.1-r0                           -fdebug-prefix-map=/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/recipe-sysroot=                           -fdebug-prefix-map=/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden  -march=armv8-a+crc -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/recipe-sysroot -I/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/recipe-sysroot-native/usr/local/cuda-10.2/include -I=/usr/local/cuda-10.2/include -fvisibility=hidden -std=c++11 -Wno-deprecated-declarations -DNDEBUG -fPIC   -std=gnu++11 -MD -MT CMakeFiles/tensorrt.dir/src/parsers/pyCaffe.cpp.o -MF CMakeFiles/tensorrt.dir/src/parsers/pyCaffe.cpp.o.d -o CMakeFiles/tensorrt.dir/src/parsers/pyCaffe.cpp.o -c /home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/git/python/src/parsers/pyCaffe.cpp
In file included from /home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/git/python/src/parsers/pyCaffe.cpp:20:0:
/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/git/python/include/utils.h: In function 'pybind11::function tensorrt::utils::getOverride(const T*, const string&, bool)':
/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/git/python/include/utils.h:114:33: error: 'get_override' is not a member of 'tensorrt::utils::py'
     py::function overload = py::get_override(self, overloadName.c_str());
                                 ^~~~~~~~~~~~
/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/git/python/include/utils.h:114:33: note: suggested alternative: 'get_overload'
     py::function overload = py::get_override(self, overloadName.c_str());
                                 ^~~~~~~~~~~~
                                 get_overload
[3/12] /home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/recipe-sysroot-native/usr/bin/aarch64-poky-linux/aarch64-poky-linux-g++  -Dtensorrt_EXPORTS -I/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/recipe-sysroot/usr/include/python3.8 -I/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/git/python/../include -I/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/git/python/include -I/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/git/python/docstrings -I/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/recipe-sysroot -march=armv8-a+crc -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/recipe-sysroot  -O2 -pipe -g -feliminate-unused-debug-types -fdebug-prefix-map=/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0=/usr/src/debug/python3-tensorrt/8.2.1-r0                           -fdebug-prefix-map=/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/recipe-sysroot=                           -fdebug-prefix-map=/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/recipe-sysroot-native=  -fvisibility-inlines-hidden  -march=armv8-a+crc -fstack-protector-strong  -D_FORTIFY_SOURCE=2 -Wformat -Wformat-security -Werror=format-security  --sysroot=/home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/pyth [... cut ...]  -fPIC   -std=gnu++11 -MD -MT CMakeFiles/tensorrt.dir/src/infer/pyPlugin.cpp.o -MF CMakeFiles/tensorrt.dir/src/infer/pyPlugin.cpp.o.d -o CMakeFiles/tensorrt.dir/src/infer/pyPlugin.cpp.o -c /home/ubuntu/przemoch/safeway-by-sternkraft/build/tmp/work/armv8a_tegra194-poky-linux/python3-tensorrt/8.2.1-r0/git/python/src/infer/pyPlugin.cpp
ninja: build stopped: subcommand failed.
WARNING: exit code 1 from a shell command.
```

I took recipe from external/meta-python.

Anyone has idea how to resolve it?

madisongh, 2022-09-02:
Hmm.  That problem was fixed with an updated patch [here](ca536648612ceccc68d1fdc0d1ba884ea2fdc1c3), but it looks like that change got lost in a later commit.  I'm not sure how that happened, but it's easy enough to fix.

przemoch-dev, 2022-09-02:
Hey, glad to hear that. I'll test it out tomorrow.
Thank you

przemoch-dev, 2022-09-03:
I can confirm. Works fine now.
Thanks

PR #1056 (2022-09-02): external/meta-python: restore missing patch for python3-tensorrt
Fixes #1054 by restoring a missing additional patch.
closed 2022-09-03 by przemoch-dev