OE4T/meta-tegra #1067: Mender install fails on TX2NX because of outdated TEGRA_BUPGEN_SPECS, is it possible to prevent this?

Botigli, 2022-09-06:
Hello,

We are using TX2NX SOMs on custom carrier board, our BSP is created with Yocto (dunfell) and based on meta-tegra.

Everything is working well since months, but recently we have trouble with a new batch of SOMs: it is possible to flash them but impossible to update via mender.
The mender install command fails with this message:

```
Error: missing entries for partitions: bpmp-fw-dtb, bootloader-dtb, VER, BCT, MB1_BCT, kernel-dtb

       for TNSPEC 3636-301-0001-D.0-1-0-jetson-xavier-nx-devkit-tx2-nx-mmcblk0p1

ERR: bootloader update failed

---------- end of script output

ERRO[0056] ArtifactInstall_Leave script failed: statescript: error executing 'ArtifactInstall_Leave_80_bl-update': 1 : exit status 1

Rolling back Artifact...

INFO[0056] Rolling back to the inactive partition (33).

ERRO[0056] statescript: error executing 'ArtifactInstall_Leave_80_bl-update': 1 : exit status 1
```

Thanks to previous discussions of this repository (like [here](https://github.com/OE4T/tegra-demo-distro/discussions/208) and [here](https://github.com/OE4T/tegra-demo-distro/issues/52)) we found that a change of Nvidia SOM PN was the culprit.
The `FAB` value had to be updated in our tools and should include 301 (it was originally 300).
Adding a line with fab=301 in our `jetson-xavier-nx-devkit-tx2-nx.conf` helps have successful mender update:
```
TEGRA_BUPGEN_SPECS ?= "fab=300;boardsku=0001 \
                       fab=301;boardsku=0001;boardrev="
``` 

This workaround is great for the time being, thanks for the community.
However, this way of working is pretty inconvenient because that means we would need to generate a new release each time their is a change in the FAB number.

The role of TEGRA_BUPGEN_SPECS for the mender update is far from being clear for us (what is mandatory, what is optional, ...).
Do you know if there is a way to be future-proof right now (like a wildcard or something like that), and avoid the need of checking the PN of each Nvidia SOM and maintain the TEGRA_BUPGEN_SPECS value?

Thank you!
Adrien

madisongh, 2022-09-06:
Fortunately, such part number changes are relatively infrequent.

> The role of TEGRA_BUPGEN_SPECS for the mender update is far from being clear for us (what is mandatory, what is optional, ...).

Bootloader updates are complicated for the Jetsons, unfortunately.  Which parameters (and their values) that might affect compatibility of the boot firmware across the different SOM revisions varies by model and isn't clearly documented.  NVIDIA  recently enhanced their bootloader update tools to do some wildcarding and compatible parameter values for updates... we could update the `tegra-boot-tools` package (which we use for applying bootloader updates) o do the same. It's a bit risky, since it's not a straightforward process to set up the compatibility specs.  Getting that wrong could result in rendering a device unbootable.

Adding compatibility specs support would reduce the frequency of having to update the specs in the machine config files, but there could still be times when a SOM revision comes out that *will* require new boot firmware (change of DRAM or eMMC parts, for instance).  Those are even less frequent occurrences, though, and usually NVIDIA publishes a PCN when they do that.

dwalkes, 2022-09-06:
A few related links in case you haven't found these already:

* NVIDIA documentation regarding [Bootloader Update Payload (BUP) generation](https://docs.nvidia.com/jetson/archives/r35.1/DeveloperGuide/text/SD/Bootloader/UpdateAndRedundancy.html#generating-the-bootloader-update-payload-bup), which is where these values are used today: 
* Where the magic happens in meta-tegra regarding BUP generation - in the [image_types_tegra class](https://github.com/OE4T/meta-tegra/blob/master/classes/image_types_tegra.bbclass)
* The documentation for [tegra-bootloader-update](https://github.com/OE4T/tegra-boot-tools/blob/master/doc/tegra-bootloader-update.md) in the [tegra-boot-tools](https://github.com/OE4T/tegra-boot-tools/) used as a replacement for `nv_update_engine`.

Botigli, 2022-09-06:
Ok that's clearer. Let's see when the next FAB change occur.
Thanks a lot for your (very) quick answers @madisongh and @dwalkes !

closed 2022-09-06 by Botigli