OE4T/meta-tegra #1069: Jetpack 5 demo-image-full screen blank, not recovered with hotplug or mouse/keyboard movement

dwalkes, 2022-09-06:
**Describe the bug**
Using Jetpack 5.0.2 release on master branch with `agx-xavier-devkit` and `demo-image-full` I've noticed the display shuts off after some period of time when plugged into the HDMI port.  I'm unable to get the display to work again through any combination I've tried of the following:
* Hotplugging the HDMI display
* Moving keyboard or mouse in an attempt to wakeup
* Unplug/replug keyboard or mouse
* `xset -display :0 s reset`

Example Xserver logs at https://pastebin.com/BE3ivN3e

**Workarounds**
* I'm able to recover with `systemctl restart display-manager`
* I'm not able to reproduce when I use these configuration commands to xset:
```
xset s off
xset -dpms
xset s noblank
```

I'm also not able to reproduce with the stock Jetpack 5.0.2 image from NVIDIA.  
The `xset q` output with the stock NVIDIA Jetpack 5.0.2 image can be found at https://pastebin.com/hsT9Waqv

**To Reproduce**
Steps to reproduce the behavior: Note - this appears to be environment specific since others have used this same configuration and have not reproduced.
1. Build `tegra-demo-distro` branch `master` with `MACHINE` based on `jetson-agx-xavier-devkit`
2. Build with bitbake argument `demo-image-base`
3. Deploy to hardware via tegraflash, with HDMI monitor and keyboard connected.  Keyboard is connected through USB3 hub.
4. Wait ~1/2 hour for the display to blank.

**Additional context**
See thread at https://gitter.im/OE4T/jetpack-5-test?at=6312666905ad4a3701d16628

madisongh, 2022-09-08:
I've tried reproducing this, but I do see my screen unblank on a keypress.  Here's the kernel log from the blanking:
```
Sep 08 21:23:04 jetson-agx-xavier-devkit kernel: tegradc 15200000.display: blank - powerdown
Sep 08 21:23:04 jetson-agx-xavier-devkit kernel: extcon-disp-state external-connection:disp-state: cable 48 state 0
Sep 08 21:23:04 jetson-agx-xavier-devkit kernel: Extcon AUX2(HDMI) disable
Sep 08 21:23:04 jetson-agx-xavier-devkit kernel: tegra_nvdisp_handle_pd_disable: Powergated Head1 pd
Sep 08 21:23:04 jetson-agx-xavier-devkit kernel: tegra_nvdisp_handle_pd_disable: Powergated Head0 pd
Sep 08 21:23:04 jetson-agx-xavier-devkit kernel: tegradc 15210000.display: blank - powerdown
Sep 08 21:23:04 jetson-agx-xavier-devkit kernel: tegradc 15220000.display: blank - powerdown
```
and then after the keypress, this shows up in the kernel log:
```
Sep 08 21:47:39 jetson-agx-xavier-devkit kernel: tegradc 15200000.display: blank - powerdown
Sep 08 21:47:39 jetson-agx-xavier-devkit kernel: tegradc 15200000.display: unblank
Sep 08 21:47:39 jetson-agx-xavier-devkit kernel: tegra_nvdisp_handle_pd_enable: Unpowergated Head0 pd
Sep 08 21:47:39 jetson-agx-xavier-devkit kernel: tegra_nvdisp_handle_pd_enable: Unpowergated Head1 pd
Sep 08 21:47:39 jetson-agx-xavier-devkit kernel: tegradc 15200000.display: hdmi: tmds rate:138000K prod-setting:prod_c_hdmi_111m_223m
Sep 08 21:47:39 jetson-agx-xavier-devkit kernel: tegradc 15200000.display: hdmi: get YCC quant from EDID.
Sep 08 21:47:39 jetson-agx-xavier-devkit kernel: extcon-disp-state external-connection:disp-state: cable 48 state 1
Sep 08 21:47:39 jetson-agx-xavier-devkit kernel: Extcon AUX2(HDMI) enable
Sep 08 21:47:39 jetson-agx-xavier-devkit kernel: tegradc 15200000.display: unblank
Sep 08 21:47:39 jetson-agx-xavier-devkit kernel: tegradc 15210000.display: blank - powerdown
Sep 08 21:47:39 jetson-agx-xavier-devkit kernel: tegradc 15220000.display: blank - powerdown
```

Maybe the problem is specific to your monitor or cabling setup? I have a GeChic LCD panel connected via HDMI.

dwalkes, 2022-09-09:
Here's what I see on blank:
```
Sep 09 02:58:17 jetson-agx-xavier-devkit kernel: tegradc 15200000.display: blank - powerdown
Sep 09 02:58:17 jetson-agx-xavier-devkit kernel: tegradc 15200000.display: unblank
Sep 09 02:58:17 jetson-agx-xavier-devkit kernel: tegra_nvdisp_handle_pd_enable: Unpowergated Head0 pd
Sep 09 02:58:17 jetson-agx-xavier-devkit kernel: tegra_nvdisp_handle_pd_enable: Unpowergated Head1 pd
Sep 09 02:58:17 jetson-agx-xavier-devkit kernel: tegradc 15200000.display: hdmi: tmds rate:148500K prod-setting:prod_c_hdmi_111m_223m
Sep 09 02:58:17 jetson-agx-xavier-devkit kernel: tegradc 15200000.display: hdmi: get YCC quant from EDID.
Sep 09 02:58:17 jetson-agx-xavier-devkit kernel: extcon-disp-state external-connection:disp-state: cable 48 state 1
Sep 09 02:58:17 jetson-agx-xavier-devkit kernel: Extcon AUX2(HDMI) enable
Sep 09 02:58:17 jetson-agx-xavier-devkit kernel: extcon-disp-state external-connection:disp-state: cable 53 state 1
Sep 09 02:58:17 jetson-agx-xavier-devkit kernel: Extcon HDMI: HPD enabled
Sep 09 02:58:17 jetson-agx-xavier-devkit kernel: tegradc 15200000.display: hdmi: plugged
Sep 09 02:58:17 jetson-agx-xavier-devkit kernel: tegradc 15200000.display: blank - powerdown
Sep 09 02:58:17 jetson-agx-xavier-devkit kernel: extcon-disp-state external-connection:disp-state: cable 48 state 0
Sep 09 02:58:17 jetson-agx-xavier-devkit kernel: Extcon AUX2(HDMI) disable
Sep 09 02:58:17 jetson-agx-xavier-devkit kernel: tegra_nvdisp_handle_pd_disable: Powergated Head1 pd
Sep 09 02:58:17 jetson-agx-xavier-devkit kernel: tegra_nvdisp_handle_pd_disable: Powergated Head0 pd
Sep 09 02:58:17 jetson-agx-xavier-devkit kernel: tegradc 15210000.display: blank - powerdown
Sep 09 02:58:17 jetson-agx-xavier-devkit kernel: tegradc 15220000.display: blank - powerdown
```

I don't see any messages on keypress after blank.  I have verified the keyboard is still sending keypresses with
```
hexdump /dev/input/by-path/platform-3610000.xhci-usb-0\:2.4\:1.0-event-kbd
```

> Maybe the problem is specific to your monitor or cabling setup? I have a GeChic LCD panel connected via HDMI.

Could be although it seems the behavior with stock jetpack is different, I haven't been able to reproduce with the same cable and monitor.  My monitor is an ONN 100002487.

dwalkes, 2022-09-09:
Here's a comparison of `xset -q` output between Jetpack 5.0.2 NVIDIA release and meta-tegra https://www.diffchecker.com/F4LrmLEE

kekiefer, 2022-09-09:
The apparent lack of DPMS support on the oe4t version, versus the stock Jetpack, is a rather curious difference here. 

Also I'm assuming that even though it looks also like Jetpack is configured not to sleep, you have confirmed that forcing it to go to sleep it will wake up properly.

dwalkes, 2022-09-09:
> Also I'm assuming that even though it looks also like Jetpack is configured not to sleep, you have confirmed that forcing it to go to sleep it will wake up properly.

For the stock Jetpack 5.0.2 case I haven't done anything other than leave it unattended and see the monitor blank, press the keyboard key and see the monitor refresh.

dwalkes, 2022-09-09:
Fixed the `xset -q` case for meta-tegra, I was x11 forwarding on the previous one.  New one at https://www.diffchecker.com/deC6rYeI

ichergui, 2023-08-27:
Hey @dwalkes 
Should we close this ticket ?

dwalkes, 2023-08-27:
Yes

closed 2023-08-27 by dwalkes