OE4T/meta-tegra #1071: Jetpack 5 device fails to boot after the 3rd reboot with Attempting Recovery Boot UEFI prompt

dwalkes, 2022-09-08:
**Describe the bug**
After booting Jetpack 5 on Xavier AGX and running several reboots I've noticed the device failing to boot with messages "Attempting Recovery Boot" in UEFI.

 Here's a comparison serial log capture between a successful boot on the left and the next boot which ends in recovery mode on the right: https://www.diffchecker.com/2CVaxtOB 

Attempted `UBOOT_EXTLINUX = "1"` in local.conf, noticed the same issue, see logs in https://pastebin.com/Uu5VTCxk

I've completed 100 reboots with stock NVIDIA Jetpack 5.0.2 on the same setup, so the problem appears to be meta-tegra specific.

See rolling reboot script at https://github.com/Trellis-Logic/linux-test-scripts

**To Reproduce**
Steps to reproduce the behavior:
1. Build `tegra-demo-distro` branch 'master' with `MACHINE` based on 'jetson-agx-xavier-devkit'
2. Build with bitbake argument 'demo-image-full'
3. Deploy to hardware with tegraflash
4. Reboot three times in a row from the console (or run rolling reboot script at https://github.com/Trellis-Logic/linux-test-scripts)

**Additional context**
See thread at https://gitter.im/OE4T/jetpack-5-test?at=631830a572ad51741fcb2508

dwalkes, 2022-09-08:
@ichergui here's the issue I mentioned this morning in the monthly call

dwalkes, 2022-09-08:
Next I'd like to compare the minicom logs on stock Jetpack 5.0.2 to the logs obtained from meta-tegra.
Discussed in the monthly meeting: Investigating UEFI variable manipulation on stock Jetpack 5.0.2 image would also be interesting.  Suspicion there's a variable we need to set to show the boot completed successfully, similar to what was required for nv_update_engine in Jetpack 4.6.

madisongh, 2022-09-08:
Looks like the root cause of this is that we are not including the `tegra-redundant-boot` package, which runs the `nv_update_verifier` service, in our builds by default.  That package *used* to be used only if you enabled A/B redundancy on your Jetson - it would be run after rebooting after a bootloader update to back-fill the other boot slot with the updated boot firmware.   With JetPack 5, that service now runs a different command, and one of the things it does is to update (via, ugh, `/dev/mem`) bits in the scratch register used by the bootloader to track boot state, as described in the [L4T docs](https://docs.nvidia.com/jetson/archives/r35.1/DeveloperGuide/text/SD/Bootloader/UpdateAndRedundancy.html).

So that means the `nv_update_verifier` service *must* be run on every boot, regardless of whether or not you've enabled A/B redundancy.  The bootloader now checks that boot state all the time, so the OS must inform it that the rootfs it booted from was OK to prevent it from trying to fall back to the recovery image (which we don't populate in our builds anyway).

dwalkes, 2022-09-08:
Thanks @madisongh confirming `IMAGE_INSTALL:append = tegra-redundant-boot` is a workaround for this, all good through 5 reboots so far.

closed 2022-09-08 by madisongh

dwalkes, 2022-09-09:
Thanks @madisongh for finding/fixing.  I've completed 100 reboots with this setup.

PR #1072 (2022-09-08): Add nv_update_verifier to all builds
Refactor the packaging the redundant-boot recipes, so we can include the `nv_update_verifier` service in all builds.

Fixes #1071