OE4T/meta-tegra #1077: Type-C l4t-usb-device-mode not work on wip-jetpack-5-ga with jetson-agx-orin-devkit

PeiMengxin, 2022-09-17:
**Describe the bug**

Type-C l4t-usb-device-mode not work on wip-jetpack-5-ga with jetson-agx-orin-devkit.

l4tbr0 has no ip with 192.168.55.1

Boot log has some info:
``` log
[   14.311657] rndis0: HOST MAC ce:0c:8e:8d:03:20
[   14.311812] rndis0: MAC f6:81:cf:c0:67:45
[   14.311918] Aquantia AQR113C 6810000.ethernet:00: No AQR phy_mode setting in DT
[   14.312365] usb0: HOST MAC a6:de:a7:3c:b2:22
[   14.312512] usb0: MAC 96:71:44:60:96:17
[   14.316421] l4tbr0: port 1(rndis0) entered blocking state
[   14.316619] l4tbr0: port 1(rndis0) entered disabled state
[  OK  ] Started DNS forwarder and DHCP server.
[   14.317224] device rndis0 entered promiscuous mode
[   14.317431] audit: type=1700 audit(1659949804.336:8): dev=rndis0 prom=256 old_prom=0 auid=429496725
[   14.318800] l4tbr0: port 2(usb0) entered blocking state
[   14.318969] l4tbr0: port 2(usb0) entered disabled state
[   14.319298] device usb0 entered promiscuous mode
```
**To Reproduce**

Steps to reproduce the behavior:
1. Build `tegra-demo-distro` branch `wip-jetpack-5-ga` with `MACHINE` based on `jetson-agx-orin-devkit`
2. Build with bitbake argument `demo-image-base`
3. Deploy to hardware via `sodo ./doflash.sh` with type-C port near 40pin
4. After flash success and device reboot, l4tbr0 can not get ip such as 192.168.55.1

**Additional context**

Someone reported a similar issue on the NVIDIA forums, and got an effective solution to the problem, refer to: https://forums.developer.nvidia.com/t/jetpack-5-0-2-jetson-agx-orin-developer-kit-one-type-c-not-working/224158/36

madisongh, 2022-09-17:
While the pinmux config might be part of the problem for Orin, I can also reproduce this issue on an AGX Xavier device, and the root cause is a missing driver.

Try adding
```
CORE_IMAGE_EXTRA_INSTALL = "kernel-module-ucsi-ccg"
```
to your local.conf to add the missing driver to your build.  That fixed it for my AGX Xavier.  If you still run into problems, then maybe it's the pinmux config issue.  I'll have a PR soon to fix both problems in master.

PeiMengxin, 2022-09-17:
Yes, I can get a desired network via l4tbr0 after install kernel-module-ucsi-ccg.
Thanks a lot for your answers @madisongh

PR #1078 (2022-09-17): l4t-usb-device-mode fixes for USB-C platforms
* Fixup for earlier commit which broke then libusgbx build
* Add RRECOMMENDS to l4t-usb-device mode to pull in a needed kernel module for USB-C platforms
* Pinmux/PMC config file changes, derived from a dev forum post

Fixes #1077
closed 2022-09-17 by madisongh