OE4T/meta-tegra #1080: Hello, could you help me solve the issue ? https://github.com/OE4T/linux-tegra-4.9/issues/37

h2cw2l, 2022-09-18:
https://github.com/OE4T/linux-tegra-4.9/issues/37

ichergui, 2022-09-18:
Hello @h2cw2l 
Could you please give us more details `repo`, `branch` and `machine` ? Did you try on the `L4T` stock and it is working as expected ?

h2cw2l, 2022-09-18:
> Hello @h2cw2l Could you please give us more details `repo`, `branch` and `machine` ? Did you try on the `L4T` stock and it is working as expected ?

Thanks very much for your reply. 

Have you see the details in issue  https://github.com/OE4T/linux-tegra-4.9/issues/37

If there still some details lacked, I will git it tomorrow.

madisongh, 2022-09-18:
As I mention [here](https://github.com/OE4T/linux-tegra-4.9/issues/37#issuecomment-1250266939), you're unlikely to find the help you're looking for here.

closed 2022-09-19 by h2cw2l