OE4T/meta-tegra #1102: Yocto Branch LTS with Jetson Nano (Meta-Tegra)

Sherry112, 2022-10-24:
Hi,
I have a small question. I am using a Jetson Nano that uses Yocto project for the OS.
I am using the Meta-Tegra layer for Jetson Nano. For Yocto, I am using the branch "Dunfell" with meta-poky.
I wanted to ask if Dunfell has the long term support or I should switch to a different branch that has better support.

Thanks for your help.
Regards

ichergui, 2022-10-24:
Hello @Sherry112 

`Denfull` branch is an LTS release. I recommend you to use the latest BSP which is part of the branch `kirkstone-l4t-r32.7.x`

Please let us know if you need anything

madisongh, 2022-10-24:
As mentioned on [this wiki page](https://github.com/OE4T/meta-tegra/wiki/Which-branch-should-I-use%3F), our branches are named based on the upstream OE-Core/Yocto Project branches.  It's a bit more complicated, since we also have to track NVIDIA's JetPack/Jetson Linux release.  We currently have two active `dunfell`-tracking branches, although the one that is based off L4T R32.6.1 will likely be the next to be retired/deprecated.

Since L4T R32.7.2 is the latest Jetson Linux release, and the R32.7.x series is the last to support the Jetson Nano hardware, updating to our `dunfell` branch is probably a good next step.  As @ichergui mentioned, `kirkstone` is the next upstream LTS branch, so if you can migrate your poky to that branch and meta-tegra to `kirkstone-l4t-r32.7.x`, that would be even better.

I'll convert this to a discussion, since it's not really raising an issue that needs fixing.