OE4T/meta-tegra #1107: Secure Boot build fails with tegra-bup-payload

roger652, 2022-10-27:
**Description**
Build fails on `dunfell` branch when secure boot is enabled and the image contains `tegra-bup-payload`:
```
| Error: Return value 6
| Command tegrarcm --chip 0x21 0 --updatesig rcm_list_signed.xml --pubkeyhash pub_key.key
| cp: cannot stat 'signed/*': No such file or directory
| Signing with /home/roger/yocto4/build-edward/rsa_priv.pem ...
| 
| signing images succeeded
| 
| mv: cannot stat '/home/roger/yocto4/build-edward/tmp/work/edward-poky-linux/u-boot-bup-payload/1.0-r0/bup-payload/signed/boot.img.*': No such file or directory
| mv: cannot stat '/home/roger/yocto4/build-edward/tmp/work/edward-poky-linux/u-boot-bup-payload/1.0-r0/bup-payload/signed/*.dtb*': No such file or directory
| mv: cannot stat '/home/roger/yocto4/build-edward/tmp/work/edward-poky-linux/u-boot-bup-payload/1.0-r0/bup-payload/signed/*bct*': No such file or directory
```

**Reproduce**
Steps to reproduce the behavior:
1. Set `TEGRA_SIGNING_ARGS = " -u path/to/private_key.pem "` in `local.conf`
2. Add `IMAGE_INSTALL_append = " tegra-bup-payload"` in `local.conf`
3. Build `meta-tegra` branch `dunfell` with `MACHINE` based on `jetson-nano-devkit-emmc`

**Additional context**
The secure boot build works when `tegra-bup-payload` is removed.

madisongh, 2022-10-27:
I just tried a test build for a Nano-eMMC with TEGRA_SIGNING_ARGS set similarly, and it worked fine.  Can you provide more of the `log.do_deploy` from the `u-boot-bup-payload` build?  There are probably some errors before the lines in the snippet you included that will tell us more about what's gone wrong.

roger652, 2022-10-28:
Thanks for taking a look. Here is the full do_deploy log:
[log.do_deploy.txt](https://github.com/OE4T/meta-tegra/files/9887474/log.do_deploy.txt)

madisongh, 2022-10-28:
Thanks.   That definitely looks weird - the signing of the RCM messages is failing, with `tegrarcm` exiting with an error code of 6.  I have no idea why, though.  It might help to get run the sequence of commands manually, with the failing command running under `strace`, to see what might be causing that error:
```
$ cd $BUILDDIR/tmp/work/edward-poky-linux/u-boot-bup-payload/1.0-r0/bup-payload
$ mkdir xxx; cd xxx
$ ../tegrasign --getmode mode.txt --key $BUILDDIR/rsa_priv.pem
$ ../tegrarcm --listrcm rcm_list.xml --chip 0x21 0 --download rcm nvtboot_recovery.bin 0 0
$ ../tegrasign --key $BUILDDIR/rsa_priv.pem --list rcm_list.xml --pubkeyhash pub_key.key
$ strace -ff -s256 -o rcmtrace ../tegrarcm --chip 0x21 0 --updatesig rcm_list_signed.xml --pubkeyhash pub_key.key
```
That should generate a file called `rcmtrace.<pid>` where `<pid>` is the process ID number of the process that got created to run the `tegrarcm` program.  Check exit status at the end, and if it's non-zero, see if it's one of the system calls in the trace returned that error.

madisongh, 2022-10-28:
I see the public key file is 384 bytes long.  Are you trying to use a 3072-bit (3K) RSA key?  I don't think that will work with the Nanos (or TX2s) - you have to use a 2048-bit key.

roger652, 2022-10-28:
That's probably it. I generated a 3072-bit RSA key.
I will retest it next week with a 2048-bit key.

roger652, 2022-11-03:
I confirm the issue was the 3072-bit RSA key.
It works with the 2048-bit key.

Thanks @madisongh

closed 2022-11-03 by roger652