OE4T/meta-tegra #1108: [kirkstone] Build error: Header already present for mb1_t194_prod.bin

elbbit01, 2022-10-28:
Hello,
we have a running project based on your dunfell meta-tegra branch using nvidia jetson agx. Everything is working fine but we need some features from the Jetpack 5.0.2 so we are testing for updating to kirkstone. It does not build, failing in do_image of the initramfs. More specificaly it has problems with the header for mb1_t194_prod.bin. Trying with the demo distro, it builds but has a similar failure in the flashing. Deeper description follows.

I made a simple setup with poky and meta-tegra (**kirkstone**). In conf/local conf I set the following:

```

MACHINE ?= **"jetson-agx-xavier-devkit"**
DISTRO_FEATURES = "x11 opengl "
IMAGE_CLASSES += "image_types_tegra"
IMAGE_FSTYPES = "tegraflash"
```

And I get the following error in the end of the build:

```
[...]

 SUCCESS: kernel_only_payload created
| 
| Payloads saved to "/home/...build/tmp/work/jetson_agx_xavier_devkit-poky-linux/tegra-minimal-initramfs/1.0-r0/bup-payload/payloads_t19x/"
| 
| Signing with zerosbk ...
| rollback info is parsed and saved in rollback.bin
| pad ZEROs to the end of rollback.bin
| 
| Generating signed binaries for 2888-402-0005--1-2-jetson-agx-xavier-devkit-mmcblk0p1 ...
| python3 tegraflash.py --chip 0x19 --bl nvtboot_recovery_cpu_t194.bin --sdram_config tegra194-mb1-bct-memcfg-p2888.cfg,tegra194-memcfg-sw-override.cfg --odmdata 0x9190000 --bldtb tegra194-p2888-0001-p2822-0000.dtb --applet mb1_t194_prod.bin --soft_fuses tegra194-mb1-soft-fuses-l4t.cfg --cmd "sign" --skipuid --cfg flash.xml --boot_chain A --bct_backup --secondary_gpt_backup --uphy_config tegra194-mb1-uphy-lane-p2888-0000-p2822-0000.cfg --minratchet_config tegra194-mb1-bct-ratchet-p2888-0000-p2822-0000.cfg --device_config tegra19x-mb1-bct-device-sdmmc.cfg --misc_config tegra194-mb1-bct-misc-flash.cfg --misc_cold_boot_config tegra194-mb1-bct-misc-l4t.cfg --pinmux_config tegra19x-mb1-pinmux-p2888-0000-a04-p2822-0000-b01.cfg --gpioint_config tegra194-mb1-bct-gpioint-p2888-0000-p2822-0000.cfg --pmic_config tegra194-mb1-bct-pmic-p2888-0001-a04-E-0-p2822-0000.cfg --pmc_config tegra19x-mb1-padvoltage-p2888-0000-a00-p2822-0000-a00.cfg --prod_config tegra19x-mb1-prod-p2888-0000-p2822-0000.cfg --scr_config tegra194-mb1-bct-scr-cbb-mini.cfg --scr_cold_boot_config tegra194-mb1-bct-scr-cbb-mini.cfg --br_cmd_config tegra194-mb1-bct-reset-p2888-0000-p2822-0000.cfg --dev_params tegra194-br-bct-sdmmc.cfg,tegra194-br-bct_b-sdmmc.cfg --overlay_dtb L4TConfiguration.dtbo,L4TRootfsInfo.dtbo,tegra194-p2822-camera-dual-imx274-overlay.dtbo,tegra194-p2822-camera-e3331-overlay.dtbo,tegra194-p2822-camera-e3333-overlay.dtbo,tegra194-p2822-camera-imx185-overlay.dtbo,tegra194-p2822-camera-imx390-overlay.dtbo,tegra194-p2888-0005-overlay.dtbo,tegra194-p2888-0001-p2822-0000-overlay.dtbo --bins "mb2_bootloader nvtboot_recovery_t194.bin; mts_preboot preboot_c10_prod_cr.bin; mts_mce mce_c10_prod_cr.bin; mts_proper mts_c10_prod_cr.bin; bpmp_fw bpmp-2_t194.bin; bpmp_fw_dtb tegra194-a02-bpmp-p2888-0005-a04-maxn.dtb; spe_fw spe_t194.bin; tos tos-optee_t194.img; eks eks.img; bootloader_dtb tegra194-p2888-0001-p2822-0000.dtb"
| Welcome to Tegra Flash
| version 1.0.0
| Type ? or help for help and q or quit to exit
| Use ! to execute system commands
| 
| [   0.0260 ] Using default ramcode: 0
| [   0.0260 ] Disable BPMP dtb trim, using default dtb
| [   0.0260 ]
| [   0.0325 ] tegrasign_v3.py --key None --getmode mode.txt
| [   0.0332 ] Assuming zero filled SBK key
| [   0.0360 ] Generating RCM messages
| [   0.0395 ] tegrahost_v2 --chip 0x19 0 --magicid MB1B --appendsigheader mb1_t194_prod.bin zerosbk
| [   0.0423 ] Header already present for mb1_t194_prod.bin
| Error: Return value -9
| Command tegrahost_v2 --chip 0x19 0 --magicid MB1B --appendsigheader mb1_t194_prod.bin zerosbk
| WARNING: exit code 1 from a shell command.
ERROR: Task (/home/.../meta-tegra/recipes-core/images/tegra-minimal-initramfs.bb:do_image_cpio) failed with exit code '1'

```

-------------------------------------------------------------------------------------

Using your **demo distro**, I can build demo-image-base, but when flashing my device (with **doflash.sh**), I get somehow a similar error:
```
$ ./doflash.sh
Welcome to Tegra Flash
version 1.0.0
Type ? or help for help and q or quit to exit
Use ! to execute system commands

[   0.0255 ] Generating RCM messages
[   0.0286 ] tegrahost_v2 --chip 0x19 0 --magicid MB1B --appendsigheader mb1_t194_prod.bin zerosbk
[   0.0303 ] Header already present for mb1_t194_prod.bin
Error: Return value -9
Command tegrahost_v2 --chip 0x19 0 --magicid MB1B --appendsigheader mb1_t194_prod.bin zerosbk
ERR: could not retrieve EEPROM board information
```
---------------------------------------------------------------------------------------------------

Am I missing something?

Could somebody please give me a hint?

Thanks in advance,
Alvaro.

ichergui, 2022-10-28:
Hi @elbbit01 
Make sure that your device is in recovery mode then run doflash.sh with sudo when using tegra-demo-distro

elbbit01, 2022-10-28:
Hi @ichergui, 
yes. of course.
Actually, with my "old" build with dunfell branch it is flashing properly.

madisongh, 2022-10-28:
That "Header already present" message is normal.  The error return code of `-9` from the `tegrahost_v2` is the actual error, but it looks like the tool isn't providing any message to indicate what that return code means.

What OS are you running on your build host?   Could be something environmental.  Running the `doflash.sh` script under strace, then tracking down the strace dump for the failing process, could give us a pointer as to what's going on.

madisongh, 2022-10-28:
Glad you found the cause.

closed 2022-10-28 by madisongh

ichergui, 2022-10-28:
Glad that it is working fine now :smile: