OE4T/meta-tegra #1110: Default orin machine (jetson-agx-orin-devkit) hangs on S20nv_update_verifier

frosteyes, 2022-11-07:
Have just tested jetson-agx-orin-devkit in combination with latest kirkstone of meta-tegra and poky.

It hangs on S20nv_update_verifier init script, from tegra-redundant-boot

The problem is that it waits on a partition label not available on the default orin flash image.

From the script

```
root@jetson-agx-orin-devkit:/etc/rc5.d# cat /etc/init.d/nv_update_verifier 
#! /bin/sh
NAME=nv_update_verifier
DESC="OTA bootloader update verifier"

case "$1" in
  start|restart)
      while [ ! -e /dev/disk/by-partlabel/kernel ]; do sleep 1; done
      sleep 2
      /usr/sbin/nvbootctrl verify < /dev/null
      ;;
...
```

And the default partition labels are

```
root@jetson-agx-orin-devkit:/dev/disk/by-partlabel# ls -la
drwxr-xr-x    2 root     root           260 Jan  1  1970 .
drwxr-xr-x    7 root     root           140 Jan  1  1970 ..
lrwxrwxrwx    1 root     root            15 Jan  1  1970 APP -> ../../mmcblk0p1
lrwxrwxrwx    1 root     root            15 Jan  1  1970 A_kernel -> ../../mmcblk0p2
lrwxrwxrwx    1 root     root            15 Jan  1  1970 A_kernel-dtb -> ../../mmcblk0p3
lrwxrwxrwx    1 root     root            15 Jan  1  1970 A_reserved_on_user -> ../../mmcblk0p4
lrwxrwxrwx    1 root     root            15 Jan  1  1970 B_kernel -> ../../mmcblk0p5
lrwxrwxrwx    1 root     root            15 Jan  1  1970 B_kernel-dtb -> ../../mmcblk0p6
lrwxrwxrwx    1 root     root            15 Jan  1  1970 B_reserved_on_user -> ../../mmcblk0p7
lrwxrwxrwx    1 root     root            16 Jan  1  1970 esp -> ../../mmcblk0p10
lrwxrwxrwx    1 root     root            15 Jan  1  1970 recovery -> ../../mmcblk0p8
lrwxrwxrwx    1 root     root            15 Jan  1  1970 recovery-dtb -> ../../mmcblk0p9
lrwxrwxrwx    1 root     root            16 Jan  1  1970 reserved -> ../../mmcblk0p11
```

So it will hang on this init script forever.

dwalkes, 2022-11-07:
Related follow-up discussion at https://gitter.im/OE4T/community?at=63698045ff5546644b34691a

madisongh, 2022-11-08:
Yeah, I haven't tested a sysvinit-based build recently.  I'm trying one now, and will address this and any other issues I find.

frosteyes, 2022-11-08:
Fantastic work @madisongh  big thanks.

closed 2022-11-08 by madisongh

madisongh, 2022-11-08:
Fixes are in now (and cherry-picked to kirkstone and langdale).  No hangs or failures now, but there may well be other, less-obvious, issues with sysvinit-based startups that I didn't see.  Patches from folks that actually use sysvinit instead of systemd to address those issues would be welcome!

PR #1111 (2022-11-08): Fixes for sysvinit-based startup issues
* Fix the installation of the optee supplicant init script to match the INITSCRIPT_NAME
* Drop the wait on the kernel partition for `nv_update_verifier`

Fixes #1110