OE4T/meta-tegra #1160: Issues Flashing to Jetson AGX Xavier

plyons72, 2023-02-03:
**Describe the bug**
We are attempting to build an image for the Jetson AGX Xavier Industrial with yocto. We have successfully used meta-tegra to build and flash on another board with a Jetson AGX Xavier. When we attempt to flash (running `sudo ./doflash.sh`), we get the following error:

```
Welcome to Tegra Flash
version 1.0.0
Type ? or help for help and q or quit to exit
Use ! to execute system commands
 
[   0.0033 ] Generating RCM messages
[   0.0050 ] tegrahost_v2 --chip 0x19 0 --magicid MB1B --appendsigheader mb1_t194_prod.bin zerosbk
[   0.0055 ] Header already present for mb1_t194_prod.bin
[   0.0097 ] tegrasign_v3.py --key None --getmode mode.txt
[   0.0098 ] Assuming zero filled SBK key
[   0.0115 ] tegrasign_v3.py --file mb1_t194_prod_sigheader.bin --key None --length 1136 --offset 2960 --pubkeyhash pub_key.key
[   0.0116 ] Assuming zero filled SBK key
[   0.0137 ] Warning: pub_key.key is not found
[   0.0151 ] tegrahost_v2 --chip 0x19 0 --updatesigheader mb1_t194_prod_sigheader.bin mb1_t194_prod_sigheader.hash zerosbk
[   0.0200 ] tegrabct_v2 --chip 0x19 0 --sfuse tegra194-mb1-soft-fuses-l4t.cfg.pdf sfuse.bin
[   0.0222 ] tegrarcm_v2 --listrcm rcm_list.xml --chip 0x19 0 --sfuses sfuse.bin --download rcm mb1_t194_prod_sigheader.bin 0 0
[   0.0227 ] RCM 0 is saved as rcm_0.rcm
[   0.0248 ] RCM 1 is saved as rcm_1.rcm
[   0.0250 ] RCM 2 is saved as rcm_2.rcm
[   0.0250 ] List of rcm files are saved in rcm_list.xml
[   0.0250 ] 
[   0.0250 ] Signing RCM messages
[   0.0273 ] tegrasign_v3.py --key None --list rcm_list.xml --getmontgomeryvalues montgomery.bin --pubkeyhash pub_key.key
[   0.0274 ] Assuming zero filled SBK key
[   0.0302 ] Warning: pub_key.key is not found
[   0.0298 ] Copying signature to RCM mesages
[   0.0313 ] tegrarcm_v2 --chip 0x19 0 --updatesig rcm_list_signed.xml
[   0.0321 ] Boot Rom communication
[   0.0338 ] tegrarcm_v2 --chip 0x19 0 --rcm rcm_list_signed.xml --skipuid
[   0.0343 ] RCM version 0X190001
[   0.0351 ] Boot Rom communication completed
[   2.0628 ] tegrarcm_v2 --isapplet
[   2.0644 ] Applet version 01.00.0000
[   2.0704 ] tegrarcm_v2 --ismb2
[   2.0767 ] tegrahost_v2 --chip 0x19 --align nvtboot_applet_t194_aligned.bin
[   2.0788 ] tegrahost_v2 --chip 0x19 0 --magicid PLDT --appendsigheader nvtboot_applet_t194_aligned.bin zerosbk
[   2.0793 ] adding BCH for nvtboot_applet_t194_aligned.bin
[   2.0876 ] tegrasign_v3.py --key None --list nvtboot_applet_t194_aligned_sigheader.bin_list.xml --pubkeyhash pub_key.key
[   2.0879 ] Assuming zero filled SBK key
[   2.0910 ] Warning: pub_key.key is not found
[   2.0933 ] tegrahost_v2 --chip 0x19 0 --updatesigheader nvtboot_applet_t194_aligned_sigheader.bin.encrypt nvtboot_applet_t194_aligned_sigheader.bin.hash zerosbk
[   2.1008 ] tegrarcm_v2 --download mb2 nvtboot_applet_t194_sigheader.bin.encrypt
[   2.1021 ] Applet version 01.00.0000
[   2.1045 ] Sending mb2
[   2.1046 ] [................................................] 100%
[   2.1251 ] tegrarcm_v2 --boot recovery
[   2.1256 ] Applet version 01.00.0000
[   3.1365 ] tegrarcm_v2 --isapplet
[   3.1445 ] tegrarcm_v2 --ismb2
[   3.1449 ] MB2 Applet version 01.00.0000
[   3.1490 ] tegrarcm_v2 --ismb2
[   3.1494 ] MB2 Applet version 01.00.0000
[   3.1538 ] Retrieving board information
[   3.1554 ] tegrarcm_v2 --oem platformdetails chip chip_info.bin
[   3.1559 ] MB2 Applet version 01.00.0000
[   3.1607 ] Saved platform info in chip_info.bin
[   3.1650 ] Chip minor revision: 2
[   3.1650 ] Bootrom revision: 0xf
[   3.1650 ] Ram code: 0x1
[   3.1651 ] Chip sku: 0x99
[   3.1651 ] Chip Sample: non es
[   3.1661 ] Retrieving EEPROM data
[   3.1661 ] tegrarcm_v2 --oem platformdetails eeprom cvm ~/cvm.bin.sUbMo
[   3.1670 ] MB2 Applet version 01.00.0000
[   3.1718 ] Saved platform info in ~/cvm.bin.sUbMo
[   3.2130 ] Rebooting to recovery mode
[   3.2149 ] tegrarcm_v2 --ismb2
[   3.2154 ] MB2 Applet version 01.00.0000
[   3.2178 ] Rebooting to recovery mode
[   3.2197 ] tegrarcm_v2 --reboot recovery
[   3.2202 ] MB2 Applet version 01.00.0000
ERR: unrecognized board version 600
```

It seems the `tegra194-flash-helper.sh` script may just need to be updated with newer version numbers, but we cant say for sure. 

**To Reproduce**
Steps to reproduce the behavior:
1. Build `meta-tegra` branch 'kirkstone' with `MACHINE` based on 'jetson-agx-xavier-devkit'
2. Build with bitbake argument 'jetson-min'
3. Deploy to hardware with method 'doflash.sh'

madisongh, 2023-02-03:
Yes, unfortunately, there is no support for JAXI in the layer, mainly because none of us has one to test with.  Adding it would mainly involve:
* creating a `jetson-agx-xavier-industrial.conf` for it in `conf/machine`
* adding a `flashvars` file for it (for the `tegra-flashvars` recipe)
* updating the `tegra194-flash-helper.sh` as needed

The first two are largely derived from the corresponding `.conf` file in the L4T kit, and the third bit is derived from the `process_board` function in that `.conf` file (and/or the common config file(s) it includes).

plyons72, 2023-02-06:
Thank you for your response. I may try to assist with this while I have one at my disposal. If you already have any drafts of these changes lying around waiting to be tested on hardware, I'd appreciate you sharing them, but I'll mark this as closed regardless.

closed 2023-02-06 by plyons72