OE4T/meta-tegra #1172: wrong data image filename with secure boot enabled

hesmar, 2023-02-13:
Hi, when building an image with mender and secure boot enabled, flashing is not working correctly because the filename of the data image is wrong.  The resulting tegraflash directory contains a data image file `<IMAGE_DESCRIPTOR>.dataimg` but the UDA partition defined inside the `secureflash.yml` references the file `<IMAGE_DESCRIPTOR>.dataimg.img`. 

```
...
        <partition name="UDA" type="data">
            <allocation_policy> sequential </allocation_policy>
            <filesystem_type> basic </filesystem_type>
            <size> 2097152 </size>
            <allocation_attribute> 0x808 </allocation_attribute>
            <filename> <IMAGE_DESCRIPTOR>.dataimg.img </filename>
        </partition>
...
```

So the `.img` suffix is missing in the filename. The flash tool simply skips that partition. Adapting the filename to `<IMAGE_DESCRIPTOR>.dataimg.img` works.

I am on kirkstone-r32.7.x.

madisongh, 2023-02-13:
Which Jetson are you building for?

hesmar, 2023-02-13:
1. I build for Jetson Nano emmc
2. I use the TEGRA_SIGNING_ARGS variable for signing during the build.

It seems the issue is in the `image_types_tegra.bbclass`:

```
tegraflash_post_sign_pkg() {
    mksparse -b ${TEGRA_BLBLOCKSIZE} --fillpattern=0 "${IMAGE_TEGRAFLASH_ROOTFS}" ${IMAGE_BASENAME}.img
    cp secureflash.xml initrd-secureflash.xml
    sed -i -e"s,APPFILE,${IMAGE_BASENAME}.img," secureflash.xml
    if [ -n "${IMAGE_TEGRAFLASH_DATA}" -a -n "${DATAFILE}" ]; then
        sed -i -e"s,DATAFILE,${DATAFILE}.img," secureflash.xml
    fi
    cat > .presigning-vars <<EOF
${@'\n'.join(d.getVar("TEGRA_SIGNING_ENV").split())}
EOF
}
```

Here it generates the secureflash.xml and appends the `.img` suffix to the DATAFILE variable but the filename of the data image is without the `.img` file extension.

madisongh, 2023-02-13:
Yep, that looks like the problem.  The `mksparse` command that used to be in that function got lost when I refactored the per-SoC post-sign functions into a single common function.

madisongh, 2023-02-13:
Fix has been merged, thanks

PR #1173 (2023-02-13): Restore mksparse in common post-sign function
Fixes #1172
closed 2023-02-13 by madisongh