OE4T/meta-tegra #1177: install command broke in "tegra-uefi-prebuilt_35.2.1.bb"

CHNtentes, 2023-02-16:
I'm using Orin NX 16GB on a custom board, with kirkstone branch of meta-tegra.

When I tried `bitbake tegra-uefi-prebuilt`, it returned an error, which said:
> install: missing destination file operand after '/home/zenghuipeng/LTG/build/tmp/work/p3509_a02_p3767_0000-poky-linux/tegra-uefi-prebuilt/35.2.1-r0/image/EFI/BOOT/bootaa64.efi'

Then I went check the recipe and found line 16 seemed to lack a parameter:
`install -m 0644  ${D}${EFIDIR}/${EFI_BOOT_IMAGE}`

Compared to other install commands, like line 49:
`install -m 0644 ${B}/BOOTAA64.efi ${D}${EFIDIR}/${EFI_BOOT_IMAGE}`

CHNtentes, 2023-02-16:
And log is here:
[log.do_compile.txt](https://github.com/OE4T/meta-tegra/files/10752722/log.do_compile.txt)

madisongh, 2023-02-16:
Yeah, that's a stray that should be removed.  Since we switched to compiling UEFI from sources by default, this recipe doesn't get as much exercise as it once did.  Thanks.

PR #1178 (2023-02-16): tegra-uefi-prebuilt: fix typo in do_compile
Fixes #1177
closed 2023-02-16 by madisongh