OE4T/meta-tegra #1190: Outdated patches for qtbase 5.15.7

VitaliiBlagodir, 2023-03-02:
**Describe the bug**
Build for `tegra-demo-distro` fails with the latest `meta-qt` layer. It is caused by outdated patches in https://github.com/OE4T/meta-tegra/tree/kirkstone-l4t-r32.7.x/external/qt5-layer/recipes-qt/qt5

**To Reproduce**
Steps to reproduce the behavior:
1. Checkout `tegra-demo-distro` branch `kirkstone-l4t-r32.7.x` commit `96469b6aeec6aa863eaa4da8e045acd627093109`
2. Add layer `meta-qt` layer branch `kirkstone` commit `aec1934cb2708236c033f2d205d5c3fbbebf7555` from https://github.com/meta-qt5/meta-qt5/tree/kirkstone
3. Build `qtbase`: `bitbake qtbase`
4. See error 

```
ERROR: qtbase-5.15.7+gitAUTOINC+358aebba72-r0 do_patch: Applying patch '0001-eglfs-Newer-Nvidia-libdrm-provide-device-instead-dri.patch' on target directory
```

madisongh, 2023-03-02:
@kekiefer Is f779da5c from kirkstone applicable here?

VitaliiBlagodir, 2023-03-02:
> @kekiefer Is [f779da5](https://github.com/OE4T/meta-tegra/commit/f779da5c7028d44d6c1b3bf9f44e795128ce3d7b) from kirkstone applicable here?

@madisongh Yes. Solution in https://github.com/OE4T/meta-tegra/commit/f779da5c7028d44d6c1b3bf9f44e795128ce3d7b fixed build.

I've temporary added to my `qtbase_%.bbappend`:
```
SRC_URI:remove = "file://0001-eglfs-Newer-Nvidia-libdrm-provide-device-instead-dri.patch"
```

kekiefer, 2023-03-02:
Yes, we have a lot of branches :)

madisongh, 2023-03-02:
I've cherry-picked f779da5 to the kirkstone-l4t-r32.7.x branch, so you should be able to drop your workaround when you update to the latest.

closed 2023-03-02 by madisongh