OE4T/meta-tegra #1214: Fetcher Failure: Nvidia-container-toolkit on Dunfell branch

ervgan, 2023-03-31:
**Describe the bug**
Hello,
I am getting the error message:
`nvidia-container-toolkit-1.0.5-r0 do_fetch: Fetcher failure: Unable to find revision 60f165ad6901f85b0c3acbf7ce2c66cd759c4fb8 in branch master even from upstream `
when trying to build the Meta tegra Dunfell branch. I think it's because "branch=main" is missing at the end of 

`SRC_URI = "git://github.com/NVIDIA/nvidia-container-toolkit.git;protocol=https"`

But I am not sure, can someone advise on how to solve the fetcher error?

I need this to work for L4T 32.7.2 on Dunfell branch, so I guess the best way is for me to create a new Dunfell branch dunfell-l4t-r32.7.2. with the necessary changes for nvidia-container-toolkit.

Thanks!

madisongh, 2023-03-31:
That got fixed a couple of weeks ago.  Had you not updated your local copy of the repo recently?

ervgan, 2023-03-31:
Yes sorry, I just saw you made the changes to the main dunfell branch but I was on an ealier commit since I need this to work for L4T 32.7.2. Is it possible to create a new branch dunfell-l4t-r32.7.2 which only has L4T 32.7.2 versions? I think you can create a branch from commit c69687d2f4fae77e016f8646351d6eb345d9f7f4, then I can create a PR to update the SRC_URI link for nvidia-toolkit-container for that branch. 

Thanks!

dwalkes, 2023-03-31:
@ervgan you might just want to make your own branch for this.

ervgan, 2023-03-31:
Thanks for the quick reply! I created my own dunfell-l4t-r32.7.2 branch but it is actually more desirable to have it in the main repo, otherwise the whole build process will have one layer depend on my private repo and we would need to clone from two separate repos. I think maintaining it would not be a big problem since it will mostly be updating links, paths and hashes, what do you think?

madisongh, 2023-04-02:
You don't actually need to fork and maintain a private branch of the meta-tegra repo to fix this particular problem - just add a bbappend in your distro layer (or some layer you do control) to override the SRC_URI setting.   I've had to do this from time to time for other recipes, when I haven't been able to update layers in my projects.

> I think maintaining it would not be a big problem since it will mostly be updating links, paths and hashes, what do you think?

There are [several](https://github.com/OE4T/meta-tegra/wiki/Which-branch-should-I-use%3F) active branches already, and I just don't have the bandwidth to add yet another, unfortunately.

ervgan, 2023-04-03:
Oh you mean I can override just the SRC_URI variable from the nvidia-toolkit-container.bb file? This would indeed be a better solution I'm not sure how I would use bbappend, is there an example somewhere in the repo? This is what my .bb file looks like now: 

`CORE_IMAGE_BASE_INSTALL += " \
  cuda-libraries \
  cuda-samples \
  cuda-toolkit \
  tensorrt-core \
  tensorrt-plugins \
  tensorrt-samples \
  tensorrt-trtexec \
  cudnn \
  libvisionworks \
  gstreamer1.0-plugins-nvvideo4linux2 \
  nvidia-docker \
  nvidia-container-runtime \
"`

dwalkes, 2023-04-03:
@ervgan that might work but I think what Matt was suggesting was for you to add the .bbappend in your own layer.  See https://docs.yoctoproject.org/dev-manual/common-tasks.html?highlight=override+src_uri#understanding-and-creating-layers for an overview.  This way you don't need to add to `BBFILES` and you can track other layer customizations you'll ultimately need.

closed 2023-04-04 by ervgan