OE4T/meta-tegra #1215: No serial console and cannot boot from extlinux.conf on Jetson Orin AGX

Tom2096, 2023-04-01:
Hi all. I am new to the yocto project and am still getting used to the meta-tegra layer. I have been playing around with the `kirkstone` branch of the `tegra-demo-distro` repo trying to get it working on the `Jetson Orin AGX`. I set up the enviroment as per the instructions listed in the Readme file and built the image with 

```bitbake demo-image-full```

Upon booting the Orin, I can see the sato graphics UI but nothing on the serial console:
```
L4TLauncher: Attempting Direct Boot
OpenAndReadUntrustedFileToBuffer: Failed to open boot\extlinux\extlinux.conf: No
t Found
ProcessExtLinuxConfig:sds Failed to Authenticate boot\extlinux\extlinux.conf (No
t Found)
L4TLauncher: Unable to process extlinux config: Not Found
L4TLauncher: Attempting Kernel Boot
EFI stub: Booting Linux Kernel...
EFI stub: Using DTB from configuration table
EFI stub: Loaded initrd from LINUX_EFI_INITRD_MEDIA_GUID device path
EFI stub: Exiting boot services and installing virtual address map...
```
From what's shown on the serial console I'm guessing that the L4T Launcher is not finding the `/boot/extlinux/extlinux.conf` file. Looking into the `/boot` directory with the graphic interface shows that there is only a `/boot/efi/EFI/BOOT/bootaa64.efi` file nested in that directory. 

After some digging around in the meta-tegra layer, I found a `/meta-tegra/classes/l4t-extlinux-config.bbclass` file that seems to provide instructions for creating `/boot/extlinux/extlinux.conf` as well as both `/boot/Image` and `/boot/initrd`. Tracking this down led me to the `UBOOT_EXTLINUX` variable in `/meta-tegra/conf/machine/include/tegra-common.inc` that starts the `l4t-launcher-extlinux` task which generates those files. 

Setting `UBOOT_EXTLINUX ?= '1'` and building the image again seems to have populated `/boot` with `/boot/extlinux/extlinux.conf`, `/boot/Image`, and `/boot/initrd`. However, upon booting the board again the serial console still complains with the same message shown before and just seems to default to booting from the kernel partition. 

I have reviewed a similar issue https://github.com/OE4T/meta-tegra/issues/998 but the problem I'm facing differs in that 
- I cannot switch to booting from `/boot/extlinux/extlinux.conf`.
- There is no serial console output when forced to boot from the kernel partition. Moreover, `cat /proc/cmdline` does seem to show the correct kernel arguments: 
```mminit_loglevel=4 console=tty0 console=ttyTCU0,115200 firmware_class.path=/etc/firmware fbcon=map:0 net.ifnames=0```

Sorry for the long read and for any incorrect terminologies. Thanks in advance and I am happy to provide any additional files.

madisongh, 2023-04-02:
* For the serial console output problem, try removing `console=tty0` from the `KERNEL_ARGS` setting.
* For the `extlinux.conf` problem, once you have set the `UBOOT_EXTLINUX` variable to generate it in your rootfs image, are you still seeing the same messages from the EFI bootloader about the `extlinux.conf` file not being found?

Tom2096, 2023-04-02:
Hi, thanks for the response.

After booting the image with `UBOOT_EXTLINUX ?= 1` this is what I see now in the serial console
```
L4TLauncher: Attempting Direct Boot
EFI stub: Booting Linux Kernel...
EFI stub: Using DTB from configuration table
EFI stub: Loaded initrd from LINUX_EFI_INITRD_MEDIA_GUID device path
EFI stub: Exiting boot services and installing virtual address map...
```

Which seems to indicate that it has found and is booting from the `extlinux.conf` file. To verify this, I removed `console=tty0` from the kernel arguments in `extlinux.conf` and used `cat /proc/cmdline` to verify that the change persisted:
```
mminit_loglevel=4 console=ttyTCU0,115200 firmware_class.path=/etc/firmware fbcon=map:0 net.ifnames=0
```

However, I am still not seeing anything except for the message above on the serial console.

Tom2096, 2023-04-02:
Changing the kernel argument from `console=ttyTCU0,115200` to `console=ttyAMA0,115200` seemed to have fixed the issue. Maybe for micro-usb connections it's a different port? Thanks for the help!

closed 2023-04-02 by Tom2096

om26er, 2024-11-23:
@madisongh using scarthgap branch on jetson orin agx (64gb) -- I have below variables set
```
UBOOT_EXTLINUX = "1"
UBOOT_EXTLINUX_FDT = "/boot/${DTBFILE}"
```
and yet it seems the boot process loads the kernel and dtb from the partitions instead of the rootfs' `/boot/` directory.

Here are the boot logs
```
      L4TLauncher: Attempting Direct Boot
OpenAndReadUntrustedFileToBuffer: Failed to open boot\extlinux\extlinux.conf: No
t Found
ProcessExtLinuxConfig:sds Failed to Authenticate boot\extlinux\extlinux.conf (No
t Found)
L4TLauncher: Unable to process extlinux config: Not Found
L4TLauncher: Attempting Kernel Boot
EFI stub: Booting Linux Kernel...
EFI stub: Using DTB from configuration table
EFI stub: Loaded initrd from LINUX_EFI_INITRD_MEDIA_GUID device path
EFI stub: Exiting boot services...
```

Is there a way I can change that?

dwalkes, 2024-11-24:
Have you tried to reproduce on tegra-demo-distro?  If not I'd start there.  Also might be better to open a new issue than comment on this closed one which was referencing kirkstone.  UBOOT_EXTLINUX [should default to 1 already on scarthgap](https://github.com/OE4T/meta-tegra/blob/1dbb1d1a1aa4f65301bfc1c0685f6eda572d11b3/conf/machine/include/tegra-common.inc#L81) I believe.

madisongh, 2024-11-24:
IIRC, for the extlinux.conf support to work, `/boot` has to be on the APP partition, and that partition has to be on a filesystem that UEFI can mount - VFAT or ext4 (since NVIDIA added a UEFI driver to support ext4). If you are using one of those filesystems for your APP partition, and it's not working, you might consider increasing the log level in UEFI/L4TLauncher to find out why it's unable to locate the file it's trying to read.

If you've customized your system to use some other filesystem, or you're using LUKS to encrypt your rootfs, you'll need to split `/boot` out into a separate partition with a usable filesystem and no LUKS. The stock L4T storage layouts that NVIDIA provides when you opt for encrypted rootfs uses this approach.

om26er, 2024-11-24:
That would explain it. My rootfs is squashfs. I am now working to see if I can just dd images to kernel and dtb partitions as part of my rauc installation script