OE4T/meta-tegra #1243: Enable NVMM support for `gstreamer1.0-plugins-base >= 1.20`

molysgaard, 2023-04-29:
**Describe the bug**
In `gstreamer1.0-plugins-base >= 1.20`, the GStreamer elements `glupload` and `gldownload` has direct support for the tegra `NVMM` type buffers.
This means that one can convert to and from `OpenGL` buffers and `NVMM` without doing a full buffer copy, which is critical for not tanking performance in high-resolution live video applications.

Currently, there are some challenges to getting this support enabled though.

Going through the GStreamer build system, one needs to get the `HAVE_NVMM` define set up.
Reference: https://gitlab.freedesktop.org/gstreamer/gstreamer/-/blob/main/subprojects/gst-plugins-base/ext/gl/gstgldownloadelement.c#L37

To get this define set up, one needs OpenGL and EGL support, which is already supported through the PACKAGECONFIG system :tada:
_But_, one also needs Nvidia-specific proprietary headers available during the build phase of `gstreamer1.0-plugins-base`, namely `nvbuf_utils.h`.
Reference: https://gitlab.freedesktop.org/gstreamer/gstreamer/-/blob/main/subprojects/gst-plugins-base/gst-libs/gst/gl/meson.build#L958

This header is supplied by `tegra-mmapi`.
I tried letting `gstreamer1.0-plugins-base` DEPEND on `tegra-mmapi`, but that led to a dependency cycle.
The cycle seems to be the following:
`tegra-mmapi` `DEPENS` on `tegra-libraries-multimedia` which `DEPENDS` on `gstreamer1.0-plugins-base`.

I believe this cycle is not really needed, I think `tegra-mmapi` should be possible to build without any reference to `gstreamer1.0-plugins-base`. But I do not know the best way to break this cycle.

I would be very thankful for any insight that could lead to fixing this.

I am basing my conclusions on this revision of meta-tegra: `9d58efcf`

Please tell me if there is any more information I can supply to help with this issue.

dwalkes, 2023-04-29:
Hi @molysgaard 

> I am basing my conclusions on this revision of meta-tegra: `9d58efcf`

I think this is https://github.com/OE4T/meta-tegra/commit/9d58efcfad33a0496b2099e01111c4c937bf3287, a commit which is several months old on the kirkstone branch.

However I've reproduced on the latest master with

```
diff --git a/recipes-multimedia/gstreamer/gstreamer1.0-plugins-base_%.bbappend b/recipes-multimedia/gstreamer/gstreamer1.0-plugins-base_%.bbappend
new file mode 100644
index 00000000..86eff608
--- /dev/null
+++ b/recipes-multimedia/gstreamer/gstreamer1.0-plugins-base_%.bbappend
@@ -0,0 +1 @@
+DEPENDS += "tegra-mmapi"
```

Full error below:

```
Dependency loop #4 found:
  Task /home/dan/trellis/tegra-demo-distro/layers/meta-tegra/recipes-bsp/tegra-binaries/tegra-libraries-camera_35.3.1.bb:do_compile (dependent Tasks ['tegra-libraries-camera_35.3.1.bb:do_configure'])
  Task /home/dan/trellis/tegra-demo-distro/layers/meta-tegra/recipes-bsp/tegra-binaries/tegra-libraries-camera_35.3.1.bb:do_install (dependent Tasks ['pseudo_git.bb:do_populate_sysroot', 'tegra-libraries-camera_35.3.1.bb:do_compile'])
  Task /home/dan/trellis/tegra-demo-distro/layers/meta-tegra/recipes-bsp/tegra-binaries/tegra-libraries-camera_35.3.1.bb:do_populate_sysroot (dependent Tasks ['tegra-libraries-camera_35.3.1.bb:do_install', 'binutils-cross_2.40.bb:do_populate_sysroot'])
  Task /home/dan/trellis/tegra-demo-distro/layers/meta-tegra/recipes-multimedia/argus/tegra-mmapi_35.3.1.bb:do_prepare_recipe_sysroot (dependent Tasks ['tegra-libraries-multimedia_35.3.1.bb:do_populate_sysroot', 'pseudo_git.bb:do_populate_sysroot', 'tegra-libraries-multimedia-utils_35.3.1.bb:do_populate_sysroot', 'tegra-mmapi_35.3.1.bb:do_fetch', 'tegra-libraries-camera_35.3.1.bb:do_populate_sysroot', 'gcc-cross_12.2.bb:do_populate_sysroot', 'gcc-runtime_12.2.bb:do_populate_sysroot', 'glibc_2.37.bb:do_populate_sysroot', 'libglvnd_1.6.0.bb:do_populate_sysroot'])
  Task /home/dan/trellis/tegra-demo-distro/layers/meta-tegra/recipes-multimedia/argus/tegra-mmapi_35.3.1.bb:do_configure (dependent Tasks ['tegra-mmapi_35.3.1.bb:do_prepare_recipe_sysroot', 'tegra-mmapi_35.3.1.bb:do_patch', 'tegra-mmapi_35.3.1.bb:do_deploy_source_date_epoch'])
  Task /home/dan/trellis/tegra-demo-distro/layers/meta-tegra/recipes-multimedia/argus/tegra-mmapi_35.3.1.bb:do_compile (dependent Tasks ['tegra-mmapi_35.3.1.bb:do_configure'])
  Task /home/dan/trellis/tegra-demo-distro/layers/meta-tegra/recipes-multimedia/argus/tegra-mmapi_35.3.1.bb:do_install (dependent Tasks ['tegra-mmapi_35.3.1.bb:do_compile', 'pseudo_git.bb:do_populate_sysroot'])
  Task /home/dan/trellis/tegra-demo-distro/layers/meta-tegra/recipes-multimedia/argus/tegra-mmapi_35.3.1.bb:do_populate_sysroot (dependent Tasks ['tegra-mmapi_35.3.1.bb:do_install', 'binutils-cross_2.40.bb:do_populate_sysroot'])
  Task /home/dan/trellis/tegra-demo-distro/layers/meta/recipes-multimedia/gstreamer/gstreamer1.0-plugins-base_1.22.0.bb:do_prepare_recipe_sysroot (dependent Tasks ['libtheora_1.1.1.bb:do_populate_sysroot', 'pseudo_git.bb:do_populate_sysroot', 'libxrender_0.9.11.bb:do_populate_sysroot', 'glib-2.0_2.74.6.bb:do_populate_sysroot', 'alsa-lib_1.2.8.bb:do_populate_sysroot', 'gstreamer1.0_1.22.0.bb:do_populate_sysroot', 'libvorbis_1.3.7.bb:do_populate_sysroot', 'wayland-protocols_1.31.bb:do_populate_sysroot', 'qemu-native_7.2.0.bb:do_populate_sysroot', 'ninja_1.11.1.bb:do_populate_sysroot', 'libsm_1.2.4.bb:do_populate_sysroot', 'orc_0.4.33.bb:do_populate_sysroot', 'libxv_1.0.12.bb:do_populate_sysroot', 'wayland_1.22.0.bb:do_populate_sysroot', 'libjpeg-turbo_2.1.5.1.bb:do_populate_sysroot', 'tegra-mmapi_35.3.1.bb:do_populate_sysroot', 'gcc-cross_12.2.bb:do_populate_sysroot', 'meson_1.0.1.bb:do_populate_sysroot', 'wayland_1.22.0.bb:do_populate_sysroot', 'glibc_2.37.bb:do_populate_sysroot', 'gobject-introspection_1.74.0.bb:do_populate_sysroot', 'gettext_0.21.1.bb:do_populate_sysroot', 'iso-codes_4.13.0.bb:do_populate_sysroot', 'libogg_1.3.5.bb:do_populate_sysroot', 'libglvnd_1.6.0.bb:do_populate_sysroot', 'orc_0.4.33.bb:do_populate_sysroot', 'gobject-introspection_1.74.0.bb:do_populate_sysroot', 'zlib_1.2.13.bb:do_populate_sysroot', 'util-linux_2.38.1.bb:do_populate_sysroot', 'libpng_1.6.39.bb:do_populate_sysroot', 'libx11_1.8.4.bb:do_populate_sysroot', 'pkgconfig_git.bb:do_populate_sysroot', 'libdrm_2.4.115.bb:do_populate_sysroot', 'pango_1.50.14.bb:do_populate_sysroot', 'gcc-runtime_12.2.bb:do_populate_sysroot', 'gstreamer1.0-plugins-base_1.22.0.bb:do_fetch', 'python3_3.11.2.bb:do_populate_sysroot'])
  Task /home/dan/trellis/tegra-demo-distro/layers/meta/recipes-multimedia/gstreamer/gstreamer1.0-plugins-base_1.22.0.bb:do_configure (dependent Tasks ['gstreamer1.0-plugins-base_1.22.0.bb:do_write_config', 'gstreamer1.0-plugins-base_1.22.0.bb:do_patch', 'gstreamer1.0-plugins-base_1.22.0.bb:do_prepare_recipe_sysroot', 'gstreamer1.0-plugins-base_1.22.0.bb:do_deploy_source_date_epoch'])
  Task /home/dan/trellis/tegra-demo-distro/layers/meta/recipes-multimedia/gstreamer/gstreamer1.0-plugins-base_1.22.0.bb:do_compile (dependent Tasks ['gstreamer1.0-plugins-base_1.22.0.bb:do_configure'])
  Task /home/dan/trellis/tegra-demo-distro/layers/meta/recipes-multimedia/gstreamer/gstreamer1.0-plugins-base_1.22.0.bb:do_install (dependent Tasks ['gstreamer1.0-plugins-base_1.22.0.bb:do_compile', 'pseudo_git.bb:do_populate_sysroot'])
[... cut ...] .bb:do_populate_sysroot', 'libglvnd_1.6.0.bb:do_populate_sysroot'])
  Task /home/dan/trellis/tegra-demo-distro/layers/meta-tegra/recipes-bsp/tegra-binaries/tegra-libraries-camera_35.3.1.bb:do_configure (dependent Tasks ['tegra-libraries-camera_35.3.1.bb:do_prepare_recipe_sysroot', 'tegra-libraries-camera_35.3.1.bb:do_patch', 'tegra-libraries-camera_35.3.1.bb:do_deploy_source_date_epoch'])
```

I'd need to dig into this to help diagnose, I just thought I'd start by looking for a change on master.

molysgaard, 2023-04-30:
Yeah, the issue can be worked on from master.

I am not an bitbake expert, but I can help out with the actual work if someone can guide me on the best way to split this cycle.

madisongh, 2023-04-30:
I've looked at this, and have opened #1244 with the changes that should break this dependency loop.

PR #1244 (2023-05-05): Refactor tegra-libraries-multimedia
Creates a new recipe to install the `libnvdsbufferpool.so` library so that the main `tegra-libraries-multimedia` recipe doesn't have to depend on `gstreamer1.0-plugins-base`.

Fixes #1243
closed 2023-05-05 by madisongh