OE4T/meta-tegra #1246: gcc-8 internal recipe broken since crosssdk is droped in oe-core

quaresmajose, 2023-05-02:
**Describe the bug**

oe-core [1] drop crosssdk suffix from virtual provides to improve dependency handling
and this breakes meta-tegra on master branch 

[1] https://git.yoctoproject.org/poky/commit/?id=9a20fbe23f63d87e30b0291ae1dba863c0f5de31

```
ERROR: Nothing PROVIDES 'virtual/x86_64-oesdk-linux-binutils-crosssdk' (but /srv/oe/build/conf/../../layers/meta-tegra/recipes-devtools/gcc-for-nvcc/gcc-8-cross-canadian_8.5.bb DEPENDS on or otherwise requires it). Close matches:
  virtual/x86_64-oesdk-linux-binutils
  virtual/x86_64-oesdk-linux-cuda-g++-crosssdk
  virtual/x86_64-oesdk-linux-cuda-gcc-crosssdk
ERROR: Nothing PROVIDES 'virtual/x86_64-oesdk-linux-gcc-crosssdk' (but /srv/oe/build/conf/../../layers/meta-tegra/recipes-devtools/gcc-for-nvcc/gcc-8-cross-canadian_8.5.bb DEPENDS on or otherwise requires it). Close matches:
  virtual/x86_64-oesdk-linux-cuda-g++-crosssdk
  virtual/x86_64-oesdk-linux-cuda-gcc-crosssdk
  virtual/x86_64-oesdk-linux-gcc
ERROR: Nothing PROVIDES 'virtual/x86_64-oesdk-linux-binutils-crosssdk' (but /srv/oe/build/conf/../../layers/meta-tegra/recipes-devtools/gcc-for-nvcc/gcc-8-cross-canadian_8.5.bb DEPENDS on or otherwise requires it). Close matches:
  virtual/x86_64-oesdk-linux-binutils
  virtual/x86_64-oesdk-linux-cuda-g++-crosssdk
  virtual/x86_64-oesdk-linux-cuda-gcc-crosssdk
ERROR: Nothing PROVIDES 'virtual/x86_64-oesdk-linux-gcc-crosssdk' (but /srv/oe/build/conf/../../layers/meta-tegra/recipes-devtools/gcc-for-nvcc/gcc-8-cross-canadian_8.5.bb DEPENDS on or otherwise requires it). Close matches:
  virtual/x86_64-oesdk-linux-cuda-g++-crosssdk
  virtual/x86_64-oesdk-linux-cuda-gcc-crosssdk
  virtual/x86_64-oesdk-linux-gcc
ERROR: Nothing PROVIDES 'virtual/x86_64-oesdk-linux-binutils-crosssdk' (but /srv/oe/build/conf/../../layers/meta-tegra/recipes-devtools/gcc-for-nvcc/gcc-8-cross-canadian_8.5.bb DEPENDS on or otherwise requires it). Close matches:
  virtual/x86_64-oesdk-linux-binutils
  virtual/x86_64-oesdk-linux-cuda-g++-crosssdk
  virtual/x86_64-oesdk-linux-cuda-gcc-crosssdk
ERROR: Nothing PROVIDES 'virtual/x86_64-oesdk-linux-gcc-crosssdk' (but /srv/oe/build/conf/../../layers/meta-tegra/recipes-devtools/gcc-for-nvcc/gcc-8-cross-canadian_8.5.bb DEPENDS on or otherwise requires it). Close matches:
  virtual/x86_64-oesdk-linux-cuda-g++-crosssdk
  virtual/x86_64-oesdk-linux-cuda-gcc-crosssdk
  virtual/x86_64-oesdk-linux-gcc
ERROR: Nothing PROVIDES 'virtual/x86_64-oesdk-linux-binutils-crosssdk' (but /srv/oe/build/conf/../../layers/meta-tegra/recipes-devtools/gcc-for-nvcc/gcc-8-cross-canadian_8.5.bb DEPENDS on or otherwise requires it). Close matches:
  virtual/x86_64-oesdk-linux-binutils
  virtual/x86_64-oesdk-linux-cuda-g++-crosssdk
  virtual/x86_64-oesdk-linux-cuda-gcc-crosssdk
ERROR: Nothing PROVIDES 'virtual/x86_64-oesdk-linux-gcc-crosssdk' (but /srv/oe/build/conf/../../layers/meta-tegra/recipes-devtools/gcc-for-nvcc/gcc-8-cross-canadian_8.5.bb DEPENDS on or otherwise requires it). Close matches:
  virtual/x86_64-oesdk-linux-cuda-g++-crosssdk
  virtual/x86_64-oesdk-linux-cuda-gcc-crosssdk
  virtual/x86_64-oesdk-linux-gcc
ERROR: Nothing PROVIDES 'virtual/x86_64-oesdk-linux-binutils-crosssdk' (but /srv/oe/build/conf/../../layers/meta-tegra/recipes-devtools/gcc-for-nvcc/gcc-8-cross-canadian_8.5.bb DEPENDS on or otherwise requires it). Close matches:
  virtual/x86_64-oesdk-linux-binutils
  virtual/x86_64-oesdk-linux-cuda-g++-crosssdk
  virtual/x86_64-oesdk-linux-cuda-gcc-crosssdk
ERROR: Nothing PROVIDES 'virtual/x86_64-oesdk-linux-gcc-crosssdk' (but /srv/oe/build/conf/../../layers/meta-tegra/recipes-devtools/gcc-for-nvcc/gcc-8-cross-canadian_8.5.bb DEPENDS on or otherwise requires it). Close matches:
  virtual/x86_64-oesdk-linux-cuda-g++-crosssdk
  virtual/x86_64-oesdk-linux-cuda-gcc-crosssdk
  virtual/x86_64-oesdk-linux-gcc
```

**To Reproduce**
Steps to reproduce the behavior:

Can be reproduces building something that needs gcc.

**Additional context**

ichergui, 2023-05-02:
Hello @quaresmajose 
Is this master branch of OE core ?

ichergui, 2023-05-02:
Thanks for sharing this with us. 
I will take closely to it over the weekend

quaresmajose, 2023-05-03:
It build in my side backporting the changes from oe-core

closed 2023-05-03 by quaresmajose