OE4T/meta-tegra #1248: Updated R35.3.1 USB firmware

teebr, 2023-05-03:
Hello,

Just flagging that there is an update to [USB firmware](https://developer.nvidia.com/downloads/embedded/l4t/r35_release_v3.1/overlay_xusb_35.3.1.tbz2/) for [R35.3.1](https://developer.nvidia.com/embedded/jetson-linux-r3531) which fixes USB-serial issues, and doesn't seem to be in the Yocto layer yet. I've patched it in my Orin AGX build by doing the following:
- `bbappend`-ing `tegra_bootfiles` to delete the existing file
- add a new recipe to install the new file in the same path
- add `DEPENDS` to the new recipe in`tegra-bootfiles`

It's a bit messy because `do_fetch` is a no-op for the original recipe; there's some discussion on how to handle this [here](https://github.com/OE4T/meta-tegra/discussions/1094). I'm not sure if that approach would suffice for this layer, or whether to try and split up the `tegra-bootfiles` and make it more amenable to overwriting certain files?

madisongh, 2023-05-03:
The approach you took sounds like the right one for now. Now that I know about the update (thanks for that), I'll update the `tegra-binaries` recipe to add that overlay to the `SRC_URI` for that recipe.

teebr, 2023-05-05:
great, thanks Matt!

madisongh, 2023-05-05:
I've update the `tegra-binaries` recipe in master, mickledore, and kirkstone branches to pull in the overlay, so you shouldn't need your workaround any longer.

closed 2023-05-05 by madisongh