OE4T/meta-tegra #1259: TX2 support for Langdale and newer

rustand2, 2023-05-25:
From what I can see Kirkstone is the latest Yocto release that is supported for Jetson TX2 by this layer. Are there any plans to add support for TX2 in the newer meta-tegra branches?

rustand2, 2023-05-26:
So Nvidia has no plans to keep supporting TX2 in newer Jetpacks? And meta-tegra is dependent on Nvidia to release newer Jetpacks in order to update the kernel?

madisongh, 2023-05-26:
> So Nvidia has no plans to keep supporting TX2 in newer Jetpacks? And meta-tegra is dependent on Nvidia to release newer Jetpacks in order to update the kernel?

That's correct, TX1/Nano and TX platforms are not supported in JetPack 5/L4T R35.x. It might be possible to switch to a mainline kernel, but that wouldn't be compatible with any of the JetPack packages, since those rely on the downstream kernel changes and drivers that NVIDIA supplies.

rustand2, 2023-05-26:
Well, as far as I know I do not use any Jetpack packages unless they are used by the system in some way. I just use it as a  normal Linux machine. I was under the impression that these things were somehow required in order for Linux to work on these machines.

madisongh, 2023-05-26:
You might be able to reuse the tegraflash packaging and scripts from here, but otherwise this layer is pretty closely coupled with NVIDIA's BSP, and would try to install their userland libraries, which are tied to their downstream kernel and GPU driver and wouldn't be usable with a mainline kernel.

I'd expect this to be a bunch of work, and unfortunately I can't even dig up any links to provide as references to start with.

dwalkes, 2023-05-26:
> I'd expect this to be a bunch of work, and unfortunately I can't even dig up any links to provide as references to start with.

See https://github.com/OE4T/meta-tegra/discussions/965 for a thread on nano, and https://elinux.org/Jetson/Nano/Upstream for a page on an upstream kernel build for nano.

>  I do not use any Jetpack packages unless they are used by the system in some way. I just use it as a normal Linux machine

If you don't need the GPU and can consider another platform you might just use an RPi 4 and save yourself some headaches and cash.

closed 2023-07-15 by madisongh